ros2/rmw_fastrtps #382: rmw_fastrtps_dynamic_cpp shouldn't require rosidl_typesupport_fastrtps_*

ivanpauno, 2020-05-12:
## Feature request

#### Feature description

Currently, it's impossible to make a single rmw single typesupport build for `rmw_fastrtps_dynamic_cpp`, as it currently depends on `rosidl_typesupport_fastrtps_c/cpp`:

https://github.com/ros2/rmw_fastrtps/blob/98d14729df0aa6d1ce01d7165d91e7cee257dd5c/rmw_fastrtps_dynamic_cpp/package.xml#L27-L28

That dependency was introduced to use some utility functions available there.
It could be removed if the utility functions are moved to a common package.

MiguelCompany, 2020-10-02:
> That dependency was introduced to use some utility functions available there.
> It could be removed if the utility functions are moved to a common package

The dependency is there just for the `u16string <-> wstring` conversions. Perhaps `rosidl_runtime_c` and/or `rosidl_runtime_cpp` are good candidates for those functions?

ivanpauno, 2020-10-02:
Sounds good to me

ZhenshengLee, 2023-11-29:
Can be closed?

MiguelCompany, 2023-11-29:
I don't think so, since `rmw_fastrtps_dynamic_cpp` still requires `rosidl_typesupport_fastrtps_*` [here](https://github.com/ros2/rmw_fastrtps/blob/rolling/rmw_fastrtps_dynamic_cpp/CMakeLists.txt#L47-L48)

PR #740 (2024-02-20): Improve wide string (de)serialization in rwm_dynamic_fastrtps_cpp
This PR:

1. Moves private headers to `src`
2. Changes (de)serialization of wide strings so they don't perform `u16string` <-> `wstring` conversions
3. Fixes an off-by-one mistake in the calculation of the serialized size of wide strings
4. Removes dependencies to `rosidl_typesupport_fastrtps_c(pp)`, i.e. fixes #382
closed 2024-02-20 by fujitatomoya