ros2/rmw_fastrtps #391: WARNING: ThreadSanitizer: data race

fujitatomoya, 2020-05-27:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/0444bff7b23484fffdb78d170801e39cadf514f9
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
> cd colcon_ws
> cd src
> git clone git@github.com:fujitatomoya/ros2_test_prover.git
> cd ..
> colcon build --symlink-install --packages-select ros2_test_prover
> source install/local_setup.bash
> for i in {1..300}; do ros2 run ros2_test_prover rclcpp_1121_repro; done
```

#### Expected behavior

finish w/o error/warning.

#### Actual behavior

Thread Sanitizer detects the following data race condition.

```
WARNING: ThreadSanitizer: data race (pid=12656)
  Write of size 8 at 0x7ba000004070 by thread T2 (mutexes: write M195760417463603536, write M199138117184131472, write M208712873003479200):
    #0 close <null> (libtsan.so.0+0x345aa)
    #1 eprosima::fastdds::rtps::UDPChannelResource::release() <null> (libfastrtps.so.2+0x320cf7)
    #2 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() /usr/include/c++/9/bits/shared_ptr_base.h:730 (rclcpp_1121_repro+0x39026)
    #3 std::__shared_ptr<rcl_node_t, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() /usr/include/c++/9/bits/shared_ptr_base.h:1169 (rclcpp_1121_repro+0x3d829)
    #4 std::shared_ptr<rcl_node_t>::~shared_ptr() /usr/include/c++/9/bits/shared_ptr.h:103 (rclcpp_1121_repro+0x3d859)
    #5 rclcpp::node_interfaces::NodeBase::~NodeBase() <null> (librclcpp.so+0x594d17)
    #6 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() /usr/include/c++/9/bits/shared_ptr_base.h:730 (rclcpp_1121_repro+0x39026)
    #7 std::__shared_ptr<rclcpp::node_interfaces::NodeBaseInterface, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() /usr/include/c++/9/bits/shared_ptr_base.h:1169 (rclcpp_1121_repro+0x3838d)
    #8 std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>::~shared_ptr() /usr/include/c++/9/bits/shared_ptr.h:103 (rclcpp_1121_repro+0x383f9)
    #9 rclcpp::Node::~Node() <null> (librclcpp.so+0x58431f)
    #10 void std::allocator_traits<std::allocator<rclcpp::Node> >::destroy<rclcpp::Node>(std::allocator<rclcpp::Node>&, rclcpp::Node*) /usr/include/c++/9/bits/alloc_traits.h:497 (rclcpp_1121_repro+0x51b54)
    #11 std::_Sp_counted_ptr_inplace<rclcpp::Node, std::allocator<rclcpp::Node>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() /usr/include/c++/9/bits/shared_ptr_base.h:557 (rclcpp_1121_repro+0x501dc)
    #12 std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() /usr/include/c++/9/bits/shared_ptr_base.h:155 (rclcpp_1121_repro+0x3bf20)
    #13 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() /usr/include/c++/9/bits/shared_ptr_base.h:730 (rclcpp_1121_repro+0x39026)
    #14 std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() /usr/include/c++/9/bits/shared_ptr_base.h:1169 (rclcpp_1121_repro+0x387f1)
    #15 std::shared_ptr<rclcpp::Node>::~shared_ptr() /usr/include/c++/9/bits/shared_ptr.h:103 (rclcpp_1121_repro+0x38821)
    #16 runService(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::atomic<bool>&) /root/docker_ws/ros2_colcon/src/fujitatomoya/ros2_test_prover/src/rclcpp_1121_repro.cpp:12 (rclcpp_1121_repro+0x209b3)
    #17 operator() /root/docker_ws/ros2_colcon/src/fujitatomoya/ros2_test_prover/src/rclcpp_1121_repro.cpp:113 (rclcpp_1121_repro+0x21432)
    #18 __invoke_impl<void, main(int, char**)::<lambda()> > /usr/include/c++/9/bits/invoke.h:60 (rclcpp_1121_repro+0x3209b)
    #19 __invoke<main(int, char**)::<lambda()> > /usr/include/c++/9/bits/invoke.h:95 (rclcpp_1121_repro+0x31d2a)
    #20 _M_invoke<0> /usr/include/c++/9/thread:244 (rclcpp_1121_repro+0x319a4)
    #21 operator() /usr/include/c++/9/thread:251 (rclcpp_1121_repro+0x3162f)
    #22 operator() /usr/include/c++/9/future:1362 (rclcpp_1121_repro+0x30b7f)
    #23 _M_invoke /usr/include/c++/9/bits/std_function.h:286 (rclcpp_1121_repro+0x2f696)
    #24 std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>::operator()() const /usr/include/c++/9/bits/std_function.h:688 (rclcpp_1121_repro+0x3981c)
    #25 std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) /usr/include/c++/9/future:561 (rclcpp_1121_repro+0x371e6)
    #26 void std::__invoke_impl<void, void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::__invoke_memfun_deref, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) /usr/include/c++/9/bits/invoke.h:73 (rclcpp_1121_repro+0x3f0fc)
    #27 std::__invoke_result<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>::type std::__invoke<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) /usr/include/c++/9/bits/invoke.h:95 (rclcpp_1121_repro+0x3c1ee)
    #28 std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&)::{lambda()#1}::operator()() const /usr/include/c++/9/mutex:671 (rclcpp_1121_repro+0x393bb)
    #29 std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&)::{lambda()#2}::operator()() const /usr/include/c++/9/mutex:676 (rclcpp_1121_repro+0x39430)
    #30 std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::fu [... cut ...]
    #31 __invoke<std::__future_base::_Async_state_impl<_BoundFn, _Res>::_Async_state_impl(_BoundFn&&) [with _BoundFn = std::thread::_Invoker<std::tuple<main(int, char**)::<lambda()> > >; _Res = void]::<lambda()> > /usr/include/c++/9/bits/invoke.h:95 (rclcpp_1121_repro+0x35417)
    #32 _M_invoke<0> /usr/include/c++/9/thread:244 (rclcpp_1121_repro+0x34f22)
    #33 operator() /usr/include/c++/9/thread:251 (rclcpp_1121_repro+0x348ad)
    #34 _M_run /usr/include/c++/9/thread:195 (rclcpp_1121_repro+0x32986)
    #35 <null> <null> (libstdc++.so.6+0xd6cb3)

SUMMARY: ThreadSanitizer: data race (/lib/x86_64-linux-gnu/libtsan.so.0+0x345aa) in __interceptor_close
```

fujitatomoya, 2020-05-29:
@MiguelCompany 

friendly ping,

according to Tthread Sanitizer,

https://github.com/eProsima/Fast-RTPS/blob/23b4664d20f6194789a526d36c9b60f4576ed068/include/fastdds/rtps/transport/UDPChannelResource.h#L172 should be protected by mutex.

There is a data race condition between recv and close.

MiguelCompany, 2020-05-29:
As `socket_` is only assigned on the construction of `UDPChannelResource`, I don't see the data race as harmful.

Protecting with a mutex while we are doing a blocking call on `recv` would prevent the listening thread to stop until something is received. Closing the socket makes `recv` return with an error, which is exactly what we need.

Is there a way to annotate that we know this race condition is not harmful, so Thread Sanitizer does not complain?

fujitatomoya, 2020-05-29:
with https://github.com/eProsima/Fast-RTPS/pull/1239, i cannot reproduce the Thread Sanitizer warning.

> Protecting with a mutex while we are doing a blocking call on recv would prevent the listening thread to stop until something is received.

totally understood, shutdown() disables send/receive.

> Closing the socket makes recv return with an error, which is exactly what we need.

probably right, but i believe it does not make it exclusive access.

https://github.com/eProsima/Fast-RTPS/pull/1239 is proposal, So if you are not comfortable you can close it.

just in case i confirmed,

- ROS2 colcon test. (colcon test & colcon test-result --verbose)
- Cannot reproduce the reported sanitizer warning.

fujitatomoya, 2020-05-30:
@MiguelCompany 

I am closing this, refer to my comment https://github.com/eProsima/Fast-RTPS/pull/1239#issuecomment-636288469.

maybe we can open the issue when we actually meet the problem.

closed 2020-05-30 by fujitatomoya