ros2/rmw_fastrtps #392: Improve service discovery

hidmic, 2020-05-28:
## Feature request

#### Feature description
Follow-up issue after #390. Prior to that PR, services' discovery would rely on built-in request/reply topics discovery, which can and did lead to races (see https://github.com/ros2/ros2/issues/922). A short-term workaround was then introduced by #390, but not without caveats i.e. blocking behavior of `rmw_send_response`, non-standard (ab)use of sample identity data.

[OMG DDS-RPC 1.0 spec](https://www.omg.org/spec/DDS-RPC/1.0/PDF) describes both the issues associated with independent discovery of request/reply topics (see _Basic Service Mapping_ under section 7.6.1) and an algorithm to prevent them (see _Enhanced Service Mapping_ under section 7.6.2). Services' implementation in `rmw_fastrtps*_cpp` could and probably should move in that direction.

sloretz, 2020-06-05:
@richiware Friendly FYI. Does eProsima have any plans to work on this one?

richiware, 2020-06-08:
We were talking this morning about this. We have in mind to support deeply in the future the DDS-RPC standard on FastDDS. But right now the service discovery issue will be addressed by #390 solution. There are other features/issues that need our focus.

hidmic, 2020-06-08:
@richiware #390 is not a solution, it is a workaround at best and seemingly not an effective one for all cases (see https://github.com/ros2/rclcpp/issues/1152 and https://github.com/ros2/ros2/issues/931). To the extent that it is possible, please, do consider addressing this issue at some point in the near future.

ivanpauno, 2020-06-08:
I agree with @hidmic that #390 is a workaround.

Though a full implementation of `DDS-RPC` standard would be great, that's a lot of effort.
Support of an enhanced discovery mechanism to avoid the race condition should be enough to solve the problem.

richiware, 2020-06-12:
We've added your suggestion on our [roadmap](https://github.com/eProsima/Fast-DDS/pull/1255).

SteveMacenski, 2020-06-18:
To summarize issues from https://github.com/ros2/ros2/issues/931 with respect to navigation reported by @daisukes

Upgrading to 2.x.x from 1.x.x results in services failing to function (<5% successful calls), which is a safety issue with fast-dds.

dirk-thomas, 2020-06-18:
@richiware With the report mention in the previous comment I think this is a clear regression and should be addressed asap.

richiware, 2020-07-02:
@MiguelCompany could you help in this?

SteveMacenski, 2020-07-07:
@JaimeMartin can we get an update on this? This is a safety critical issue and we may have to switch primary support and instructions for Navigation2 to Cyclone if we can't reliably call services in Foxy. It could genuinely cause complete system failure.

MiguelCompany, 2020-07-08:
@SteveMacenski Looking at ros2/ros2#931 I see all the analysis was done before #390 was merged. I know it is a workaround and it has not made its way to a Foxy patch release (it is only in master), but have you checked if it improves the behavior?

Regarding [the commit](eProsima/Fast-DDS@a9bd1a9) that seems to be responsible, it is the merge commit of eProsima/Fast-DDS#1190. Among other things, that PR fixes a wrong behavior with volatile late-joiners. Before the PR, volatile late-joiners were (incorrectly) receiving old samples in some cases. Commits affecting volatile late-joiners are eProsima/Fast-DDS@a1c35b5bf2f991e381770f24c3131184efce4c39 and eProsima/Fast-DDS@30f8f6f4fff54c99244b01d3d304f0edd11a0aad and Fixing that to be compliant with RTPS standard, makes the service discovery race more likely to happen.

Nevertheless, we have just started analyzing ros-planning/navigation2#1772 and, at the same time, designing a proper service discovery on rmw_fastrtps.

SteveMacenski, 2020-07-08:
We also had to drop some unit tests as well because of this issue causing them to fail with Fast-RTPS only https://github.com/ros-planning/navigation2/pull/1856 and reports of various elements of navigation2 not working (recovery behaviors) due to services not working. 

Have you released this update so that it will be in ROS2 foxy after the sync Friday (https://discourse.ros.org/t/preparing-for-foxy-sync-and-patch-release-2020-07-10/15265)?

jacobperron, 2020-07-13:
> Have you released this update so that it will be in ROS2 foxy after the sync Friday (https://discourse.ros.org/t/preparing-for-foxy-sync-and-patch-release-2020-07-10/15265)?

The workaround in #390 dropped from my radar. I've added it to the [project board](https://github.com/orgs/ros2/projects/25) and make sure it gets into the next Foxy sync.

MiguelCompany, 2020-07-20:
@dirk-thomas @jacobperron As you can see [here](https://github.com/ros-planning/navigation2/issues/1772#issuecomment-658229941) the issue has been solved by eProsima/Fast-DDS#1295. I hope this can be released in foxy ASAP.

jacobperron, 2020-07-20:
@MiguelCompany With eProsima/Fast-DDS#1295, does it mean that the changes introduced in #390 are no longer required (or should #390 still be backported to Foxy)?

MiguelCompany, 2020-07-21:
> With [eProsima/Fast-DDS#1295](https://github.com/eProsima/Fast-DDS/pull/1295), does it mean that the changes introduced in #390 are no longer required (or should #390 still be backported to Foxy)?

@jacobperron No, the changes in #390 are still necessary, as there are still chances of asymmetric discovery of endpoints. Nevertheless, we are preparing a design for a different approach to how the service discovery race is handled. Although the new mechanism will be inspired by the DDS-RPC standard, it will not strictly follow it.

We will open a PR in a few days with the proposed design.

jacobperron, 2020-07-21:
@MiguelCompany If you can create a new release tag for the Fast-DDS `2.x.x` branch, I can help bloom a new release into rosdistro for Foxy.

dirk-thomas, 2020-07-21:
I would suggest to wait until eProsima/Fast-DDS#1304 has landed too (which should hopefully be soon - CI is running as we speak).

MiguelCompany, 2020-07-23:
@dirk-thomas @jacobperron The warnings related PRs have been merged, and I have created tag `2.0.1-rc`, which will most likely become release v2.0.1 after we proceed with release testing.

IkerLuengo, 2020-07-31:
We have done the design to address the problem with the service discovery. You can find it documented on PR #418

aditya2592, 2022-06-22:
Hi, I am facing several issues with service reliability when using ROS2 Galactic with Fast DDS over SHM. Also using discovery server. Most common occurrences:
1. Request is sent by client and response isn't received : This happens even with `RELIABLE` and `KEEP_ALL` QOS settings
2. Client fails to discover service during startup when using `wait_for_service`, even though the service has started up fine.

Is there anything on the ROS2 roadmap that will help address service reliability in the near term? We just switched to ROS2 and this issue is hurting our ability to build a stack around it.

MarcoMatteoBassa, 2022-06-24:
HI @aditya2592 , are you guys running into these issues while using docker or also on a normal installation?
We could reproduce the first issue (see https://github.com/MarcoMatteoBassa/reproduce_client_bug) but only while running on a docker container (humble version, will try on galactic too), so we are wondering if there's any docker-specific problem.

MarcoMatteoBassa, 2022-06-24:
@aditya2592 did you run into those issues even when running on cyclone-DDS (without the discovery server) on galactic?
For Humble apparently something is moving https://discourse.ros.org/t/nav2-issues-with-humble-binaries-due-to-fast-dds-rmw-regression/26128/2

fujitatomoya, 2022-06-24:
@MiguelCompany this could be related to https://github.com/ros2/rmw_fastrtps/pull/616?

aditya2592, 2022-06-24:
> @aditya2592 did you run into those issues even when running on cyclone-DDS (without the discovery server) on galactic? For Humble apparently something is moving https://discourse.ros.org/t/nav2-issues-with-humble-binaries-due-to-fast-dds-rmw-regression/26128/2

We had a lot more issues with cyclone so we stopped experimenting with that early on. There were memory leaks and general latency was not good with cyclone. But I have seen issues on both galactic and humble using Fast DDS with/without discovery server, with all services and clients on the same machine

aditya2592, 2022-07-05:
> @aditya2592 did you run into those issues even when running on cyclone-DDS (without the discovery server) on galactic? For Humble apparently something is moving https://discourse.ros.org/t/nav2-issues-with-humble-binaries-due-to-fast-dds-rmw-regression/26128/2

@MarcoMatteoBassa Were you able to run your experiment with the fix in this MR - https://github.com/ros2/rmw_fastrtps/pull/616 ?

MarcoMatteoBassa, 2022-07-07:
@fujitatomoya thanks for trying, JFI, here is the corresponding issue (opend on the docker images repo because I can only reproduce it on docker) https://github.com/osrf/docker_images/issues/628

aditya2592, 2022-07-07:
@MarcoMatteoBassa I incorporated the fix into my Docker image like below by only building `rmw_fastrtps` from source on the fixed branch. Rest of ROS2 is still from the existing docker image on Dockerhub. @clalancette This approach should work right?

```
ARG ROS_DISTRO=galactic
FROM ros:${ROS_DISTRO}-ros-base

WORKDIR /root/ws_ros
# RMW Fast DDS
# NOTE: this branch is used to incorporate fix for services
RUN git clone https://github.com/ros2/rmw_fastrtps/ -b mergify/bp/${ROS_DISTRO}/pr-616 src/rmw_fastrtps
RUN . /opt/ros/${ROS_DISTRO}/setup.sh && \
    rosdep update && \
    apt-get -q update && \
    DEBIAN_FRONTEND=noninteractive \
    # Install RMW Fast RTPS deps
    rosdep install -y --from-paths src/rmw_fastrtps --ignore-src --rosdistro ${ROS_DISTRO} --as-root=apt:false && \
    colcon build \
            --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=OFF -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
            --ament-cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo \
            --event-handlers desktop_notification- status- && \
    # Remove the source code from this container
    rm -rf src build && \
    ccache -s
```

Also another thing I am trying is to limit transports to only shared memory. This for sure improved service response getting dropped on one of our PCs but they still happen occasionally:

```
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
        <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <transport_descriptors>
            <!-- Create a descriptor for the new transport -->
            <transport_descriptor>
                <transport_id>shm_transport</transport_id>
                <type>SHM</type>
            </transport_descriptor>
        </transport_descriptors>

        <participant profile_name="super_client_profile" is_default_profile="true">
            <rtps>
                <!-- Link the Transport Layer to the Participant -->
                <userTransports>
                    <transport_id>shm_transport</transport_id>
                </userTransports>
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>
    </profiles>
</dds>

```

MarcoMatteoBassa, 2022-07-08:
@clalancette I verified that on osrf/ros2:nightly the installed rmw_fastrtps_shared_cpp/custom_client_info.hpp and rmw_fastrtps_shared_cpp/custom_service_info.hpp were modified as in the fix https://github.com/ros2/rmw_fastrtps/pull/616/files.
But sure, in the coming days I can also try it from source. @aditya2592  Thanks for the dockerfile!

MarcoMatteoBassa, 2022-07-18:
Hi @clalancette , just tested with a full src installation, but this is unfortunately still reproducible

aditya2592, 2022-07-18:
Update from my end on this, used the following config to completely eliminate any transport except SHM and also assigned a 75MB segment size. This based on experiments so far removed service drops on Humble. It also seems to allow discovery without using discovery server. Other things I tried that didn't work : 

- SHM only transport but no `segment_size` increase. This caused an enormous amount of drops.
- `export ROS_LOCALHOST_ONLY=1`

```
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <transport_descriptors>
            <!-- SHM transport should be the only one used for IPC on same machine -->
            <transport_descriptor>
                <transport_id>shm_transport</transport_id>
                <type>SHM</type>
                <!-- 75MB is assigned to each participant which means every process, one participant is equivalent to one process -->
                <!-- Multiply this by total processes in system to get overall SHM consumption -->
                <segment_size>78643200</segment_size>
                <maxMessageSize>78643200</maxMessageSize>
            </transport_descriptor>
        </transport_descriptors>

        <participant profile_name="super_client_profile" is_default_profile="true">
            <rtps>
                <!-- SHM only is enabled to enforce SHM use because currently all ROS2 comms happen on same machine -->
                <userTransports>
                    <transport_id>shm_transport</transport_id>
                </userTransports>
                <!-- Prevent builtin which automatically chooses transport -->
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>
    </profiles>
</dds>
```

Jayden-F, 2023-05-10:
https://github.com/ros2/rmw_fastrtps/issues/691

Is this planning to be fixed in Humble?