ros2/rmw_fastrtps #395: Add support to message lost event status

ivanpauno, 2020-06-19:
## Feature request

#### Feature description

Tracks the implementation of the feature described in https://github.com/ros2/rmw/issues/226 for `rmw_fastrtps_*`.

#### Implementation considerations

FastRTPS subscriber doesn't support the feature AFAIS (https://github.com/eProsima/Fast-DDS/blob/9125239274040384af8b1ad981817333fc4f9d58/include/fastrtps/subscriber/Subscriber.h#L161-L169).

FastDDS neither seems to have support https://github.com/eProsima/Fast-DDS/blob/9125239274040384af8b1ad981817333fc4f9d58/include/fastdds/dds/subscriber/DataReader.hpp#L281-L283.

ivanpauno, 2020-06-19:
@MiguelCompany FYI.
You might want to implement this in `rmw_fastrtps_*` when `FastRTPS` has support for the feature.

clalancette, 2022-01-28:
@MiguelCompany @JaimeMartin With Fast-DDS now being the default for Humble, support for this has become urgent.  In point of fact, we currently cannot start `rviz2` without this support, since rviz2 depends on the rclcpp `message_lost` callback working properly, and `rmw_fastrtps_cpp` throws an exception:

```
$ rviz2
[INFO] [1643405798.086552793] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1643405798.086719965] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[INFO] [1643405798.155550628] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [1643405798.303867112] [rviz2]: Could not load display config: Failed to initialize event: provided event_type is not supported by rmw_fastrtps_cpp, at /home/ubuntu/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_event.cpp:69
```

(I printed out the event type locally, and the one that failed was 3, which corresponds to https://github.com/ros2/rmw/blob/de38e79ca8906c8a7757557acdef57f8474351ff/rmw/include/rmw/event.h#L39)

One thing we could do here is to just not throw an error when we see that event type, but not really support it either.  That's not a great option, so I'm interested in hearing your thoughts about getting this developed in Fast-DDS itself and then in `rmw_fastrtps_cpp`.

EduPonz, 2022-02-01:
Hi everyone,

We have added this support to the roadmap for Fast DDS v2.6.0 (version for Humble) and we are currently evaluating the effort.

Thanks for bringing this to our attention!

MiguelCompany, 2022-02-18:
See #583 for the `rmw_fastrtps` part.

clalancette, 2022-09-07:
We fixed this with #583, so closing.

closed 2022-09-07 by clalancette