ros2/rmw_fastrtps #446: Timers being executed more than once by the multithread executor

ivanpauno, 2020-09-23:
## Bug report

**Required Info:**

- Operating System:
  -  Any
- Installation type:
  -  Any
- Version or commit hash:
  -  https://github.com/ros2/rclcpp/commit/2531787550084307a5c8144a68c0195fcdd6442a or previous commits (i.e. current master)
- DDS implementation:
  -  Any
- Client library (if applicable):
  -  rclcpp

#### Steps to reproduce issue

The test [here](https://github.com/ros2/rclcpp/blob/master/rclcpp/test/rclcpp/executors/test_multi_threaded_executor.cpp) sometimes fails.

e.g.: https://ci.ros2.org/view/nightly/job/nightly_osx_release/1712/testReport/projectroot/test/test_multi_threaded_executor/

#### Additional information

From comment [here](https://github.com/ros2/rclcpp/pull/1241#issuecomment-670530455), a timer can be triggered twice in a row when using the `MultiThreadedExecutor`:

- Threads A adds the timer to the wait set.
- The timer is ready to be executed by thread A.
- Thread B adds the timer to the wait set, and it's waiting it to be ready.
- Thread B sees that the timer is ready to be executed ([get_next_executable](https://github.com/ros2/rclcpp/blob/0e7fed993dcfa0eec28f094fb8518fc0877fb03d/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp#L83)).
- Thread A executes the timer and removes it from the scheduled timer collection.
- Thread B checks if the timer has been scheduled ... but thread A has already removed it from the collection. So thread B executes the timer again.

Related PRs:
- PR that introduced the "scheduled timers" mechanism: https://github.com/ros2/rclcpp/pull/383
- PR fixing hangs caused by the "scheduler timers" mechanism: https://github.com/ros2/rclcpp/pull/836
- PR trying to improve the "scheduled timers" mechanism to avoid spurious wake ups: https://github.com/ros2/rclcpp/pull/621
- PR (opened) to avoid threads being locked https://github.com/ros2/rclcpp/pull/1168 (almost reverting https://github.com/ros2/rclcpp/pull/621).

I guess this is complex :joy:.

fujitatomoya, 2020-09-29:
@ivanpauno 

I don't catch up with you the following sequence, could you enlighten me a bit?

> Thread B sees that the timer is ready to be executed (get_next_executable).
> Thread A executes the timer and removes it from the scheduled timer collection.
> Thread B checks if the timer has been [...]

with above sequence how `Thread A` can be on running to access `scheduled_timers_` in the middle of `Thread B`? `Thread B` is protected with `wait_mutex_` https://github.com/ros2/rclcpp/blob/0e7fed993dcfa0eec28f094fb8518fc0877fb03d/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp#L78-L98. so is `scheduled_timers_` https://github.com/ros2/rclcpp/blob/0e7fed993dcfa0eec28f094fb8518fc0877fb03d/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp#L106-L109.

I think that it should be the following sequence,

> Thread B sees that the timer is ready to be executed (get_next_executable).
> Thread A executes the timer (but not removes it from the scheduled timer collection.)
> Thread B checks if the timer has been scheduled, and see it is scheduled the continue.
> Thread A removes it from the scheduled timer collection.

ivanpauno, 2020-09-29:
> with above sequence how Thread A can be on running to access scheduled_timers_ in the middle of Thread B? Thread B is protected with wait_mutex_

Yeah, I was writing the temporal sequence in the code and I think you're right.

> I think that it should be the following sequence,

The sequence you comment can make the timer miss a trigger, but cannot produce double execution AFAIS.

---

I have to double check my reasoning here.

ivanpauno, 2020-10-05:
Wrong repository :confused:, moving to `rclcpp`.