ros2/rmw_fastrtps #454: occasional long message delay in remote node subscription of pointcloud

Jconn, 2020-09-29:
Bug report
Required Info:

Operating System:
Ubuntu 18.04
Installation type:
foxy - from source
Version or commit hash:
f54c74b80981

I have a 96x128 pointcloud where each point is 16 bytes. I am publishing this with Sensor QoS at 15 hz using foxy, built from source last week.

When I am on the same computer as the publisher, and I check the message rate with:
`ros2 topic hz /camera/pointcloud
`
I receive messages at 15 hz.

I have another computer on the same ethernet network as this publisher, also ubuntu 18.04, with foxy built from source. When I check the subscription rate on this second computer using the same `ros2 topic hz /camera/pointcloud`, I am still hovering at 15hz reception. But every few minutes there is a ~10 second period where no pointclouds are received on the second computer.

clalancette, 2020-10-15:
@MiguelCompany @richiprosima Any thoughts about this one?

MiguelCompany, 2020-10-15:
Using Sensor QoS means using best-effort reliability. Each pointcloud will have 96x128x16 bytes, which means around 200 KBytes. This will require fragmentation and, with best-effort that means that if a fragment gets lost, the whole sample would be lost.

Now, how could the fragment get lost?
1. It may be the case that the network drops it, but that is unlikely happening, unless you are on a lossy network. 
2. The reception buffer is overrun. That doesn't necessarily mean that the reader is slow taking data, it may be that the network is sending data in bursts. You could avoid this either by increasing the reception buffer.
3. The sending buffer is overrun. That could happen if the asynchronous sending thread doesn't get scheduled for whatever the reason. You could avoid this by changing the publishing mode to synchronous, or by adding a throughput controller on the publishing side.

I would first try to change the publishing mode following the instructions [here](https://github.com/ros2/rmw_fastrtps#example).

Changing the buffer sizes would require to add a `<participant>` profile. Check [here](https://fast-dds.docs.eprosima.com/en/v2.0.1/fastdds/xml_configuration/domainparticipant.html#rtps-element-type) for `listenSocketBufferSize` and `sendSocketBufferSize`

Adding a throughput controller can be done on either the `publisher` profile or the `publisher`. More information [here](https://fast-dds.docs.eprosima.com/en/v2.0.1/fastdds/xml_configuration/common.html#throughput-configuration)

clalancette, 2022-02-03:
No response from the original submitter in a long time, so closing this out.  Feel free to reopen if you have more information to share.

closed 2022-02-03 by clalancette