ros2/rmw_fastrtps #461: Unccaught exception causing test failures in macOS

nuclearsandwich, 2020-10-09:
## Bug report

**Required Info:**

- Operating System:
  - macOS Mojave
- Installation type:
  - source
- Version or commit hash:
  - master as of 23f9c4c09d499c6d1cd5cdeb127f823bbf37520c or possibly earlier
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

It doesn't seem to be reliably reproducible. Repeated jobs have it in at least one test over the last three nights which is the most reliability I've seen yet.
* https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2116/testReport/(root)/rcl/test_node__rmw_fastrtps_cpp_gtest_missing_result/
* https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2117/testReport/(root)/rcl/test_node__rmw_fastrtps_cpp_gtest_missing_result/

* https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2118/testReport/(root)/rcl/test_node__rmw_fastrtps_cpp_gtest_missing_result/

#### Additional information

The earliest build I've found with the error is https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2109//rcl/test_node__rmw_fastrtps_cpp_gtest_missing_result/ but even in the repeated job it has come and gone at least once between then and now.

Additional examples are https://ci.ros2.org/view/nightly/job/nightly_osx_release/1834/testReport/(root)/rcl/test_node__rmw_fastrtps_cpp_gtest_missing_result/ and https://ci.ros2.org/view/nightly/job/nightly_osx_extra_rmw_release/953/testReport/(root)/test_rmw_implementation/test_create_destroy_node__rmw_fastrtps_dynamic_cpp_gtest_missing_result/

nuclearsandwich, 2020-10-09:
There was one new instance of this in today's nightlies:

https://ci.ros2.org/view/nightly/job/nightly_osx_extra_rmw_release/lastCompletedBuild/testReport/projectroot/test/test_node__rmw_fastrtps_cpp/

mjcarroll, 2020-10-16:
Another instance of this, but this time in `test_create_destroy_node__rmw_fastrpts_cpp`: https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2126/testReport/junit/(root)/projectroot/test_create_destroy_node__rmw_fastrtps_cpp/

chapulina, 2020-11-05:
I'd like to point out that this test has been failing consistently (20/24 past builds):

https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/lastBuild/testReport/projectroot/test/test_node__rmw_fastrtps_cpp/history/

If it can't be fixed, I suggest marking it as `xfail` and leaving this issue open until it can be addressed.

hidmic, 2020-12-22:
I believe this is a duplicate of #491, which since #496 has been fixed.

clalancette, 2020-12-22:
The last time this failed was December 16, which is the day that #496 was merged.  So I'm going to go ahead and say that this is fixed, and close it out.  Do feel free to reopen if this occurs again.

closed 2020-12-22 by clalancette