ros2/rmw_fastrtps #464: Example xml does not change the qos of the simple ROS2 publisher

SomaGallai, 2020-10-13:
## Bug report

**Required Info:**

- Operating System:
  - OS X Catalina 10.15.5
- Installation type:
  - Installed as the default middleware for ROS2
- Version or commit hash:
  - foxy version
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export FASTRTPS_DEFAULT_PROFILES_FILE=path/to/xml/ros_example.xml
export RMW_FASTRTPS_USE_QOS_FROM_XML=1
ros2 run demo_nodes_cpp talker

#### Expected behavior
No publishing until listener is started as the example xml is set to synchronous.
#### Actual behavior
Starts publishing as soon as it is started

JLBuenoLopez-eProsima, 2020-10-13:
Hi @SomaGallai,

There seems to be a misunderstanding related with the synchronous publishing mode. This mode does not wait for a matched subscriber before starting to publish as you imply in your bug ticket. [Fast DDS documentation](https://fast-dds.docs.eprosima.com/en/latest/fastdds/dds_layer/core/policy/eprosimaExtensions.html#publishmodeqospolicy) explains that data will be publish every time the user thread calls the write operation. Consequently, if you wish that your publisher waits until there is a matched subscriber, you should not call the write operation until this condition is fulfilled. This logic should be implemented in your application's code (in this case modifying the demo_nodes_cpp example) as there is no QoS that defines the behavior that you wish.

SomaGallai, 2020-10-13:
Hi @JLBuenoLopez-eProsima 

Thank you for clearing up my confusion!

closed 2020-10-13 by SomaGallai