ros2/rmw_fastrtps #491: Possible regression on rmw_fastrps on OS X

caguero, 2020-12-03:
## Bug report

**Required Info:**

- Operating System:
  OS X

#### Steps to reproduce issue
See https://ci.ros2.org/view/nightly/job/nightly_osx_release/1870/#showFailuresLink

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

hidmic, 2020-12-14:
@jacobperron I'll take a look.

hidmic, 2020-12-14:
This one's tough. It's coming from within Fast-DDS. And it doesn't happen consistently.

EduPonz, 2020-12-15:
Hi @hidmic 

Do you know how/where is it coming from Fast DDS? Do you have some traces or dumps that we can take a look at to see if there's something to be done in our side?

hidmic, 2020-12-15:
Turns out I spoke too soon. It was aborting within a Fast-DDS object destructor:
 
<details>
<summary> Backtrace on SIGABRT </summary>
```
 thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff6931b2c2 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff693d6bf1 libsystem_pthread.dylib`pthread_kill + 284
    frame #2: 0x00007fff692856a6 libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff66460641 libc++abi.dylib`abort_message + 231
    frame #4: 0x00007fff664607c7 libc++abi.dylib`default_terminate_handler() + 243
    frame #5: 0x00007fff67a13eeb libobjc.A.dylib`_objc_terminate() + 105
    frame #6: 0x00007fff6646c19e libc++abi.dylib`std::__terminate(void (*)()) + 8
    frame #7: 0x00007fff6646c213 libc++abi.dylib`std::terminate() + 51
    frame #8: 0x00000001013a9caf libfastrtps.2.dylib`__clang_call_terminate + 15
    frame #9: 0x0000000101537e7d libfastrtps.2.dylib`eprosima::fastrtps::SubscriberImpl::~SubscriberImpl(this=0x000000010700fe00) at SubscriberImpl.cpp:93:1
    frame #10: 0x0000000101537e95 libfastrtps.2.dylib`eprosima::fastrtps::SubscriberImpl::~SubscriberImpl(this=0x000000010700fe00) at SubscriberImpl.cpp:82:1
    frame #11: 0x0000000101537eb9 libfastrtps.2.dylib`eprosima::fastrtps::SubscriberImpl::~SubscriberImpl(this=0x000000010700fe00) at SubscriberImpl.cpp:82:1
    frame #12: 0x000000010150345b libfastrtps.2.dylib`eprosima::fastrtps::ParticipantImpl::removeSubscriber(this=0x00000001006085b0, sub=0x0000000100601cc0) at ParticipantImpl.cpp:108:13
    frame #13: 0x00000001014edcc1 libfastrtps.2.dylib`eprosima::fastrtps::Domain::removeSubscriber(sub=0x0000000100601cc0) at Domain.cpp:139:36
    frame #14: 0x0000000101226485 librmw_fastrtps_shared_cpp.dylib`rmw_fastrtps_shared_cpp::destroy_subscription(identifier="rmw_fastrtps_cpp", participant_info=0x00000001006003a0, subscription=0x0000000104600d40) at subscription.cpp:52:8
    frame #15: 0x000000010101776d librmw_fastrtps_cpp.dylib`init_context_impl(this=0x00007ffeefbfac58, sub=0x0000000104600d40)::$_2::operator()(rmw_subscription_t*) const at init_rmw_context_impl.cpp:125:25
    frame #16: 0x0000000101017712 librmw_fastrtps_cpp.dylib`decltype(__f=0x00007ffeefbfac58, __args=0x00007ffeefbfa8e0)::$_2&>(fp)(std::__1::forward<rmw_subscription_t*>(fp0))) std::__1::__invoke<init_context_impl(rmw_context_t*)::$_2&, rmw_subscription_t*>(init_context_impl(rmw_context_t*)::$_2&, rmw_subscription_t*&&) at type_traits:4339:1
    frame #17: 0x00000001010176a2 librmw_fastrtps_cpp.dylib`void std::__1::__invoke_void_return_wrapper<void>::__call<init_context_impl(__args=0x00007ffeefbfac58, __args=0x00007ffeefbfa8e0)::$_2&, rmw_subscription_t*>(init_context_impl(rmw_context_t*)::$_2&, rmw_subscription_t*&&) at __functional_base:349:9
    frame #18: 0x00000001010164d9 librmw_fastrtps_cpp.dylib`std::__1::__function::__func<init_context_impl(rmw_context_t*)::$_2, std::__1::allocator<init_context_impl(rmw_context_t*)::$_2>, void (rmw_subscription_t*)>::operator(this=0x00007ffeefbfac50, __arg=0x00007ffeefbfa8e0)(rmw_subscription_t*&&) at functional:1562:12
    frame #19: 0x000000010101b665 librmw_fastrtps_cpp.dylib`std::__1::function<void (rmw_subscription_t*)>::operator(this= Lambda in File init_rmw_context_impl.cpp at Line 124, __arg=0x0000000104600d40)(rmw_subscription_t*) const at functional:1913:12
    frame #20: 0x000000010101b5af librmw_fastrtps_cpp.dylib`std::__1::unique_ptr<rmw_subscription_t, std::__1::function<void (rmw_subscription_t*)> >::reset(this=0x00007ffeefbfac40, __p=0x0000000000000000) at memory:2648:7
    frame #21: 0x000000010101b53d librmw_fastrtps_cpp.dylib`std::__1::unique_ptr<rmw_subscription_t, std::__1::function<void (rmw_subscription_t*)> >::~unique_ptr(this=0x00007ffeefbfac40) at memory:2602:19
    frame #22: 0x0000000101011f75 librmw_fastrtps_cpp.dylib`std::__1::unique_ptr<rmw_subscription_t, std::__1::function<void (rmw_subscription_t*)> >::~unique_ptr(this=0x00007ffeefbfac40) at memory:2602:17
    frame #23: 0x0000000101011887 librmw_fastrtps_cpp.dylib`init_context_impl(context=0x0000000100503d78) at init_rmw_context_impl.cpp:184:1
    frame #24: 0x0000000101010fa5 librmw_fastrtps_cpp.dylib`rmw_fastrtps_cpp::increment_context_impl_ref_count(context=0x0000000100503d78) at init_rmw_context_impl.cpp:195:21
    frame #25: 0x000000010103d622 librmw_fastrtps_cpp.dylib`::rmw_create_node(context=0x0000000100503d78, name="my_node", namespace_="/my_ns") at rmw_node.cpp:60:21
    frame #26: 0x000000010025c97d librmw_implementation.dylib`::rmw_create_node(v3=0x0000000100503d78, v2="my_node", v1="/my_ns") at functions.cpp:222:1
    frame #27: 0x0000000100005f3f test_create_destroy_node__rmw_fastrtps_cpp`TestNodeConstructionDestruction__rmw_fastrtps_cpp_create_with_internal_errors_Test::TestBody(this=0x0000000100503d00) at test_create_destroy_node.cpp:184:3
    frame #28: 0x0000000100050dbe test_create_destroy_node__rmw_fastrtps_cpp`void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(object=0x0000000100503d00, method=21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00, location="the test body")(), char const*) at gtest.cc:2433:10
    frame #29: 0x000000010001adcb test_create_destroy_node__rmw_fastrtps_cpp`void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(object=0x0000000100503d00, method=21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00, location="the test body")(), char const*) at gtest.cc:2469:14
[... cut ...]
    frame #49: 0x0000000100029500 test_create_destroy_node__rmw_fastrtps_cpp`testing::UnitTest::Run(this=0x000000010006bfc8) at gtest.cc:4925:10
    frame #50: 0x000000010000cf71 test_create_destroy_node__rmw_fastrtps_cpp`RUN_ALL_TESTS() at gtest.h:2473:46
    frame #51: 0x000000010000cf50 test_create_destroy_node__rmw_fastrtps_cpp`main(argc=1, argv=0x00007ffeefbfb908) at gtest_main.cc:45:10
    frame #52: 0x00007fff691e03d5 libdyld.dylib`start + 1
    frame #53: 0x00007fff691e03d5 libdyld.dylib`start + 1
```
</details>

I think #496 should fix the issue.

hidmic, 2020-12-21:
Hmm, https://ci.ros2.org/view/nightly/job/nightly_osx_extra_rmw_release/lastCompletedBuild/testReport/(root)/projectroot/test_create_destroy_node__rmw_fastrtps_dynamic_cpp/ I still see it. New error this time.

hidmic, 2020-12-22:
Issue above is a completely different beast. Closing this and opening another ticket.

PR #496 (2020-12-16): Set rmw_dds_common::GraphCache callback after init succeeds.
Otherwise, during stack unwinding, the graph cache may end up with a callback that captures and uses an already destroyed guard condition.

This patch (hopefully) fixes #491. 

Ci up `test_rmw_implementation`:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]
closed 2020-12-22 by hidmic