ros2/rmw_fastrtps #498: rmw_fastrtps_dynamic_cpp node create/destruct tests segfault on MacOS

hidmic, 2020-12-22:
## Bug report

**Required Info:**

- Operating System:
  - MacOS Mojave
- Installation type:
  - From source
- Version or commit hash:
  - [`master`](4d38e385b8474cd604ff4f491d058f14b3a49a5d)

#### Steps to reproduce issue

Run node creation/destruction tests for `rmw_fastrtps_dynamic_cpp` in MacOS: 

```sh
colcon test --event-handlers=console_direct+ --packages-select test_rmw_implementation --ctest-args ' -R' ' test_create_destroy_node__rmw_fastrtps_dynamic_cpp'
```

#### Expected behavior
Test passes consistently.

#### Actual behavior
Test segfaults every now and then **on exit**:

```
[----------] Global test environment tear-down
[==========] 6 tests from 2 test suites ran. (2246 ms total)
[  PASSED  ] 6 tests.
[foonathan::memory] Allocator foonathan::memory::new_allocator (at 0x0) leaked 288 bytes.
[foonathan::memory] Allocator foonathan::memory::heap_allocator (at 0x0) leaked 34056 bytes.
Process 99589 stopped
* thread #5, stop reason = EXC_BAD_ACCESS (code=1, address=0x102447e0a)
    frame #0: 0x0000000102447e0a
error: memory read failed for 0x102447e00
Target 0: (test_create_destroy_node__rmw_fastrtps_dynamic_cpp) stopped.
```

#### Additional information
This does not occur with `rmw_fastrps_cpp`. Basic triaging suggests a thread may be trying to access `rmw` code or state past library unloading:

```
(lldb) thread list
Process 99589 stopped
* thread #1: tid = 0xcae860c, 0x00000001001cb656 dyld`__munmap + 10, queue = 'com.apple.main-thread'
  thread #2: tid = 0xcae86e8, 0x00007fff69318866 libsystem_kernel.dylib`__psynch_cvwait + 10
  thread #3: tid = 0xcae86e9, 0x00007fff6931a786 libsystem_kernel.dylib`__recvmsg + 10
  thread #4: tid = 0xcae86ea, 0x00007fff6931a786 libsystem_kernel.dylib`__recvmsg + 10
  thread #5: tid = 0xcae86eb, 0x0000000102447e0a, stop reason = EXC_BAD_ACCESS (code=1, address=0x102447e0a)
  thread #6: tid = 0xcae86ec, 0x00007fff6931a786 libsystem_kernel.dylib`__recvmsg + 10
  thread #7: tid = 0xcae86ed, 0x00007fff69318f2e libsystem_kernel.dylib`__semwait_signal + 10
  thread #8: tid = 0xcae86ef, 0x00007fff69318866 libsystem_kernel.dylib`__psynch_cvwait + 10
(lldb) bt
* thread #1, queue = 'com.apple.main-thread'
  * frame #0: 0x00000001001cb656 dyld`__munmap + 10
    frame #1: 0x00000001001a2cf9 dyld`ImageLoaderMachOCompressed::~ImageLoaderMachOCompressed() + 25
    frame #2: 0x000000010018bbb8 dyld`dyld::garbageCollectImages() + 857
    frame #3: 0x0000000100195623 dyld`dlclose + 191
    frame #4: 0x00007fff691cc6d2 libdyld.dylib`dlclose + 157
    frame #5: 0x000000010041b9c0 librcutils.dylib`rcutils_unload_shared_library(lib=0x0000000100601220) at shared_library.c:163:20
    frame #6: 0x00000001003f3185 librcpputils.dylib`rcpputils::SharedLibrary::~SharedLibrary(this=0x0000000100601218) at shared_library.cpp:45:25
    frame #7: 0x00000001003f3335 librcpputils.dylib`rcpputils::SharedLibrary::~SharedLibrary(this=0x0000000100601218) at shared_library.cpp:43:1
    frame #8: 0x0000000100260589 librmw_implementation.dylib`std::__1::__shared_ptr_emplace<rcpputils::SharedLibrary, std::__1::allocator<rcpputils::SharedLibrary> >::__on_zero_shared(this=0x0000000100601200) at memory:3706:23
    frame #9: 0x000000010025f3ba librmw_implementation.dylib`std::__1::__shared_count::__release_shared(this=0x0000000100601200) at memory:3540:9
    frame #10: 0x000000010025f35f librmw_implementation.dylib`std::__1::__shared_weak_count::__release_shared(this=0x0000000100601200) at memory:3582:27
    frame #11: 0x000000010025f32c librmw_implementation.dylib`std::__1::shared_ptr<rcpputils::SharedLibrary>::~shared_ptr(this=0x00000001002621b8) at memory:4518:19
    frame #12: 0x000000010025b5a5 librmw_implementation.dylib`std::__1::shared_ptr<rcpputils::SharedLibrary>::~shared_ptr(this=0x00000001002621b8) at memory:4516:1
    frame #13: 0x00007fff692863cf libsystem_c.dylib`__cxa_finalize_ranges + 319
    frame #14: 0x00007fff692866b3 libsystem_c.dylib`exit + 55
    frame #15: 0x00007fff691e03dc libdyld.dylib`start + 8
```

hidmic, 2020-12-22:
@EduPonz @MiguelCompany any chance you can take a look at this at some point?

EduPonz, 2020-12-23:
Hi @hidmic ,

We'll take a look at it in January. Thanks for the heads up!

clalancette, 2021-01-07:
@EduPonz Friendly reminder to take a look at this one when you have a chance.  Thanks.

rsanchez15, 2021-01-13:
Hi @hidmic 

In order to reproduce the issue I need more information about the development environment. In particular:
- Could you post the exact state of all ROS 2 packages? You can do this by navigating to the root folder of your ROS 2 workspace and running the following command:
```bash 
vcs export --exact src > issue_498.repos
```
- It would also be helpful to know which test of the test suite belonging to `test_create_destroy_node` is failing.

hidmic, 2021-03-09:
Circling back. 

> Could you post the exact state of all ROS 2 packages? You can do this by navigating to the root folder of your ROS 2 workspace and running the following command:

@rsanchez15 .repos file can be found [here](https://gist.github.com/hidmic/da4d018e7f48dabeea860f303f877ab0).

> It would also be helpful to know which test of the test suite belonging to test_create_destroy_node is failing.

@rsanchez15 It doesn't manifest for any specific test. It happens **after** all have run, on process termination.

hidmic, 2021-03-19:
@rsanchez15 @EduPonz friendly ping

hidmic, 2021-04-07:
@rsanchez15 @EduPonz @MiguelCompany periodic ping

MiguelCompany, 2021-04-08:
@hidmic We have been quite busy getting Fast DDS 2.3 out and performing the refactor of the rmw to our DDS API. I finally have time to check this today, will keep you informed.

MiguelCompany, 2021-04-08:
@hidmic Couldn't reproduce here. These are the steps I have taken:

* Followed instructions on https://docs.ros.org/en/rolling/Installation/macOS-Development-Setup.html
* `colcon build --symlink-install --base-paths "src" --build-base "build" --install-base "install" --event-handlers console_cohesion+ console_package_list+ --packages-up-to test_rmw_implementation --cmake-args -DBUILD_TESTING=ON --no-warn-unused-cli -DCMAKE_BUILD_TYPE=Release -DINSTALL_EXAMPLES=OFF -DSECURITY=ON`
    * Note: I had to `python3 -m pip install numpy` in order to succeed
* `colcon test --event-handlers=console_direct+ --packages-select test_rmw_implementation --ctest-args ' -R' ' test_create_destroy_node__rmw_fastrtps_dynamic_cpp' --repeat-until-fail 1000`

On a side note, this was checked on the only mac we have in the office, which is the one running our CI. I did an `export ROS_DOMAIN_ID=169` to avoid collisions

hidmic, 2021-04-08:
@MiguelCompany thank you for checking ! 

> Couldn't reproduce here.

That's unexpected :thinking: Perhaps some of the recent changes to `rmw_fastrtps_cpp` indirectly fixed it? What if you don't set a `CMAKE_BUILD_TYPE ` (and just use the default build configuration)?

MiguelCompany, 2021-04-09:
> Perhaps some of the recent changes to `rmw_fastrtps_cpp` indirectly fixed it?

Indeed. I think [these lines](https://github.com/ros2/rmw_fastrtps/pull/518/files#diff-d8e97ec75e34b3fa50933914e85806c8edc034a5fa37fffd2df82fd3b3885438R332-R334) may be responsible of the fix

> What if you don't set a `CMAKE_BUILD_TYPE ` (and just use the default build configuration)?

Will do, but I just copied the cmake args from the console output of https://ci.ros2.org/view/nightly/job/nightly_osx_release/2022

MiguelCompany, 2021-04-09:
@hidmic Same result if I don't set `CMAKE_BUILD_TYPE`

hidmic, 2021-04-09:
Let's run MacOS CI once just to be completely sure: [image]

hidmic, 2021-04-09:
Alright, no longer an issue. Thanks for checking @MiguelCompany !

closed 2021-04-09 by hidmic