ros2/rmw_fastrtps #507: Occasionnally crash in fastrtps

BaiCai1990, 2021-01-29:
## Bug report

**Required Info:**

- Operating System:
  Debian 10 buster
- Installation type:
  from source
- Version or commit hash:
  - 
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue

```

```

backtrace:
#0  0x00007ffff2159704 in eprosima::fastrtps::rtps::StatefulReader::send_acknack(eprosima::fastrtps::rtps::WriterProxy const*, eprosima::fastrtps::rtps::RTPSMessageSenderInterface const&, bool) () from /opt/ros2_foxy/install/fastrtps/lib/libfastrtps.so.2
#1  0x00007ffff2156c7c in std::_Function_handler<bool (), eprosima::fastrtps::rtps::WriterProxy::WriterProxy(eprosima::fastrtps::rtps::StatefulReader*, eprosima::fastrtps::rtps::RemoteLocatorsAllocationAttributes const&, eprosima::fastrtps::ResourceLimitedContainerConfig const&)::{lambda()#1}>::_M_invoke(std::_Any_data const&) () from /opt/ros2_foxy/install/fastrtps/lib/libfastrtps.so.2
#2  0x00007ffff212a080 in eprosima::fastrtps::rtps::TimedEventImpl::trigger(std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >) () from /opt/ros2_foxy/install/fastrtps/lib/libfastrtps.so.2
#3  0x00007ffff2128bbc in eprosima::fastrtps::rtps::ResourceEvent::do_timer_actions() ()
   from /opt/ros2_foxy/install/fastrtps/lib/libfastrtps.so.2
#4  0x00007ffff21290bb in eprosima::fastrtps::rtps::ResourceEvent::event_service() ()
   from /opt/ros2_foxy/install/fastrtps/lib/libfastrtps.so.2
#5  0x00007ffff67c5b2f in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7e6ffa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
#7  0x00007ffff64a34cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

#### Expected behavior
run normal
#### Actual behavior
occasionally crash.
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2021-02-01:
@BaiCai1990 Can you provide a reproducing example here?  That is, how do you get into this situation?

@MiguelCompany @EduPonz This seems to be somewhere in the Fast-DDS stack.  Could you take a look?

BaiCai1990, 2021-02-04:
@clalancette I switched to **cyclonedds**， the probability of problems is greatly reduced.

BaiCai1990, 2021-02-04:
The key code I have provide @clalancette

hidmic, 2021-02-08:
@BaiCai1990 I believe @clalancette is asking for an example we can actually compile and execute to reproduce. It's quite difficult to track down the issue otherwise.

hidmic, 2021-04-08:
@BaiCai1990 friendly ping

BaiCai1990, 2021-04-11:
@hidmic The complete code i can not provide, Because it involves company secrets。But i will provide a simplified version recently, thinks for your attention.

clalancette, 2022-09-07:
Since we cannot reproduce this and there is no example we can run, I'm going to close this out.  Please feel free to reopen if you are still having problems and can provide an example showing the problem.

closed 2022-09-07 by clalancette