ros2/rmw_fastrtps #509: Restarting nodes causes other nodes to "disappear"

v-lopez, 2021-02-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - [foxy-pkgs.txt](https://github.com/ros2/rmw_fastrtps/files/5970338/foxy-pkgs.txt)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Terminal 1: `ros2 run  examples_rclcpp_minimal_publisher publisher_member_function`
Terminal 2: `ros2 run  examples_rclcpp_minimal_subscriber subscriber_member_function`
Terminal 3: `ros2 launch nav2_bringup slam_launch.py`

Kill the `slam_launch.py` on Terminal 3 and restart it, repeat this several times. Eventually /clock will disappear.
`ros2 node list` will not list gazebo node (neither with `ros2 node list --no-daemon`).

#### Expected behavior
That clock keeps being published and all the nodes are visible.

#### Actual behavior
Nodes and topics are no longer findable through ros2 cli.
Also a rclcpp subscriber to the same topic will stop receiving messages, so it seems not related to ros2 cli.

#### Additional information
It can also be reproduced with gazebo (how I found out initially), for instance running:
Terminal 1: `ros2 launch  gazebo_ros gazebo.launch.py `
Terminal 2: `ros2 topic echo /clock`

It can also be reproduced with ros2_control, running:
Terminal 1: `ros2 launch ros2_control_demo_robot rrbot_system_position_only.launch.py`
Terminal 2: `ros2 control load_start_controller joint_state_controller`
Terminal 3: `ros2 topic echo /joint_states`

Also instead of restarting `ros2 launch nav2_bringup slam_launch.py` it can also be reproduced restarting `ros2 launch nav2_bringup rviz_launch.py`

It does NOT happen with `RMW_IMPLEMENTATION=rmw_connext_cpp`

I don't know if it has some relation with nav2 packages or https://github.com/ros2/rmw_fastrtps/issues/341

I see no crashes, no weird messages, just the nodes silently stop publishing (or receiving) and are not findable. 
I also attached with the debugger and saw that the publish function keeps being called.

hidmic, 2021-02-12:
@EduPonz @MiguelCompany thoughts?

fujitatomoya, 2021-03-05:
https://github.com/ros2/rmw_fastrtps/pull/514 could be related for localhost communication?

IkerLuengo, 2021-03-15:
Thank you @v-lopez for the detailed description on how to reproduce it. It was really helpful to find the problem. We found this to be a failure in the transport, that stops sending the data from the publisher. We are looking for the root cause, and some patch should be available shortly.

hidmic, 2021-03-26:
Any updates @IkerLuengo ?

IkerLuengo, 2021-03-29:
Solution is implemented on [this PR](https://github.com/eProsima/Fast-DDS/pull/1856). It is currently targeted to master, but will be backported to all relevant versions in maintenance.

IkerLuengo, 2021-04-08:
[PR #1856](https://github.com/eProsima/Fast-DDS/pull/1856) has been merged into master

Backports to affected branches under maintenance have been created:
- [PR @1893](https://github.com/eProsima/Fast-DDS/pull/1893) for 2.2.x
- [PR @1894](https://github.com/eProsima/Fast-DDS/pull/1894) for 2.1.x
- [PR @1895](https://github.com/eProsima/Fast-DDS/pull/1895) for 2.0.x

MiguelCompany, 2021-05-31:
@v-lopez Could you confirm if this has been solved by the PRs mentioned in the previous comment

clalancette, 2022-02-03:
No response from the original reporter in over 6 months, so closing.  Feel free to keep commenting or reopen if you are still having an issue.

closed 2022-02-03 by clalancette