ros2/rmw_fastrtps #510: FastDDS to allow devices to communicate across different network in Gazebo with RMF

destkk, 2021-02-16:
Operating System: Ubuntu 20.04
DDS: FastDDS

Hi there. I would like to check how should we set FastDDS to allow devices to communicate across different network? I am trying to allow a server to send a service command to another server with Gazebo installed with RMF (rmf_demos) across different network.

I have created a default_fastrtps_profiles.xml with the following code below to include both separate IPV4 addresses,
```
<?xml version="1.0" encoding="UTF-8" ?>

<profiles>
    <transport_descriptors>
        <transport_descriptor>
            <transport_id>veelpeers</transport_id> <!-- string -->
            <type>UDPv4</type> <!-- string -->
            <maxInitialPeersRange>100</maxInitialPeersRange> <!-- uint32 -->
        </transport_descriptor>
    </transport_descriptors>
    <participant profile_name="participant_somename" is_default_profile="true">
        <rtps>
            <builtin>
                <metatrafficUnicastLocatorList> 
                    <locator/> 
                </metatrafficUnicastLocatorList> 
                <domainId>4</domainId>
                <initialPeersList>
                    <locator>
                        <udpv4>
                                <address>192.168.1.18</address>
                        </udpv4>
                    </locator>
                    <locator>
                        <udpv4>
                                <address>192.168.1.7</address>
                        </udpv4>
                    </locator>
                </initialPeersList>
            </builtin>
            <userTransports>
            <transport_id>veelpeers</transport_id>
            </userTransports>
            <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
    </participant>
</profiles>
```

In the env file i have included `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp `and `FASTRTPS_DEFAULT_PROFILES_FILE=file:///home/brobotics00/Desktop/default_fastrtps_profiles.xml`

After launching Gazebo I have received the following error below which shows an error opening the file,
```
[gzclient-14] 2021-02-16 10:02:01.688 [XMLPARSER Error] Error opening 'file:///home/$USER/Desktop/default_fastrtps_profiles.xml' -> Function loadXML
[gzclient-14] 2021-02-16 10:02:01.688 [XMLPARSER Error] Error parsing 'file:///home/$USER/Desktop/default_fastrtps_profiles.xml' -> Function loadXMLFile
```
Does anyone have tried launching gazebo with RMF and receiving commands from another server across different network?

hidmic, 2021-02-16:
@destkk skimming through https://github.com/eProsima/Fast-DDS/, it seems that the issue here is that `<domainId/>` should be under `<participant/>` not `<builtin/>`, i.e.:

```
<?xml version="1.0" encoding="UTF-8" ?>

<profiles>
    <transport_descriptors>
        <transport_descriptor>
            <transport_id>veelpeers</transport_id> <!-- string -->
            <type>UDPv4</type> <!-- string -->
            <maxInitialPeersRange>100</maxInitialPeersRange> <!-- uint32 -->
        </transport_descriptor>
    </transport_descriptors>
    <participant profile_name="participant_somename" is_default_profile="true">  
        <domainId>4</domainId>        
        <rtps>
            <builtin>
                <metatrafficUnicastLocatorList> 
                    <locator/> 
                </metatrafficUnicastLocatorList> 
                
                <initialPeersList>
                    <locator>
                        <udpv4>
                                <address>192.168.1.18</address>
                        </udpv4>
                    </locator>
                    <locator>
                        <udpv4>
                                <address>192.168.1.7</address>
                        </udpv4>
                    </locator>
                </initialPeersList>
            </builtin>
            <userTransports>
            <transport_id>veelpeers</transport_id>
            </userTransports>
            <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
    </participant>
</profiles>
```

CC @EduPonz to confirm.

destkk, 2021-02-19:
> @destkk skimming through https://github.com/eProsima/Fast-DDS/, it seems that the issue here is that `<domainId/>` should be under `<participant/>`, not `<builtin/>`. That is:
> 
> ```
> <?xml version="1.0" encoding="UTF-8" ?>
> 
> <profiles>
>     <transport_descriptors>
>         <transport_descriptor>
>             <transport_id>veelpeers</transport_id> 
>             <type>UDPv4</type> 
>             <maxInitialPeersRange>100</maxInitialPeersRange> 
>         </transport_descriptor>
>     </transport_descriptors>
>     <participant profile_name="participant_somename" is_default_profile="true">  
>         <domainId>4</domainId>        
>         <rtps>
>             <builtin>
>                 <metatrafficUnicastLocatorList> 
>                     <locator/> 
>                 </metatrafficUnicastLocatorList> 
>                 
>                 <initialPeersList>
>                     <locator>
>                         <udpv4>
>                                 <address>192.168.1.18</address>
>                         </udpv4>
>                     </locator>
>                     <locator>
>                         <udpv4>
>                                 <address>192.168.1.7</address>
>                         </udpv4>
>                     </locator>
>                 </initialPeersList>
>             </builtin>
>             <userTransports>
>             <transport_id>veelpeers</transport_id>
>             </userTransports>
>             <useBuiltinTransports>false</useBuiltinTransports>
>         </rtps>
>     </participant>
> </profiles>
> ```
> 
> CC @EduPonz to confirm.

Hi there. Thank you for the suggestion. I have tried changing it but I am receiving errors when launching gazebo or publishing a message.
```
2021-02-19 11:04:28.769 [XMLPARSER Error] Error opening 'file:///home/$USER/Desktop/default_fastrtps_profiles.xml' -> Function loadXML
2021-02-19 11:04:28.769 [XMLPARSER Error] Error parsing 'file:///home/$USER/Desktop/default_fastrtps_profiles.xml' -> Function loadXMLFile
```
In the env file i have included `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp` and `FASTRTPS_DEFAULT_PROFILES_FILE=file:///home/$USER/Desktop/default_fastrtps_profiles.xml`

May I know if it is ok for you to send me your XML file? Or you have any solution that could help to solve this issue. Thank you

EduPonz, 2021-02-19:
Hi @destkk ,

Sorry for the late reply. I have tested that the following file loads without problems:

```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles>
    <transport_descriptors>
        <transport_descriptor>
            <transport_id>veelpeers</transport_id> <!-- string -->
            <type>UDPv4</type> <!-- string -->
            <maxInitialPeersRange>100</maxInitialPeersRange> <!-- uint32 -->
        </transport_descriptor>
    </transport_descriptors>
    <participant profile_name="participant_somename" is_default_profile="true">
        <domainId>4</domainId>
        <rtps>
            <builtin>
                <initialPeersList>
                    <locator>
                        <udpv4>
                            <address>192.168.1.18</address>
                        </udpv4>
                    </locator>
                    <locator>
                        <udpv4>
                            <address>192.168.1.7</address>
                        </udpv4>
                    </locator>
                </initialPeersList>
            </builtin>
            <userTransports>
                <transport_id>veelpeers</transport_id>
            </userTransports>
            <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
    </participant>
</profiles>
```

Are you sure that `FASTRTPS_DEFAULT_PROFILES_FILE` is pointing to the right address? The same logs will appear for malformed files and missing files.

EduPonz, 2021-02-19:
Hi @destkk ,

Is it possible that you are using Windows and not Ubuntu? I suspect that what you are seeing is simply that Fast DDS is trying to load the same file twice. Fast DDS has two mechanisms for loading XML:

1. Environment variable `FASTRTPS_DEFAULT_PROFILES_FILE`
2. Placing a file `DEFAULT_FASTRTPS_PROFILES.xml` in the run directory.

From your comment, I can see that your file is called `default_fastrtps_profiles.xml`. Now, mind that Windows file names are not case sensitive, which means that if the file is on your run directory, then it will load it. Then, it will load the one in the environment variable, and if they are the same it will complain. We have a PR eProsima/Fast-DDS#1657 addressing this issue, it's currently under review.

In the mean time, you could try simply changing the name of the file and updating the environment variable accordingly. That, in combination with the XML I posted in my previous commit, should do the trick.

Please do let me know how it goes!

clalancette, 2022-09-07:
No response to the latest comment in over a year, so closing this out.  Feel free to reopen if you are still having problems.

closed 2022-09-07 by clalancette