ros2/rmw_fastrtps #522: Terminate called after throwing an instance of 'NotEnoughMemoryException' with Nav2

ruffsl, 2021-04-02:
The Nav2 stack crashes immediately upon startup when using `rmw_fastrtps_cpp` with security.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - foxy/binary
- Version or commit hash:
  - `ros-foxy-fastrtps/now 2.0.2-2focal.20210225.202742 amd64`
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Clone the turtlebot3_demo repo from the ros security working group:

```
git clone https://github.com/ros-swg/turtlebot3_demo.git --branch foxy
```

Apply the following patch to use the short and permissive root enclave:

<details>

``` diff
diff --git a/configs/common.conf b/configs/common.conf
index b4bd118..af35d98 100644
--- a/configs/common.conf
+++ b/configs/common.conf
@@ -1,10 +1,10 @@
 # setup secure session
 new-window -n turtlebot
-send-keys 'export ROS_SECURITY_ENCLAVE_OVERRIDE=/gazebo' Enter
+send-keys 'export ROS_SECURITY_ENCLAVE_OVERRIDE=/' Enter
 send-keys 'ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py' Enter
 
 split-window -h
-send-keys 'export ROS_SECURITY_ENCLAVE_OVERRIDE=/teleop' Enter
+send-keys 'export ROS_SECURITY_ENCLAVE_OVERRIDE=/' Enter
 send-keys 'ros2 run turtlebot3_teleop teleop_keyboard' Enter
 
 split-window -v
@@ -15,7 +15,7 @@ select-pane -t 1
 
 # create navigation window
 new-window -n navigation
-send-keys 'export ROS_SECURITY_ENCLAVE_OVERRIDE=/nav2_map' Enter
+send-keys 'export ROS_SECURITY_ENCLAVE_OVERRIDE=/' Enter
 send-keys 'ros2 launch nav2_bringup tb3_simulation_launch.py use_simulator:=False' Enter
 
 split-window -h
@@ -29,7 +29,7 @@ select-pane -t 1
 
 # create mapping window
 new-window -n mapping
-send-keys 'export ROS_SECURITY_ENCLAVE_OVERRIDE=/nav2_slam' Enter
+send-keys 'export ROS_SECURITY_ENCLAVE_OVERRIDE=/' Enter
 send-keys 'ros2 launch nav2_bringup tb3_simulation_launch.py use_simulator:=False slam:=True' # Enter
 
 split-window -h
```

</details>

Follow instructions in [Setting the Demo](https://github.com/ros-swg/turtlebot3_demo#setting-the-demo) for installing docker and rocker, etc. Then build and run either the demo with or without DDS security, using byobu to switch to the navigation tmux tab to either view the stdout or debug launching nav2 stack.
[... 359 lines cut ...]
[lifecycle_manager-5] 2021-03-16 16:55:39.519 [SECURITY_CRYPTO Error] Not valid SecureDataHeader submessage id -> Function decode_rtps_message
[lifecycle_manager-5] 2021-03-16 16:55:39.523 [SECURITY_CRYPTO Error] Not enough memory to deserialize SecureDataTag length -> Function decode_rtps_message
[waypoint_follower-10] 2021-03-16 16:55:39.535 [SECURITY_CRYPTO Error] Not valid SecureDataHeader submessage id -> Function decode_rtps_message
[rviz2-2] [WARN] [1615938939.610710332] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-2] [WARN] [1615938939.634312144] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-2] [WARN] [1615938939.662063819] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[ERROR] [planner_server-7]: process has died [pid 1627, exit code -11, cmd '/opt/ros/foxy/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmp8qk5tipf -r /tf:=tf -r /tf_static:=tf_static'].
[rviz2-2] [INFO] [1615938939.698829392] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[ERROR] [controller_server-6]: process has died [pid 1625, exit code -11, cmd '/opt/ros/foxy/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmp8n3w9kq6 -r /tf:=tf -r /tf_static:=tf_static'].
[ERROR] [waypoint_follower-10]: process has died [pid 1633, exit code -11, cmd '/opt/ros/foxy/lib/nav2_waypoint_follower/waypoint_follower --ros-args -r __node:=waypoint_follower --params-file /tmp/tmp4lr6d23h -r /tf:=tf -r /tf_static:=tf_static'].
[rviz2-2] [INFO] [1615938940.699127998] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-2] [INFO] [1615938941.699518268] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-2] [INFO] [1615938942.699835648] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
```

</details>

#### Additional information

Given the size of the enclave permission.p7s file (2.8K for the root enclave) is not the issue here, and the same demo works when using `rmw_connextdds`, I suspect the issue lies here or with FastRTPS itself.

Contexts:

* https://github.com/ros2/rmw_connextdds/issues/8
* https://github.com/ros-swg/turtlebot3_demo/pull/35#issuecomment-800692416

ruffsl, 2021-04-02:
cc @MiguelCompany and @JaimeMartin . It'd be great to have the security demo working again for Foxy.

IkerLuengo, 2021-04-09:
The issue is caused by a known problem on FastDDS v2.0.2, which is the version tag linked to the ROS2 foxy binaries. It was solved on [PR #1644](https://github.com/eProsima/Fast-DDS/pull/1644) and already merged to FastDDS branch 2.0.x. This branch also contains fixes for other security-related issues that were found since v2.0.2 was released.

I run the test with the latest commit of 2.0.x (which is the version linked to ROS2 foxy sources) and the demo runs correctly. The correction is also included in the latest of 2.1.x (and v2.1.1). There is a [pull request](https://github.com/ros2/ros2/pull/1114) to  ros2:foxy-release to upgrade FastDDS to v2.1.1 in foxy, which would make the demo work again.

clalancette, 2022-09-07:
Given that we have updated Foxy to Fast-RTPS 2.1 quite a while ago, I'm going to close this out.  Feel free to reopen if you are still having problems.

closed 2022-09-07 by clalancette