ros2/rmw_fastrtps #525: Build failure due to deleted function

anamud, 2021-04-13:
# Build Fail

Using sources from latest Rolling Ridley repos list. Files involved were updated recently.

```
--- stderr: rmw_fastrtps_shared_cpp                                                                                                                           
~/ros2_rolling/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/utils.cpp: In function ‘bool rmw_fastrtps_shared_cpp::find_and_check_topic_and_typ
e(const CustomParticipantInfo*, const string&, const string&, eprosima::fastdds::dds::TopicDescription**, eprosima::fastdds::dds::TypeSupport*)’:             
~/ros2_rolling/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/utils.cpp:107:71: error: use of deleted function ‘eprosima::fastdds::dds::TypeSupp
ort& eprosima::fastdds::dds::TypeSupport::operator=(const eprosima::fastdds::dds::TypeSupport&)’                                                              
   *returned_type = participant_info->participant_->find_type(type_name);                                                                                     
                                                                       ^                                                                                      
In file included from ~/ros2_rolling/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/utils.hpp:22:0,                 
                 from ~/ros2_rolling/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/utils.cpp:17:                                               
~/ros2_rolling/install/fastrtps/include/fastdds/dds/topic/TypeSupport.hpp:50:7: note: ‘eprosima::fastdds::dds::TypeSupport& eprosima::fastdds::dds:
:TypeSupport::operator=(const eprosima::fastdds::dds::TypeSupport&)’ is implicitly declared as deleted because ‘eprosima::fastdds::dds::TypeSupport’ declares 
a move constructor or move assignment operator                                                                                                                
 class TypeSupport : public std::shared_ptr<fastdds::dds::TopicDataType>                                                                                      
       ^~~~~~~~~~~                                                                                                                                            
make[2]: *** [CMakeFiles/rmw_fastrtps_shared_cpp.dir/src/utils.cpp.o] Error 1
```

## Build Setup

- L4T 32.5.0  based on Ubuntu 18.04.5
- gcc/g++ 7.5.0
- eProsima/Fast-CDR v1.0.20
- eProsima/FastDDS 2.3.x

MiguelCompany, 2021-04-13:
@anamud Could you please check if eProsima/Fast-DDS#1912 fixes this issue?

anamud, 2021-04-13:
@MiguelCompany Thanks for the fast response. I confirm that build succeeds in my setup with change [1912](https://github.com/eProsima/Fast-DDS/pull/1912).

MiguelCompany, 2021-04-13:
@anamud PR has been merged and is now present on `2.3.x`, so I think this can be closed.

anamud, 2021-04-13:
Fixed with lightning speed. Thanks!

closed 2021-04-13 by anamud