ros2/rmw_fastrtps #526: Specify network interface names instead of static IP addresses

fujitatomoya, 2021-04-14:
we'd like to open discussion against our use case, which is container with multiple network interface including container orchestration.

## Description

whitelist https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/whitelist.html, it only supports static IP addresses but not with network interface name, since the NIC name almost is fixed but IP maybe dynamic assigned by DHCP or container orchestration framework. so the whitelist profile only support IP seems more static in some use cases.

probably it would be nice that we can specify the network card name instead of static IP addresses.

clalancette, 2021-04-29:
This actually looks like a feature request for Fast-DDS itself; `rmw_fastrtps` wouldn't really be involved here.  Is that correct?  If so, I'd suggest closing this one and opening an issue over on https://github.com/eProsima/Fast-DDS

fujitatomoya, 2021-04-29:
@clalancette correct, i will move this to https://github.com/eProsima/Fast-DDS

fujitatomoya, 2021-04-29:
note: moved to https://github.com/eProsima/Fast-DDS/issues/1943

closed 2021-04-29 by fujitatomoya