ros2/rmw_fastrtps #528: Downgrade rmw_fastrtps_dynamic_cpp to Tier 3 and/or remove the code

clalancette, 2021-04-29:
The rmw_fastrtps_dynamic_cpp code has been at [Tier 2 support](https://www.ros.org/reps/rep-2000.html#galactic-geochelone-may-2021-november-2022) since at least Crystal, and it has fallen behind the fastrtps static typesupport implementation in terms of features and overall quality.  We're wondering what to do with this implementation for both Galactic and going forward into H-Turtle.  Some of the options that come to mind are:

1. Put some serious effort into it and bring it up to quality standards, aiming for it to be Tier 1.  This work would probably have to be done by eProsima or another external entity, as the core ROS 2 team wouldn't have time for this.
2. Reduce the support level of it to Tier 3, and stop running nightly build-n-test against it.
3. Remove it from https://github.com/ros2/rmw_fastrtps_cpp completely.
4. Remove it from https://github.com/ros2/rmw_fastrtps_cpp, and move it to another repository so it can be released separately into ROS 2 distributions.

We're considering option 2 for Galactic, and option 3 or 4 for Rolling and H-Turtle.  But we'd like to hear feedback, especially from eProsima about it.  Ping @MiguelCompany @EduPonz @JaimeMartin for feedback on what we should do here.

MiguelCompany, 2021-09-10:
@clalancette I think the best approach would be something similar to what has been done with rmw_connextdds: having a single rmw implementation that could handle both type supports. We could name it `rmw_fastdds`.

That said, I don't know if/when we'll have resources for that.

Crola1702, 2023-08-09:
https://github.com/Mergifyio backport iron

ZhenshengLee, 2023-11-29:
What't the status for now?

clalancette, 2023-11-29:
> What't the status for now?

It is still Tier 2, as shown in https://www.ros.org/reps/rep-2000.html#iron-irwini-may-2023-november-2024