ros2/rmw_fastrtps #533: NotEnoughMemoryException with Actions

DLu, 2021-05-20:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ros-foxy-rmw-fastrtps-cpp 1.2.5-1focal.20210513.200901
- DDS implementation:
  - FastDDS
- Client library (if applicable):
  - rclcpp AND rclpy

#### Steps to reproduce issue
Self contained example here: https://github.com/DLu/my_happy_package
including stack trace, expected behavior and actual behavior. 

#### Additional information
[Related answers.ros.org question](https://answers.ros.org/question/378355/eprosimafastcdrexceptionnotenoughmemoryexception-with-rclpyactionclient/) with example of error in rclpy

clalancette, 2021-05-20:
Hm, so far, I've been unable to replicate this result.  I built `my_happy_package` on two separate machines on my network, and I was able to run the server/client pair on either one, and have it generate the expected result with Foxy and Fast-RTPS.  For completeness I also was able to successfully run with CycloneDDS on Foxy as well.

The "NotEnoughMemoryException" actually comes from Fast-CDR, not Fast-RTPS: https://github.com/eProsima/Fast-CDR/blob/v1.0.13/include/fastcdr/exceptions/NotEnoughMemoryException.h .  So one additional piece of information that would be interesting is what version of fastcdr you have installed.

It looks like that error thrown when Fast-CDR fails to serialize a field into a message buffer.  That should only happen when the buffer handed to Fast-CDR is too small, but I don't really know why that would be.  Do you have a lot of other DDS-related traffic on the network at the same time you are trying this?

On other thing you can attempt to build Fast-RTPS 2.1.x, as we are thinking about doing for Foxy here: https://github.com/ros2/ros2/pull/1113 .  As it stands now, you'll realistically have to build the entire stack with that in place, but it could give us an idea on whether that update fixes the problem you are seeing.  Alternatively, you could try Rolling or Galactic with `RMW_IMPLEMENTATION=rmw_fastrtps_cpp`, which would also give us a bit more information.

LouisChen1905, 2022-03-04:
@DLu Hi, I encountered a similar issue recently. How do you fix it?

DLu, 2022-03-24:
I believe I just switched to a different DDS

MiguelCompany, 2025-04-30:
Service and actions are not interoperable between Cyclone DDS and other DDS rmw implementations

MiguelCompany, 2025-04-30:
> I'm using `rmw_cyclone_dds` both in my robot and my pc.

But micro ros uses serialization mechanisms which are not interoperable with `rmw_cyclone_dds`