ros2/rmw_fastrtps #534: Data sharing doesn't work while `publishMode` is `SYNCHRONOUS`

Barry-Xu-2018, 2021-05-27:
## Bug report

Data sharing doesn't work while `publishMode` is `SYNCHRONOUS`.  If set `ASYNCHRONOUS`, data sharing can work well.

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
    - Fast-DDS
      version: cec6e080b8c00ae8c0ab8dd8fc658a2be5b8aaa4
    - rmw_fastrtps
      version: 7cb3bd0f73f332f9c27caf505b5bcd1b055c0ec7
    - rosidl_typesupport_fastrtps
      version: 748be563227b889e07ab516460e17813cdf7d338
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

1. Download test codes from https://github.com/Barry-Xu-2018/bug_reproduce
2. Build codes
    ```bash
   $ colcon build --packages-select bug_reproduce  --symlink-install
    ```
3. Source install/setup.bash
4. Enter bug_reproduce/src/data_share
5. Modify PUB.xml at line 18
    ```xml
    <kind>ASYNCHRONOUS</kind>
    ```  
    to
    ```xml
    <kind>SYNCHRONOUS</kind>
    ``` 
 6. Execute talker
     ```bash
     RMW_FASTRTPS_USE_QOS_FROM_XML=1 FASTRTPS_DEFAULT_PROFILES_FILE=PUB.xml RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run bug_reproduce data_sharing_talker
     ```
  7. Execute listener
     ```bash
     RMW_FASTRTPS_USE_QOS_FROM_XML=1 FASTRTPS_DEFAULT_PROFILES_FILE=SUB.xml RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run bug_reproduce data_sharing_listener
     ```
 
 To confirm if data sharing works, I check if `send_all_datasharing_changes`  is called.   
https://github.com/eProsima/Fast-DDS/blob/c1c7c797d7c0cab1c2e8978b222ee4aa1fcc2cd5/src/cpp/rtps/writer/StatefulWriter.cpp#L1101-L1104

#### Expected behavior

Data sharing works.

#### Actual behavior

Use other transport.

#### Additional information

Barry-Xu-2018, 2021-05-27:
According to my understanding, no matter what `publishMode` for publisher is set, this should not affect data sharing.    
Please correct me if I am wrong.

MiguelCompany, 2021-05-27:
I haven't tried to reproduce this yet, but your analysis is not correct. 

Method `send_all_datasharing_changes` is only called on `ASYNCHRONOUS` mode. When the writer is using `SYNCHRONOUS`, the reader is directly notified within `unsent_change_added_to_history` via `sync_delivery` [here](https://github.com/eProsima/Fast-DDS/blob/c1c7c797d7c0cab1c2e8978b222ee4aa1fcc2cd5/src/cpp/rtps/writer/StatefulWriter.cpp#L490)

All this is for reliable writers. For best-effort ones, datasharing delivery is always done [here](https://github.com/eProsima/Fast-DDS/blob/c1c7c797d7c0cab1c2e8978b222ee4aa1fcc2cd5/src/cpp/rtps/writer/StatelessWriter.cpp#L352).

Barry-Xu-2018, 2021-05-28:
> I haven't tried to reproduce this yet, but your analysis is not correct.
>
> Method `send_all_datasharing_changes` is only called on `ASYNCHRONOUS` mode. When the writer is using `SYNCHRONOUS`, the [...]

Thanks for your explanation. I reconfirm whether data sharing work while publishMode is set to `SYNCHRONOUS`.

Barry-Xu-2018, 2021-05-28:
@MiguelCompany 

I confirm data sharing also work while publishMode is set to SYNCHRONOUS.  
I close this issue.

closed 2021-05-28 by Barry-Xu-2018