ros2/rmw_fastrtps #538: rmw_fastrtps_shared_cpp fails to build due to undeclared RMW_DURATION_INFINITE

YVbakker, 2021-06-02:
## Bug report

**Required Info:**

- Operating System:
  - Mac OS Big Sur 11.4
- Installation type:
  - From source
- Version or commit hash:
  - release/galactic
- DDS implementation:
  - Fast-RTPS (default)
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

just build rmw_fastrtps_shared_cpp package as usual with colcon
```
colcon build --symlink-install --packages-skip-by-dep python_qt_binding --packages-skip-build-finished --executor sequential
```

#### Expected behavior

Package builds succesfully

#### Actual behavior

rmw_fastrtps_shared_cpp fails to build because:
tests fail to build:
specifically this test:
src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/test/test_dds_attributes_to_rmw_qos.cpp fail
error: use of undeclared identifier 'RMW_DURATION_INFINITE' (on lines 147, 148, 149, 157, 158, 159 in test_dds_attributes_to_rmw_qos.cpp)
this constant comes from rmw/time.h
which is included in rmw/types.h
which is included in rmw/rmw.h
which is included in rmw_fastrtps_shared_cpp/qos.hpp
which is included in test_dds_attributes_to_rmw_qos.cpp

for some reason the GTEST macro cannot resolve this constant

#### Additional information

Experimental setup, compiling for M1 (aarch64/arm64) hardware. No other unusual steps taken. Perhaps it is due to sequential build, and a wrong dependency order somewhere? But due to most people building parallel this might not have popped up before?

clalancette, 2021-06-02:
> Experimental setup, compiling for M1 (aarch64/arm64) hardware. No other unusual steps taken. Perhaps it is due to sequential build, and a wrong dependency order somewhere? But due to most people building parallel this might not have popped up before?

It's possible, but unlikely.  It seems more like you are on the wrong branch of one of the repositories.  `RMW_DURATION_INFINITE` is defined in https://github.com/ros2/rmw/blob/a50f3bb1e24ac5338677990d74ab096a9d1776d7/rmw/include/rmw/time.h#L54, but only on the `master` and `galactic` branches; it didn't exist before then.  What branch are you on in the `rmw` repository?

YVbakker, 2021-06-02:
I am on the master branch, on f9f4ef46f8dd3ef938526a1140a3c57c155a259f to be exact (tag: 3.3.1, origin/galactic). I'm already three commits behind since this morning I see.

I do also have the time.h in my src with `RMW_DURATION_INFINITE` defined in there, and it should be included through the above trace. When I manually define it in the files (for the sake of troubleshooting) it cannot resolve  method `rmw_time_equal` either.

clalancette, 2021-06-02:
Looking at it briefly, it looks like there are some places where we don't include-what-you-use (`test_dds_attributes_to_rmw_qos.cpp` is missing an `rmw/time.h` include, for instance).  That might help, but I'm honestly not sure.  This works on the buildfarm (which is admittedly macOS Mojave): https://ci.ros2.org/view/nightly/job/nightly_osx_release/

YVbakker, 2021-06-02:
Thank you, I will try it with some of the configuration options from the buildfarm.
I did also try manually including rmw/time.h in the cpp files where it was complaining, but to no avail: then it started complaining about duplicate definitions (error: redefinition of 'rmw_time_t') (which I don't understand since there are #ifndef #define #endif statements in all those headers, right?) and on top of that still gives the `error: use of undeclared identifier 'RMW_DURATION_INFINITE'  qos_profile_.lifespan = RMW_DURATION_INFINITE;` errors, while RMW_DURATION_INFINITE is literally defined in that include file (rmw/time.h)

closed 2023-06-13 by clalancette