ros2/rmw_fastrtps #541: could not create publisher/subscription: Type support not from this implementation

TSC21, 2021-06-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - rmw-fastrtps `5.0.0-2focal.20210519.024652`
- DDS implementation:
  - FastDDS 2.3.1 in ROS2 Galactic
- Client library (if applicable):
  - rclcpp in ROS2 Galactic

#### Steps to reproduce issue
The ROS2 packages are being cross-compiled using https://github.com/ros-tooling/cross_compile, which supposedly configures the build environment correctly. They are then deployed to an `arm64` platform and run there.

I am not able to start any node when setting `RMW_IMPLEMENTATION=rmw_fastrtps_cpp` in ROS2 Galactic, throwing the error bellow. The same doesn't happen with ROS2 Foxy.

#### Expected behavior
Being able to start and use the ROS2 CLI to echo topics, etc.

#### Actual behavior
The bellow shows when starting a ROS2 node with a publisher:

```sh
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
what():  could not create publisher: Type support not from this implementation. Got:
    error not set
    error not set
while fetching it, at /tmp/binarydeb/ros-galactic-rmw-fastrtps-cpp-5.0.0/src/publisher.cpp:115, at /tmp/binarydeb/ros-galactic-rcl
-3.1.2/src/rcl/publisher.c:114
```

#### Additional information

If the type support is not from "this" implementation, what should it be then? Does it mean that one has to build the packages with the rmw_implementation set in the CMakeLists.txt file of the package, targeting then RMW we are going to use the package on? Or is that not relevant here? (shouldn't be IMO)

@MiguelCompany

TSC21, 2021-06-14:
Any feedback that can be provided here?

1r0b1n0, 2021-08-25:
Hi, I had the same problem on one of my projets.
To use a ROS2 workspace with fastDDS, its library must be present during compilation.  Having installed fastDDS during runtime is not enough.
My solution was to simply recompile the workspace after installing "ros-galactic-rmw-fastrtps-cpp".

clalancette, 2021-09-09:
@MiguelCompany any thoughts on what might be happening here?

MiguelCompany, 2021-09-10:
@clalancette I think the scenario may be as follows:

* A user installs galactic-core from packages
* The user creates a package with message definitions and builds
  * As only rmw_cyclonedds_cpp is installed, and it only requires the introspection type support, only that type support is generated
* The user then installs the rmw_fastrtps_cpp package, and sets `RMW_IMPLEMENTATION=rmw_fastrtps_cpp`
* When the publisher is created, the introspection type support is passed in, giving the error

Something that can be tried, apart from @1r0b1n0 suggestion of rebuilding the workspace, is to use `RMW_IMPLEMENTATION=rmw_fastrtps_dynamic_cpp`, which expects the introspection type support to be passed in.

@TSC21 Could you try that approach?

huichengongzi, 2021-12-10:
Hi, I had the some problem in my project. reprodection as that:
1, apt-get install ros-galactic-rmw-fastrtps-cpp
2, export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
3, ros run ...
Solution is that clean my project and rebuild it, then ros run, it can work fine.

zjeffer, 2022-03-19:
Deleting the build & install folders and then rebuilding the package with `ros-galactic-rmw-fastrtps-cpp` installed fixed the issue, thank you @huichengongzi @1r0b1n0

ZhenshengLee, 2022-05-05:
this issue happens in ros2 source build too.

```sh
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```

then rebuild ros2 solves this issue.

clalancette, 2022-05-05:
Sounds good, thanks!

closed 2022-05-05 by clalancette