ros2/rmw_fastrtps #542: Some Individual Topic Subscriber and Publisher QOS settings no longer work

smnogar, 2021-06-22:
## Bug report

I am trying to implement partitions in my individual publishers and subscribers in ROS2 Galactic. However, the following PR https://github.com/ros2/rmw_fastrtps/pull/518 broke including these QOS settings on a per topic basis. As far as I can tell, publish mode and memory policy still work.

**Required Info:**

- Operating System:
  - Ubuntu 20.04, galactic
- Installation type:
  - binaries
- Version or commit hash:
 - fastrtps 2.3.1
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Using the following two XML profiles:

```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">    

    <!-- Default publisher profile -->
    <publisher profile_name="default publisher profile" is_default_profile="true">
        <qos>
            <publishMode>
                <kind>ASYNCHRONOUS</kind>
            </publishMode>
        </qos>
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </publisher>

    <subscriber profile_name="default subscriber profile" is_default_profile="true">
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </subscriber>

    <publisher profile_name="/chatter">
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        <qos>
            <publishMode>
                <kind>ASYNCHRONOUS</kind>
            </publishMode>
            <partition>
                <names>
                    <name>part1</name>
                </names>
            </partition>
        </qos>
    </publisher>

</profiles>
```

and 

```
<?xml version="1.0" encoding="UTF-8" ?>

<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">    
    <!-- Default publisher profile -->
    <publisher profile_name="default publisher profile" is_default_profile="true">
        <qos>
            <publishMode>
                <kind>ASYNCHRONOUS</kind>
            </publishMode>
        </qos>
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </publisher>

    <subscriber profile_name="default subscriber profile" is_default_profile="true">
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </subscriber>

    <subscriber profile_name="/chatter">
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        <qos>
            <partition>
                <names>
                    <name>part2</name>
                </names>
            </partition>
        </qos>
    </subscriber>
</profiles>
```

Load the profiles using FASTRTPS_DEFAULT_PROFILES_FILE
Run `ros2 run demo_nodes_cpp talker` and `ros2 run demo_nodes_cpp listener` on the other.

#### Expected behavior

No response from the listener. 

#### Actual behavior

Response from the listener.

#### Additional information

MiguelCompany, 2021-06-22:
I confirmed this behavior. The thing is that `DataWriterQos` and `DataReaderQos` do not have support for partitions, because the DDS standard specifies that partitions reside in the `PublisherQos` and `SubscriberQos`.

With the current implementation, this would require creating different publishers for each DataWriter using partitions. As explained on [this OMG issue](https://issues.omg.org/issues/spec/DDS/1.4#issue-45578), this would be resource consuming, and it makes more sense to allow the PARTITION Qos to be set directly on the DataWriter and DataReader.

I have just created eProsima/Fast-DDS#2018 to track down that we should implement that behavior.

clalancette, 2022-02-03:
No response in 3 months, so I'm going to go ahead and close this.  If it isn't fixed for you, please feel free to reopen.

closed 2022-02-03 by clalancette