ros2/rmw_fastrtps #545: FastDDS Discovery Server does not work locally without a network connection

ralph-lange, 2021-06-29:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Distro: Galactic
- Installation type: Binaries
- Version or commit hash: 5.0.0-2 (release version of ros-galactic-rmw-fastrtps-cpp)
- DDS implementation: FastDDS
- Client library (if applicable): N/A

#### Steps to reproduce issue
- Disable all network connections (unplug Ethernet, turn off WiFi)
- Source the ROS Galactic installation
- Invoke `fastdds discovery -i 0` on a console (cf. https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html#setup-discovery-server)

#### Expected behavior
Discovery server should start up or give a meaningful error message that at least one network connection is required.

#### Actual behavior
The command stops immediately with output
```
[PARTICIPANT ERROR] Problem creating RTPSParticipant -> Function createParticipant
fastdds discovery <discovery-args>

1
 fast-discovery-server tool not found!
```
... which suggests an incomplete installation, which is not true.

#### Additional information
The same error occurs on Foxy and Rolling. If the computer has a network connection (WiFi, Ethernet), then the whole demo in Section 16.2.3 from https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html#run-the-demo run smoothly.

MiguelCompany, 2021-06-30:
@ralph-lange I think this issue should be on eProsima/Fast-DDS. There's nothing we can do on this repo to fix it.

ralph-lange, 2021-06-30:
I'll move it.

ralph-lange, 2021-06-30:
Moved to https://github.com/eProsima/Fast-DDS/issues/2031.

closed 2021-06-30 by ralph-lange