ros2/rmw_fastrtps #551: Expose locator addresses to rmw_fastrtps for using ipv6

RFRIEDM-Trimble, 2021-08-06:
## Support IPv6 locators in ROS2 rmw_fastrtps
#### As a user of ROS2 in a distributed network that only uses IPv6, I do not have a documented way to run ROS2 on IPv6. 

Due to some limitations of IPv4, I have a network that only supports IPv6; IPv4 is disabled and dropped by the routers. While FastDDS middleware supports IPv6 with fixes that I have tested and contributed to, I do not see a way to utilize that. I want to change the address of all my ROS2 locators to be IPv6 multicast.

#### Implementation considerations
* Works on different ipv6 scopes such as link local
* If it follows RFC's for addressing convention0, that makes it easier for other DDS vendors can suppor the same API in the future
* I am happy to contribute to this feature; just need some direction
* Should be allowed to change the address for each "node" in ROS2, so no global environement variables

EduPonz, 2021-08-06:
Hi @RFRIEDM-Trimble 

I think I bring good news! You can currently setup ROS 2 to use IPv6 when using rmw_fasrtps. What you'd need to do is creating an XML profiles file with a default participant profile which sets UDPv6 as transport (take a look at [DomainParticipant profiles](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/domainparticipant.html#domainparticipant-profiles) and [transport descriptors](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/transports.html#transport-descriptors)). A basic example that includes a UDPv6 transport for LAN comms and a SHM transport for interprocess comms would look something like:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
	<profiles>
        <!--
            List of transport descriptors 
        -->
        <transport_descriptors>
            <!-- 
                Create a UDPv6 transport descriptor
             -->
            <transport_descriptor>
                <transport_id>udpv6_transport</transport_id>
                <type>UDPv6</type>
                <!-- Any other parameter to configure goes here -->
            </transport_descriptor>
            <!-- 
                Create a SHM transport descriptor
             -->
            <transport_descriptor>
                <transport_id>shm_transport</transport_id>
                <type>SHM</type>
                <!-- Any other parameter to configure goes here -->
            </transport_descriptor>
        </transport_descriptors>	
        <!--
            Default DomainParticipant profile
        -->
        <participant profile_name="udpv6_participant" is_default_profile="true">
            <rtps>
                <!--
                    Discard default UDPv4 and SHM transports
                -->
                <useBuiltinTransports>false</useBuiltinTransports>
                <!--
                    Select one or more transport descriptors from the list above
                -->
                <userTransports>
                    <transport_id>udpv6_transport</transport_id>
                    <transport_id>shm_transport</transport_id>
                </userTransports>
            </rtps>
        </participant>
    </profiles>
</dds>
```

To load the file, you'd need to populate environment variable `FASTRTPS_DEFAULT_PROFILES_FILE` with the path to you XML file. Mind that this is need in every shell session where you want the XML to apply. For a more detail explanation please refer to [README.md](https://github.com/ros2/rmw_fastrtps#readme)

If you're also talking about adding API to rmw to configure such parameters, I suggest you open a ticket in https://github.com/ros2/ros2 with your feature request  so we can kick-off a broader discussion that involves the entire ROS 2 stack.

RFRIEDM-Trimble, 2021-08-06:
Thank you for the prompt response. I will gladly try out the XML profile method. If it does not meet our needs, I will open a follow up ticket. I have updated the ticket name to include XML profiles; since it is already implemented, I'll close the ticket.

closed 2021-08-06 by RFRIEDM-Trimble