ros2/rmw_fastrtps #554: Unable to create publishers/subscribers with different profiles

FiIipe, 2021-09-03:
## Bug report

Hi! I'm trying to create publishers and subscribers with different profiles using the XML file. 
I started with a simple test by following the documentation available [here.](https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2_configure.html#creating-publishers-subscribers-with-different-profiles)
When I run the node, the XML profile loaded is the default profile and not the custom publisher profile. 

**Required Info:**

- Operating System: Ubuntu 20.04
  - 
- Installation type: Source
  - 
- Version or commit hash: ROS2 Galatic
  - 
- DDS implementation: Fast-RTPS
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

XML configuration file:
```
<?xml version="1.0" encoding="UTF-8"?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <profiles>
        <!-- Default publisher profile -->
        <publisher profile_name="default publisher profile" is_default_profile="true">
            <qos>
                <publishMode>
                    <kind>SYNCHRONOUS</kind>
                </publishMode>
                <reliability>
                    <kind>BEST_EFFORT</kind>
                </reliability>
            </qos>
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </publisher>

        <!-- Publisher profile for topic chatter -->
        <publisher profile_name="chatter">
            <qos>
                <publishMode>
                    <kind>SYNCHRONOUS</kind>
                </publishMode>
                <reliability>
                    <kind>RELIABLE</kind>
                </reliability>
            </qos>
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </publisher>
    </profiles>
</dds>
```

C++ publisher:

```
#include <chrono>
#include <memory>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"

using namespace std::chrono_literals;

/* This example creates a subclass of Node and uses std::bind() to register a
 * member function as a callback from the timer. */

class MinimalPublisher : public rclcpp::Node
{
public:
  MinimalPublisher()
  : Node("minimal_publisher"), count_(0)
  {
    publisher_ = this->create_publisher<std_msgs::msg::String>("chatter", rclcpp::SystemDefaultsQoS());
    timer_ = this->create_wall_timer(
      500ms, std::bind(&MinimalPublisher::timer_callback, this));
  }

private:
  void timer_callback()
  {
    auto message = std_msgs::msg::String();
    message.data = "Hello, world! " + std::to_string(count_++);
    RCLCPP_INFO(this->get_logger(), "Publishing: '%s'", message.data.c_str());
    publisher_->publish(message);
  }
  rclcpp::TimerBase::SharedPtr timer_;
  rclcpp::Publisher<std_msgs::msg::String>::SharedPtr publisher_;
  size_t count_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalPublisher>());
  rclcpp::shutdown();
  return 0;
}

```

And the command to run the node: 

```
FASTRTPS_DEFAULT_PROFILES_FILE=/home/$USER$/dev_ws/fastrtps_profiles.xml RMW_FASTRTPS_USE_QOS_FROM_XML=1 RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run cpp_pubsub talker
```

Info about the topic:

```
ros2 topic info /chatter  -v
```

#### Expected behavior

Type: std_msgs/msg/String

Publisher count: 1

Node name: minimal_publisher
Node namespace: /
Topic type: std_msgs/msg/String
Endpoint type: PUBLISHER
GID: 01.0f.74.0d.89.41.41.4b.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: **RELIABLE**
  Durability: TRANSIENT_LOCAL
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

Subscription count: 0

#### Actual behavior

Type: std_msgs/msg/String

Publisher count: 1

Node name: minimal_publisher
Node namespace: /
Topic type: std_msgs/msg/String
Endpoint type: PUBLISHER
GID: 01.0f.74.0d.89.41.41.4b.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: **BEST_EFFORT**
  Durability: TRANSIENT_LOCAL
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

Subscription count: 0

Thank you very much.

EduPonz, 2021-09-06:
Hi @FiIipe ,

I'm afraid what you're seeing is the actual expected behaviour. From https://github.com/eProsima/foonathan_memory_vendor/pull/49

```
ROS 2 QoS contained in `rmw_qos_profile_t` are always honored, unless set to *_SYSTEM_DEFAULT.
```

I'll make sure we update Fast DDS' documentation to clarify this point, thanks for the catch!

EduPonz, 2021-09-06:
Looking again at Fast DDS documentation, I can see that it's actually already detailed in https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2_configure.html#xml-configuration

FiIipe, 2021-09-06:
Hi @EduPonz, 

Thank you for your quick response. 

This is my publisher declaration: 
`publisher_ = this->create_publisher<std_msgs::msg::String>("chatter", rclcpp::SystemDefaultsQoS());`

The publisher that I'm using is declared with the following QoS settings: rclcpp::[SystemDefaultsQoS()](https://docs.ros2.org/dashing/api/rclcpp/classrclcpp_1_1SystemDefaultsQoS.html). According to the documentation, this QoS setting sets all the ROS2 QoS contained in `rmw_qos_profile_t` to *_SYSTEM_DEFAULT. 

Therefore, it is expected that, when we have a publisher profile defined in the XML file with the name of the topic that I'm publishing, the publisher would use the QoS settings defined in the XML file, correct? Or is my interpretation of the documentation wrong? 

Cheers, Filipe

EduPonz, 2021-09-07:
Hi @FiIipe 

> publisher_ = this->create_publisher<std_msgs::msg::String>("chatter", rclcpp::SystemDefaultsQoS());

This seems to be about right.

> Therefore, it is expected that, when we have a publisher profile defined in the XML file with the name of the topic that I'm publishing, the publisher would use the QoS settings defined in the XML file, correct? Or is my interpretation of the documentation wrong?

Your interpretation is indeed correct.

I do not know the details of how `ros2 topic info` get and reports QoS, so I'll need some time to investigate and replicate on our side.

FiIipe, 2021-09-07:
Hi @EduPonz,

Sure, thank you. Let me know if you need more information. 

Cheers, 
Filipe

fujitatomoya, 2021-09-07:
@EduPonz 

this problem can be observed with mainline https://github.com/ros2/ros2/commit/2fafc39365dc031c08c570078c190d5c597ef9eb. 

> ROS 2 QoS contained in rmw_qos_profile_t are always honored, unless set to *_SYSTEM_DEFAULT. In that case, XML values, or Fast DDS default values in the absence of XML ones, are applied. Setting any QoS in rmw_qos_profile_t to something other than *_SYSTEM_DEFAULT entails that specifying it via XML files has no effect, since they do not override what was used to create the publisher, subscription, service, or client.

if i am not missing, i cannot find any of above implementation in `rmw_fastrtps`.

iuhilnehc-ynos, 2021-09-08:
@FiIipe 

Could you try using the Fully Qualified Name (FQN) of the topic for the profile_name? ( "chatter" --> "/chatter" )

EduPonz, 2021-09-08:
Hi everyone,

Long story short, I have reproduced @FiIipe 's issue and @iuhilnehc-ynos is right, writing `/chatter` does the trick. I'll submit a PR to fix the documentation today. In the mean time, I leave here my procedure:

### Steps to reproduce and fix

1. I have build ROS 2 Galactic from source on Ubuntu 20.04
1. I have modified https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/topics/talker.cpp#L54 to set System Default QoS:
    ```diff
    diff --git a/demo_nodes_cpp/src/topics/talker.cpp b/demo_nodes_cpp/src/topics/talker.cpp
    index 30f97a0..6e80536 100644
    --- a/demo_nodes_cpp/src/topics/talker.cpp
    +++ b/demo_nodes_cpp/src/topics/talker.cpp
    @@ -51,7 +51,7 @@ public:
           };
         // Create a publisher with a custom Quality of Service profile.
         rclcpp::QoS qos(rclcpp::KeepLast(7));
    -    pub_ = this->create_publisher<std_msgs::msg::String>("chatter", qos);
    +    pub_ = this->create_publisher<std_msgs::msg::String>("chatter", rclcpp::SystemDefaultsQoS());
    
         // Use a timer to schedule periodic message publishing.
         timer_ = this->create_wall_timer(1s, publish_message);
    ```
1. I have verified that running the demo with the XML file provided by @FiIipe does not take the document effect. Instead of selecting the `chatter` profile, it is selecting the `default publisher profile`, tagged as `is_default_profile="true"`
    **Terminal 1:**
    ```bash
    FASTRTPS_DEFAULT_PROFILES_FILE=fastrtps_profiles.xml RMW_FASTRTPS_USE_QOS_FROM_XML=1 RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 run demo_nodes_cpp talker
    ```
   **Terminal 2:**
    ```bash
    RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 topic info /test/chatter -v
    ```
    **Expected output according to documentation:**
    ```
    Type: std_msgs/msg/String

    Publisher count: 1

    Node name: talker
    Node namespace: /
    Topic type: std_msgs/msg/String
    Endpoint type: PUBLISHER
    GID: 01.0f.d9.17.fe.92.e0.cd.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
    QoS profile:
      Reliability: RELIABLE
      Durability: TRANSIENT_LOCAL
      Lifespan: 9223372036854775807 nanoseconds
      Deadline: 9223372036854775807 nanoseconds
      Liveliness: AUTOMATIC
      Liveliness lease duration: 9223372036854775807 nanoseconds

    Subscription count: 0
    ```
    **Actual output:**
    ```
    Type: std_msgs/msg/String

    Publisher count: 1

    Node name: talker
    Node namespace: /
    Topic type: std_msgs/msg/String
    Endpoint type: PUBLISHER
    GID: 01.0f.d9.17.fe.92.e0.cd.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
    QoS profile:
      Reliability: BEST_EFFORT
      Durability: TRANSIENT_LOCAL
      Lifespan: 9223372036854775807 nanoseconds
      Deadline: 9223372036854775807 nanoseconds
      Liveliness: AUTOMATIC
      Liveliness lease duration: 9223372036854775807 nanoseconds

    Subscription count: 0
    ```
1. After modifying the profile name setting it to `/chatter`
    ```
    Type: std_msgs/msg/String

    Publisher count: 1

    Node name: talker
    Node namespace: /
    Topic type: std_msgs/msg/String
    Endpoint type: PUBLISHER
    GID: 01.0f.d9.17.fe.92.e0.cd.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
    QoS profile:
      Reliability: RELIABLE
      Durability: TRANSIENT_LOCAL
      Lifespan: 9223372036854775807 nanoseconds
      Deadline: 9223372036854775807 nanoseconds
      Liveliness: AUTOMATIC
      Liveliness lease duration: 9223372036854775807 nanoseconds

    Subscription count: 0
    ```
1. Then, I went a bit deeper and further modified the demo to add a namespace to the node:
    ```diff
    diff --git a/demo_nodes_cpp/src/topics/talker.cpp b/demo_nodes_cpp/src/topics/talker.cpp
    index 30f97a0..6ad4f2d 100644
    --- a/demo_nodes_cpp/src/topics/talker.cpp
    +++ b/demo_nodes_cpp/src/topics/talker.cpp
    @@ -35,7 +35,7 @@ class Talker : public rclcpp::Node
    public:
      DEMO_NODES_CPP_PUBLIC
      explicit Talker(const rclcpp::NodeOptions & options)
    -  : Node("talker", options)
    +  : Node("talker", "test", options)
      {
        // Create a function for when messages are to be sent.
        setvbuf(stdout, NULL, _IONBF, BUFSIZ);
    @@ -51,7 +51,7 @@ public:
          };
        // Create a publisher with a custom Quality of Service profile.
        rclcpp::QoS qos(rclcpp::KeepLast(7));
    -    pub_ = this->create_publisher<std_msgs::msg::String>("chatter", qos);
    +    pub_ = this->create_publisher<std_msgs::msg::String>("chatter", rclcpp::SystemDefaultsQoS());
    
        // Use a timer to schedule periodic message publishing.
        timer_ = this->create_wall_timer(1s, publish_message);
    ```
1. This results in the publisher publishing to topic `/test/chatter`
1. I changed the profile name in the XML from `/chatter` to `/test/chatter` and verified that everything works:
    ```
    Type: std_msgs/msg/String

    Publisher count: 1

    Node name: talker
    Node namespace: /test
    Topic type: std_msgs/msg/String
    Endpoint type: PUBLISHER
    GID: 01.0f.d9.17.fe.92.e0.cd.01.00.00.00.00.00.12.03.00.00.00.00.00.00.00.00
    QoS profile:
      Reliability: RELIABLE
      Durability: TRANSIENT_LOCAL
      Lifespan: 9223372036854775807 nanoseconds
      Deadline: 9223372036854775807 nanoseconds
      Liveliness: AUTOMATIC
      Liveliness lease duration: 9223372036854775807 nanoseconds

    Subscription count: 0
    ```

EduPonz, 2021-09-08:
I have created #557 to correct the documentation, please take a look at it if you have the time.

MiguelCompany, 2021-09-14:
@FiIipe 
Now that #557 has been merged, and the rules on how the topic name should be specified are clear, I think this issue can be closed.

FiIipe, 2021-09-24:
Thank you. Yes, we can close this.

closed 2021-09-24 by FiIipe