ros2/rmw_fastrtps #566: ros2 topics and services are missing using Discovery Server

eranroll, 2021-10-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 with Ubuntu 20.04 container - over Nvidia Jetson
- Installation type:
 - Source
- Version or commit hash:
 - ROS2 Galactic
- DDS implementation:
  - rmw_fastrtps_cpp

#### Steps to reproduce issue
When running using a discovery server, and trying to get the topic list
'''
FASTRTPS_DEFAULT_PROFILES_FILE=./super_client_configuration_file.xml ros2 topic list --no-daemon
'''

Topics are not always visible.

This occurs quite often, it also occurs when new nodes are run. I am not sure if the issues are related, but it seems to be 2 instances of the same problem.

When using Wireshark. I can see a very high number of ICMP packets which report Destination unreachable.

Any suggestions?

EduPonz, 2021-10-21:
I think some extra information may be useful:
* How do you run the container?
* Which command do you use to run the server?
* Are you trying to communicate the container and the host, or is it all communication expected to happen within the container itself?
* How do the network interfaces look like in both host and container?
* Could you attach some Wireshark capture with the ICMP packets?

Thanks in advance!

MiguelCompany, 2024-09-10:
Since this issue:

1. Refers to ROS Galactic which is EOL
2. Can't be reproduced / the OP didn't provide the necessary information for more than 2 years

I think it is appropriate to close it.

Feel free to reopen if it is still an issue on a currently supported release, making sure to provide all the necessary information for reproduction.

closed 2024-09-10 by MiguelCompany