ros2/rmw_fastrtps #570: Communication between nodes fail for non-small image ros2 topics

iqedgarmg, 2021-11-14:
Hi everyone!

I make a basic python camera publisher/subscriber using ros2 galatic on a pair of Windows 10 machines located on the same local network. I change the middleware layer on both machines to fast-rtps using this tutorial: 

https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html

and this .xml file: 

`<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <participant profile_name="super_client_profile" is_default_profile="true">
            <rtps>
                <builtin>
                    <discovery_config>
                        <discoveryProtocol>SUPER_CLIENT</discoveryProtocol>
                        <discoveryServersList>
                            <RemoteServer prefix="44.53.00.5f.45.50.52.4f.53.49.4d.41">
                                <metatrafficUnicastLocatorList>
                                    <locator>
                                        <udpv4>
                                            <address>127.0.0.1</address>
                                            <port>11811</port>
                                        </udpv4>
                                    </locator>
                                </metatrafficUnicastLocatorList>
                            </RemoteServer>
                        </discoveryServersList>
                    </discovery_config>
                </builtin>
            </rtps>
        </participant>

	
        <publisher profile_name="/usb_camera/camera/rgb">
            <qos>
                <publishMode>
                    <kind>SYNCHRONOUS</kind>
                </publishMode>
            </qos>
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </publisher>	

        
        <subscriber profile_name="/usb_camera/camera/rgb">
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </subscriber>

    </profiles>
</dds>
`

Once configured, I can publish/subscribe topics between both machines using small-size data topics. Nevertheless, when I try to subscribe to the camera topic generated by my node, the callback did not work on the subscriber node.

Curiously, if I resize the published image to a small resolution (100x100 pixels) everything works fine. So, I think that this is a problem related to the topic data size.     

I tried with some of the solutions mentioned in this issue:  

https://github.com/ros2/rmw_fastrtps/issues/460

Nevertheless still at the moment I can't subscribe to the camera topic if the image is greater than 100x100 pixels. 

Thanks in advance for your help!

## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Binaries
- DDS implementation:
  - Fastrtps
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
- Install ros2 galatic from binaries according this tutorial: 
  https://docs.ros.org/en/galactic/Installation/Windows-Install-Binary.html
- Try with a camera publisher node on a machine, and a camera subscriber node on another machine using the same network.
- Enable fast-dss discovery using this tutorial: 
   https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html
- Set FASTRTPS_DEFAULT_PROFILES_FILE to the previously provided .xml file.
- Communication between subscribers/publishers between both machines should work fine, but when large images are employed  the communication to the topic is not reached by the subscriber. 

#### Expected behavior
- Machine with the subscriber node is able to read the image topic from the publisher node using any image resolution. 

#### Actual behavior
- Communication works only if a small-sized image is employed (100x100 pixels).
- For images greater than 140 pixels, the communication did not work.

#### Additional information
- Communication works fine for small-sized topics (strings, vectors etc.)
- I use the standard sensor_msgs::Image ros topic for the publisher and subscriber. 
----

MiguelCompany, 2021-11-15:
@iqedgarmg In order to reproduce your issue, it would be nice to have a minimal reproducer with the code you are running, as long as the QoS settings.

iqedgarmg, 2021-11-16:
Hi @MiguelCompany, thanks a lot for your quick response!. 

In this drive, you can download the publisher/subscriber node that I developed using OpenCV: 

[](https://drive.google.com/drive/folders/1p4GziL_vGGfIfFkao0-Z-lpc-YTRcodE?usp=sharing)

This package has two nodes:

**a) Publisher**
`ros2 run usb_camera usb_camera`

Publisher generates two topics: 

```
* /usb_camera/rgb: sensor_msgs::Image topic with the camera image.
* /usb_camera/sub: std_msgs::Int8 topic for testing.
```

**b) Subscriber**
`ros2 run usb_camera camera_subs`

If everything works fine, you should visualize an OpenCV window with the received image topic. 

This is the configuration that I run on both machines to achieve ros2 connection: 

**1.- Change ROS middleware to fastrtps** 

`set RMW_IMPLEMENTATION=rmw_fastrtps_cpp`

2.- Call a discovery server (machine that has the publisher)

`python C:\dev\ros2-windows\tools\fastdds\fastdds.py discovery -i 0`

**3.- Setup a global discovery server (IP of the machine that has the publisher)**

`set ROS_DISCOVERY_SERVER=127.0.0.1:11811`

**4.- Set FastRTPS profile**

**set FASTRTPS_DEFAULT_PROFILES_FILE=C:\ros_projects\ros_ws\super_client_configuration_file.xml** 

with this configuration file: 

```
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <participant profile_name="super_client_profile" is_default_profile="true">
            <rtps>
                <builtin>
                    <discovery_config>
                        <discoveryProtocol>SUPER_CLIENT</discoveryProtocol>
                        <discoveryServersList>
                            <RemoteServer prefix="44.53.00.5f.45.50.52.4f.53.49.4d.41">
                                <metatrafficUnicastLocatorList>
                                    <locator>
                                        <udpv4>
                                            <address>127.0.0.1</address>
                                            <port>11811</port>
                                        </udpv4>
                                    </locator>
                                </metatrafficUnicastLocatorList>
                            </RemoteServer>
                        </discoveryServersList>
                    </discovery_config>
                </builtin>
            </rtps>
        </participant>

	<!-- Publisher profile  -->
        <publisher profile_name="/usb_camera/rgb">
            <qos>
                <publishMode>
                    <kind>SYNCHRONOUS</kind>
                </publishMode>
            </qos>
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </publisher>	

        <!-- Subscriber profile -->
        <subscriber profile_name="/usb_camera/rgb">
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </subscriber>

    </profiles>
</dds>
```

In my case, the subscriber works fine if the image resolution is less than 100x100 pixels (as well as for any topic of small size), but didn't work for larger image resolutions (publisher continues working, but subscriber callback not work).

Please let me know if you require further information. Thanks again for your help!

iqedgarmg, 2021-11-19:
Hi @MiguelCompany, I finally found the way  to make the subscriber working. I write the solution if this is useful for someone:  

1.- Set QoS on Python publisher
I forgot to define the proper QoS configuration on the publisher and subscriber nodes. In Python, this can be made as follows: 

```
#Import QoS Profile for sensor data
from rclpy.qos import qos_profile_sensor_data  

#Add QoS configuration to publisher (host computer node)
self.cam_pub = self.create_publisher(Image, '/usb_camera/rgb', qos_profile=qos_profile_sensor_data)

#Add QoS configuration to subscriber (subscriber node on remote pc)
s]elf.camera_subscription = self.create_subscription(Image, '/usb_camera/rgb', self.camera_callback, qos_profile=qos_profile_sensor_data)
```

2.- Setup SYNCHRONOUS mode
I was running the nodes in asynchronous mode, by changing it to synchronous the topic speed improves considerably.
  
`set RMW_FASTRTPS_PUBLICATION_MODE=SYNCHRONOUS`

By developing this modifications (and following the instructions described in my previous response), now the publisher and subscriber are working using the original image size at 12 Hz.

hidmic, 2021-12-07:
@audrow I take that #571 is somehow related to this.