ros2/rmw_fastrtps #573: ignore_local_publications behavior irregularities

smnogar, 2022-01-04:
## Bug report

I am using the package [domain bridge](https://github.com/ros2/domain_bridge) to try and bridge topics across domains. One of its features is "[bidirectional mapping](https://github.com/ros2/domain_bridge/pull/39)" in that nodes on say domains 0 and 1 can both publish and subscribe to the same topic.  It relies on setting [ignore_local_publications=true](https://github.com/ros2/domain_bridge/blob/2dcb49c9a659e0b9c91147379608d8f9daed8a1e/src/domain_bridge/domain_bridge.cpp#L386) on the appropriate subscriber. In cyclone DDS, this results in the desired behavior, in the message is sent across the domains at the desired frequency. When using FastDDS, it results endless looping with publication rates > 1 kHz for a topic publishing at 1 Hz. 

**Required Info:**

- Operating System:
  - Ubuntu 20.04, macOS 
- Installation type:
  - galactic Ubuntu binaries, Mac source install
- Version or commit hash:
  - galactic
- DDS implementation:
  - FastDDS
- Client library (if applicable):
  - rclcpp
  - uses domain_bridge commit [2dcb49c9a659e0b9c91147379608d8f9daed8a1e])https://github.com/ros2/domain_bridge/commit/2dcb49c9a659e0b9c91147379608d8f9daed8a1e)

#### Steps to reproduce issue

```
mkdir src
cd src
git clone https://github.com/ros2/domain_bridge.git
git checkout 2dcb49c9a659e0b9c91147379608d8f9daed8a1e
cd ../..
colcon build
. install/local_install.bash
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
ros2 launch domain_bridge domain_bridge.launch.xml config:=$PWD/src/domain_bridge/examples/example_bridge_config.yaml from_domain:=3 to_domain:=2
```
In a new terminal

```
. install/local_install.bash
export ROS_DOMAIN_ID=3
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
ros2 run demo_nodes_cpp talker
```

In a third terminal

```
export ROS_DOMAIN_ID=2
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
ros2 topic hz /chatter
```

#### Expected behavior

Publish rate of 1 Hz

#### Actual behavior

Publish rate > 1 kHz

#### Additional information

I'm not sure if the current FastDDS ignore_local_publications is the desired behavior. It would be nice if FastDDS could be used with domain bridge.

MiguelCompany, 2022-01-05:
@smnogar According to [the inline doc](https://github.com/ros2/rmw/blob/9f3c3e1b010a95013eb119ce2ae9b225bdfdc1ba/rmw/include/rmw/types.h#L158-L168) for `ignore_local_publications`, the definition of local is somewhat vague at the moment.

In the case of rmw_fastrtps local means *`published by a publisher in the same RMW context`*. I understand it is not enough for the use case of a domain bridge. In that case local should mean *`published by a publisher in the same process`*.

We could change the behavior to the second interpretation, but perhaps that would break other scenarios.
I think it might be a good idea to turn `ignore_local_publications` into an enumeration. This way the user could choose the desired behavior. I've added a point to the agenda of the next middleware WG to discuss this point.

In the mean time, you could check if changing [this line](https://github.com/ros2/rmw_fastrtps/blob/d535d3c90f193e73d12e4015831041e68f78ef10/rmw_fastrtps_shared_cpp/src/rmw_take.cpp#L96) to `if (sample_writer_guid.is_on_same_process_as(info->data_reader_->guid()) {` works as you expect.

smnogar, 2022-01-06:
@MiguelCompany Thanks for the response! I agree that it is a bit ambiguous what ignore_local_publications is supposed to do. I confirmed that your suggested change works the way I want.