ros2/rmw_fastrtps #576: how to change the shared memory segment size of fastrtps

minrui-hust, 2022-01-24:
I am using ros2-foxy,  and it seems foxy alreay use shared memory transport by default.
however, the default shared memory segment size is only 512k, which limits large msg transmission(sudden large drop rate when msg larger than 510K), how can I change the default shared memory segment size?

minrui-hust, 2022-01-24:
I use the FASTRTPS_DEFAULT_PROFILES_FIL env variable, in the profile.xml I set segment size to 1M, and default to true, but it seems /dev/shm create two profiles, one is 512k, another is 1M, but the node is still using 512k one.

EduPonz, 2022-01-24:
Hi @minrui-hust ,

How does your XML look like? Can it be that your not setting `useBuiltinTransports` to `false` when loading your custom transports?

minrui-hust, 2022-01-25:
thanks for your reply, indeed I am not setting useBuiltinTransports to false, after setting it, every thing went ok。the defaut segment size is too samll, transport large message like pointcloud will drop a lot. many people face this problem, and they dont know why,  can we give a warning if message excced the segment size?

EduPonz, 2022-01-25:
Hi @minrui-hust ,

There are several things to unpack here.

First, there are log messages related to your issues, although most of them are info messages which are not compiled by default ([see Logging](https://fast-dds.docs.eprosima.com/en/latest/fastdds/logging/intro.html)). I think you can find relevant logs by applying a [category filter](https://fast-dds.docs.eprosima.com/en/latest/fastdds/logging/logging_filtering.html#category-filtering) `(RTPS_TRANSPORT_SHM)|(RTPS_MSG_OUT)`.

The reason why there isn't an always-showing log message is that the fact that sending a sample using one transport fails does not mean sending it using another transport is going to fail as well. What I mean by that is the following: Imagine a situation where you have a sender with the default transports (UDP and SHM), and then you have a receiver with a UDP transport, and another one with only SHM. When the sender publishes a sample, the writer will send it over the two transports, which may have different outcomes. Logging that one of the sends failed at first try might be confusing to the user, specially when considering that the reliability mechanism will resend it as needed.

A main difference between UDP and SHM transports is that UDP transport blocks the send call until there is enough space in the socket for writing the data, whereas the SHM transport does not. This is because in the case of UDP, the OS will eventually consume the send buffer, but SHM transport relies on the receiver to consume it. If for whatever reason the receiver is not consuming the buffer (an unexpected crash for instance), the sender may block forever on a send call. Having the SHM transport blocking until there is space in the buffer may increase performance in your use case, but activating such feature would requiere deep knowledge of the implementation details for the users.

I'd like to point you the the [Large data use-case](https://fast-dds.docs.eprosima.com/en/latest/fastdds/use_cases/large_data/large_data.html#increasing-socket-buffers-size) from Fast DDS documentation, where it recommends to increase buffer sizes for large data samples. These values will be used to create the SHM segments.

I think the issue per-se can be closed, even if we keep an open discussion here.

minrui-hust, 2022-01-26:
thanks for your explanation, I will refer the Large data use-case for more details

closed 2022-01-26 by minrui-hust