ros2/rmw_fastrtps #579: How to enable LoanedMessages

fujitatomoya, 2022-02-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/4a36f319afcc208deae3c373493ef6d9feda1324
- DDS implementation:
  - Fast-DDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker_loaned_message
[INFO] [1644340760.069653573] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.
[INFO] [1644340760.069869940] [loaned_message_talker]: Publishing: '2.000000'
[INFO] [1644340760.078255330] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.
[INFO] [1644340760.078312705] [loaned_message_talker]: Publishing: 'Hello World: 2'
[INFO] [1644340761.069560102] [loaned_message_talker]: Publishing: '3.000000'
[INFO] [1644340761.069749556] [loaned_message_talker]: Publishing: 'Hello World: 3'
[INFO] [1644340762.069596311] [loaned_message_talker]: Publishing: '4.000000'
[INFO] [1644340762.069794361] [loaned_message_talker]: Publishing: 'Hello World: 4'
[INFO] [1644340763.069584865] [loaned_message_talker]: Publishing: '5.000000'
[INFO] [1644340763.069779696] [loaned_message_talker]: Publishing: 'Hello World: 5'
```

#### Expected behavior

`LoanedMessages` should be enabled.

#### Actual behavior

`LoanedMessages` cannot be enabled.

#### Additional information

- https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/datasharing.html, it seems that in default it should be enabled.
- documentation explains how to enable `LoanedMessages` via Fast-DDS.

fujitatomoya, 2022-02-08:
@MiguelCompany could you guide us what to do or where to see against this?
CC: @Barry-Xu-2018 @iuhilnehc-ynos

MiguelCompany, 2022-02-11:
@fujitatomoya After #568 is merged the only requirement for LoanedMessage to work would be for the data type to be plain.

To enable data-sharing on inter-process, having the following on the XML **and** setting `RMW_FASTRTPS_USE_QOS_FROM_XML=1` is required

```
<data_writer profile_name="topic_name">
    <qos>
        <data_sharing>
            <kind>AUTOMATIC</kind>
        </data_sharing>
    </qos>
</data_writer>

<data_reader profile_name="topic_name">
    <qos>
        <data_sharing>
            <kind>AUTOMATIC</kind>
        </data_sharing>
    </qos>
</data_reader>
```

fujitatomoya, 2022-02-11:
@MiguelCompany okay after https://github.com/ros2/rmw_fastrtps/pull/568, i will check if that works. and prepare document in rmw_fastsrtps as well.

fujitatomoya, 2022-03-22:
@MiguelCompany 

Just one thing to confirm,

> To enable data-sharing on inter-process, having the following on the XML and setting RMW_FASTRTPS_USE_QOS_FROM_XML=1 is required

you mean this is required currently, right?

and after this PR is merged, the above XML is NOT required but data type to be plain. my understanding is correct?

MiguelCompany, 2022-03-23:
@fujitatomoya The XML would still be required if the user wants to enable data-sharing. What #568 does is stop requiring data-sharing to enable the usage of loaned messages.

fujitatomoya, 2022-05-10:
> The XML would still be required if the user wants to enable data-sharing.

XML is required for `historyMemoryPolicy` element setting, see https://github.com/ros2/rmw_fastrtps/issues/596#issuecomment-1085852399.

the following can enable data sharing zero copy via `LoanedMessage`. (confirmed that it uses `/dev/shm`)

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">

  <!-- Default publisher profile -->
  <data_writer profile_name="default publisher profile" is_default_profile="true">
    <qos>
      <publishMode>
        <kind>SYNCHRONOUS</kind>
      </publishMode>
      <data_sharing>
        <kind>AUTOMATIC</kind>
      </data_sharing>
    </qos>
    <historyMemoryPolicy>DYNAMIC</historyMemoryPolicy>
  </data_writer>

  <data_reader profile_name="default subscription profile" is_default_profile="true">
    <qos>
      <data_sharing>
        <kind>AUTOMATIC</kind>
      </data_sharing>
    </qos>
    <historyMemoryPolicy>DYNAMIC</historyMemoryPolicy>
  </data_reader>
</profiles>
```

MiguelCompany, 2023-01-15:
@fujitatomoya Now that #568 is merged. I think we can close this one.

fujitatomoya, 2023-01-16:
@MiguelCompany i will try demos to make sure that works fine. Something i want to make sure is https://github.com/ros2/rmw_fastrtps/issues/579#issuecomment-1122709868. anyway i will build the current rolling and get back to here, and close this issue.

fujitatomoya, 2023-01-17:
now requirement for enabling zero copy data sharing is only PoD via rmw_fastrtps.

- `int` does not print any warning for `LoanedMessage`, that means application can borrow the memory from rmw_fastrtps.
- `string` is unbound message type, so that heap memory allocation happens in application.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker_loaned_message 
[INFO] [1673991712.046884487] [loaned_message_talker]: Publishing: '1.000000'
[INFO] [1673991712.047764190] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.
[INFO] [1673991712.047897265] [loaned_message_talker]: Publishing: 'Hello World: 1'
[INFO] [1673991713.046792547] [loaned_message_talker]: Publishing: '2.000000'
[INFO] [1673991713.047280114] [loaned_message_talker]: Publishing: 'Hello World: 2'
[INFO] [1673991714.046767677] [loaned_message_talker]: Publishing: '3.000000'
[INFO] [1673991714.047194316] [loaned_message_talker]: Publishing: 'Hello World: 3'
[INFO] [1673991715.046723828] [loaned_message_talker]: Publishing: '4.000000'
[INFO] [1673991715.047098620] [loaned_message_talker]: Publishing: 'Hello World: 4'
...
```

closed 2023-01-17 by fujitatomoya