ros2/rmw_fastrtps #582: Segmentation fault in in rmw_fastrtps layer when constructing an rclcpp::Node

AegisField, 2022-02-11:
## Bug report

**Required Info:**
- Operating System:
Ubuntu 20.04
- Installation type:
  - package manager way ie.:
  - `sudo apt install ros-foxy-desktop`
- Version or commit hash:
  - foxy
- DDS implementation:
  - Eprimosa's Fast-RTPS DDS impl.
- Client library (if applicable):
  - rclcpp

#### Problem Description
Hello Everyone!
We're having a segfault within rclcpp::Node constructor, and further investigation /callstack lead us here. I'll post the callstack we could extract below the minimal example... 
Do note please, that due to the nature of the issue i doubt it'll resurface just like that upon the source code's copy-paste.

#### Steps to reproduce issue

Below code is the truly minimal reproduction
CMake commands
``
``` Source code reduced ``` -->
```
#include <rclcpp/rclcpp.hpp>

int main(int argc, char** argv)
{
    rclcpp::init(argc, argv);
    rclcpp::Node some_node("TEST_SEGFAULT"); ///< [1]*
    rclcpp::shutdown();
    return 0; ///< note we aren't even executing any code below this anyway

    /*
     * Commented out code ///< This is relevant [2]*
     */
}
```
**[1]** Causes many things including:
- the segfault if i leave the codebase as is ( no commenting out )
-  If I comment the _WHOLE_ code at **[2]** out i can observe this error thrown from within Node(string) constructor:

#### Expected behavior

#### Actual behavior
Callstack of the segfault:
[callstack.txt](https://github.com/ros2/rmw_fastrtps/files/8050512/callstack.txt)

fujitatomoya, 2022-02-11:
> We aren't using ROS2 via workspaces

can you use ROS 2 colcon workspace to see if this problem is reproducible? at least, i do not meet anything like this.

AegisField, 2022-02-14:
We can certainly try it with workspaces but 1) it wouldn't solve our problem in the long run, 2) Looking at the weird behavior with the commented out "dead" code after the return 0; I wouldn't expect this to be reproducible for anyone just by copy-pasting the ros code snippet.  

One more weird detail: In the beginning of January this worked without an issue. Now I even went back to that exact commit in our repository and that fails with the above too. The ros2 installation is newer, and maybe some 3rdparties too due to auto-update of Linux.

So if you don't mind, while also trying it out with colcon, internally we'll go in this 3rdparty interference debug way, and tell you what solution we arrived at, so others - if any - will have a silver lining if they encounter this.

AegisField, 2022-02-17:
Hello there again!
Soo we've solved the issue on our end when going off on this 3rdparty clash tangent. 
**TL;DR:**
The solution was to switch the RMW implementation (chose cyclone dds for this instance but rly anything would've been fine):
```sudo apt install ros-foxy-rmw-cyclonedds-cpp
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp```

**The issue (we think*) **
1) The main difference between our 2 project (the crash vs the working) was the first few layers of how the inner classes are linked together into a coherent system. In the crashing project we did use asio and we build that by source ourselves.

2) Now if you look at [debug_deadlocked.txt](https://github.com/ros2/rmw_fastrtps/files/8050608/debug_deadlocked.txt) FastDDS impl actually deadlocked in asio's random mutex, which was weird, but it definitely showed us it is linked by that RMW impl. 

3) Since we've had issues with asio multilinkage in the past we went ahead and tried changing the rmw impl and voala~ works like a charm. Not a single error randomly thrown in a lower layer call, let alone a segfault.

*The reason i wrote 'we think', is because I didn't trace ros calls through the whole stack trace + hidden symbols, we had no idea how & what libs ros should find anyway. The deadlock showed us enough to reasonable try this solution, but the answer might not be as helpful like this.

OffsetMOSFET, 2023-07-18:
I would like to add I was getting the same issue with Ros Humble on Ubuntu 22.04. I have a similar setup working outside of colcon workspace (I have to use scons), though I was using rclc. On the rare case it was able to create a node, I could not find any publishers or subscribers my software created via the terminal, though the software could subscribe to topics itself created.

Switching to cyclone-dds seems to have addressed my issues as well.