ros2/rmw_fastrtps #593: Unpredictable behavior on machines that have multiple NICs

andrewbest-tri, 2022-03-22:
## Bug report

**Required Info:**

- Operating System:
Windows and linux 20.04 (Foxy)
- Installation type:
Apt on linux, aka.ms on windows

- Version or commit hash:
Not sure, binary install.
- DDS implementation:
FASTRTPS
- Client library (if applicable):
rclcpp and rclpy

#### Steps to reproduce issue

I'll try to do my best to be concise: behavior for ros2 is unpredictable when attempting to use a fastrtps profiles file to control traffic.
I can split to multiple issues if necessary. All these tests can be observed using the following test:

open three terminals. 

Terminal 1: 
Set the environment variable for the profiles file. 
Play a ros bag

Terminal 2:
Set the environment variable for the profiles file. 
ros2 topic list

Terminal 3
DO NOT set the profiles file
ros2 topic list
ros2 topic echo a topic from the bag

Symptom 1: Discovery server is sticky
Terminal 3 in the above test will see all the topics listed in ros2 topic list. However, it will not see any messages from the bag. Without the profiles file... it gets no traffic. But, it doesn't know this so it plods along happily getting no data at all.

We're losing experiment data to this problem. We don't even realize traffic isn't appearing.

Symptom 2: There appears to be no way to limit traffic without affecting anything else
We want to have EXACTLY the default transport behavior, except prevent the nodes from spamming themselves over both NICs. This doesn't seem possible. Here is our config (trying to be as close as we can to what we think is default... but there is no default XML).
```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
  <transport_descriptors>
    <transport_descriptor>
      <transport_id>udp_transport</transport_id>
      <type>UDPv4</type>
      <non_blocking_send>true</non_blocking_send>
      <interfaceWhiteList>
        <address>127.0.0.1</address> <!-- loopback -->
        <address>192.168.10.18</address>
        <address>192.168.10.20</address>
        <address>192.168.10.22</address>
      	<address>192.168.10.24</address> 
        <address>192.168.10.26</address>
        <address>172.17.0.1</address>  <!-- Docker -->
      </interfaceWhiteList>
    </transport_descriptor>
    <transport_descriptor>
        <transport_id>shm_transport</transport_id>
        <type>SHM</type>
    </transport_descriptor>
  </transport_descriptors>
  <participant profile_name="UDPParticipant" is_default_profile="true">
    <rtps>
      <name>profile_for_ros2_context</name>
      <userTransports>
        <transport_id>udp_transport</transport_id>
        <transport_id>shm_transport</transport_id>
      </userTransports>
      <useBuiltinTransports>false</useBuiltinTransports>
    </rtps>
  </participant>
</profiles>
```

Symptom 3: Not using FASTRTPS profile causes spam and invalid data.
We use machines on a closed network to communicate with our robot. Some of those machines connect to the internet on a second interface. If we do not use fastrtps profiles, messages will send over both network cards... and late/duplicated messages will mess up our system performance. Ideally we could just say "Only use this NIC" and not need a complex configuration with transports. 

Symptom 4: Multiple users on one machine cannot see one-anothers traffic
If we run a process as a different user, even using the profiles file, we cannot see any traffic between users. This does not occur without a profiles file. It also occurs when we don't enable shared memory transport... that was a recent debugging attempt.

© 2022 GitHub, Inc.
[Help](https://docs.github.com/enterprise/3.0)
[Support](https://github.shared-services.aws.tri.global/contact)[](https://github.shared-services.aws.tri.global/)
[API](https://docs.github.com/enterprise/3.0)

EduPonz, 2022-03-23:
Hi @andrewbest-tri,

First of all, thank you for your report. I have some questions and comments.

### Comments

> Discovery server is sticky

From what I see in your commands and configuration, you are not using [Fast DDS Discovery Server](https://fast-dds.docs.eprosima.com/en/v2.1.0/fastdds/discovery/discovery_server.html), am I correct?

> We want to have EXACTLY the default transport behavior, except prevent the nodes from spamming themselves over both NICs.

You almost nailed it. By default, [non_blocking_send](https://fast-dds.docs.eprosima.com/en/v2.1.0/fastdds/transport/udp/udp.html#udptransportdescriptor) is set to `false`, although it's a minor detail. Just make sure that you want it set to `true` before setting it.

> If we do not use fastrtps profiles, messages will send over both network cards

This is by design. Not doing so would mean that Fast DDS would need to make assumptions about packet routing based on subnet masks, which may result on no data reception depending on the user configuration. Furthermore, it is a common case in critical applications to have several interfaces on the same LAN so that there is redundancy in case some of the communication channels become unavailable for whatever reason.

> Multiple users on one machine cannot see one-anothers traffic

This can indeed be the case when using SHM transport, depending on the different permissions that the users have on the shared memory files. Is any of these users `root` by any chance? Disabling SHM transport should indeed get rid of this issue. How are you attempting to do so?

### Suggestions

1. Could you post the output of `apt show ros-foxy-fastrtps`. Fast DDS changed version in Foxy from 2.0 to 2.1, so it'd be useful to know which one you're using.
2. Would it be possible for you to build ROS 2 Foxy from sources and check whether the issue persists? We are planning on a Fast DDS v2.1.2 release in the upcoming weeks which will make its way into ROS 2 Foxy's next sync.
3. Alternatively to 2. you could build Fast DDS branch 2.1.x in a separate colcon workspace and the substitute the compiled library `libfastrtps.so` in `/opt/ros/foxy/lib` with the one from your workspace (I'd keep a backup of the original to be able to restore it). If you follow this approach, make sure to build Fast DDS with `-DSECURITY=ON`.
4. Does this issue manifest with a simpler example like the ROS 2 talker-listener demo? It'd be easier to reproduce it we do not depend on rosbag.
5. Can you post the exact commands that your are running since you open a new terminal (including setting the env variables). Also, is it possible that you have some extra environment configuration somewhere (like `~/.bashrc` for instance).

andrewbest-tri, 2022-03-24:
> From what I see in your commands and configuration, you are not using [Fast DDS Discovery Server](https://fast-dds.docs.eprosima.com/en/v2.1.0/fastdds/discovery/discovery_server.html), am I correct?

I am not sure. We did not do anything specific other than the profiles file, so whatever is the default behavior is what we are doing.

> You almost nailed it. By default, [non_blocking_send](https://fast-dds.docs.eprosima.com/en/v2.1.0/fastdds/transport/udp/udp.html#udptransportdescriptor) is set to false, although it's a minor detail. Just make sure that you want it set to true before setting it.

Is this a potential symptom of the issue? Should it affect behavior. Just to verify, I have set this to true and replicated my issue.

> This can indeed be the case when using SHM transport, depending on the different permissions that the users have on the shared memory files. Is any of these users root by any chance? Disabling SHM transport should indeed get rid of this issue. How are you attempting to do so?

I have tried that now. It still does not work. Sourcing the configuration I pasted above with the following changes still does not work:
1) non-blocking send = False
2) no shm transport

> Could you post the output of apt show ros-foxy-fastrtps. Fast DDS changed version in Foxy from 2.0 to 2.1, so it'd be useful to know which one you're using.
```
Package: ros-foxy-fastrtps
Version: 2.1.1-1focal.20211014.171109
Priority: optional
Section: misc
Maintainer: Steven! Ragnarök <stevenragnarok@osrfoundation.org>
Installed-Size: 11.3 MB
Depends: libc6 (>= 2.29), libgcc-s1 (>= 3.0), libssl1.1 (>= 1.1.1), libstdc++6 (>= 9), libtinyxml2-6a (>= 5.0.0), ros-foxy-fastcdr, libssl-dev, libtinyxml2-dev, ros-foxy-foonathan-memory-vendor, ros-foxy-ros-workspace
Download-Size: 2,533 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
Description: Implementation of RTPS standard.
```

> Would it be possible for you to build ROS 2 Foxy from sources and check whether the issue persists? We are planning on a Fast DDS v2.1.2 release in the upcoming weeks which will make its way into ROS 2 Foxy's next sync.

That's a pretty big change. I can attempt to do so on a local machine next week, but our eco-system is 8 machines currently, so that's not a solution I am able to test on our deployment.

>4 Des this issue manifest with a simpler example like the ROS 2 talker-listener demo? It'd be easier to reproduce it we do not depend on rosbag. 

Yes, I have replicated it using 
```
 ros2 run examples_rclpy_minimal_publisher publisher_member_function
```
in terminal 1.

Terminal 2 will list AND echo the topic.
Terminal 3 will list the topic but fail to echo (no data and no error or warning)
A forth terminal on the local machine as a different user will also see the topic in list but get no echo data

> Can you post the exact commands that your are running since you open a new terminal (including setting the env variables). Also, is it possible that you have some extra environment configuration somewhere (like ~/.bashrc for instance).

Sure, here is a stripped down config stored in my home directory
```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
  <transport_descriptors>
    <transport_descriptor>
      <transport_id>udp_transport</transport_id>
      <type>UDPv4</type>
      <non_blocking_send>false</non_blocking_send>
      <interfaceWhiteList>
        <address>127.0.0.1</address> <!-- loopback -->
        <address>192.168.10.18</address> 
        <address>192.168.10.20</address> 
        <address>192.168.10.22</address>
        <address>192.168.10.24</address>
        <address>192.168.10.26</address>
        <address>172.17.0.1</address>  <!-- Docker -->
      </interfaceWhiteList>
    </transport_descriptor>
  </transport_descriptors>
  <participant profile_name="UDPParticipant" is_default_profile="true">
    <rtps>
      <name>profile_for_ros2_context</name>
      <userTransports>
        <transport_id>udp_transport</transport_id>
      </userTransports>
      <useBuiltinTransports>false</useBuiltinTransports>
    </rtps>
  </participant>
</profiles>
```
Terminal 1:
```
source /opt/ros/foxy/setup.bash
export FASTRTPS_DEFAULT_PROFILES_FILE=/home/me/fastrtps_config.xml
ros2 run examples_rclpy_minimal_publisher publisher_member_function
```
Terminal 2:
```
source /opt/ros/foxy/setup.bash
export FASTRTPS_DEFAULT_PROFILES_FILE=/home/me/fastrtps_config.xml
ros2 topic list   # (shows /topic as a channel)
ros2 topic echo /topic # (will generate output along the lines of "data: 'hello world: n')
```
Terminal 3:
```
source /opt/ros/foxy/setup.bash
ros2 topic list   # (shows /topic as a channel)
ros2 topic echo /topic # (will generate no warning. No output. Nothing at all.)
```
Terminal 4 (as a different user with permissions on the fast rtps file)
```
source /opt/ros/foxy/setup.bash
export FASTRTPS_DEFAULT_PROFILES_FILE=/home/me/fastrtps_config.xml
ros2 topic list   # (shows /topic as a channel)
ros2 topic echo /topic # (will generate no warning. No output. Nothing at all.)
```

I'm no bashrc expert but I don't have anything in it that looks non-default.

MiguelCompany, 2022-05-25:
Hi @andrewbest-tri

I am trying to reproduce this and would like to have additional information.
I see that you are configuring a long list of whitelist interfaces.
Are all those interfaces present on the machine?
Would you mind sharing the output of `ifconfig` on the machine where this is reproduced?

MiguelCompany, 2022-05-26:
@andrewbest-tri @clalancette 

Using just a demo_nodes talker and listener, I've been able to reproduce the issue regarding profiles vs non-profiles communication. It is related to the interfaces being whitelisted and happens independently of the users running the talker and listener.

This is the output of `ifconfig` on my system:
```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.65.3  netmask 255.255.255.0  broadcast 192.168.65.255
        inet6 fe80::50:ff:fe00:1  prefixlen 64  scopeid 0x20<link>
        ether 02:50:00:00:00:01  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 329  bytes 92182 (92.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 3618  bytes 772556 (772.5 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3618  bytes 772556 (772.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

On one terminal I run the following commands (from the [installation guide](https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html#try-some-examples)):
```
source /opt/ros/foxy/setup.bash
ros2 run demo_nodes_py listener
```

On another terminal I run the following commands
```
source /opt/ros/foxy/setup.bash
export FASTRTPS_DEFAULT_PROFILES_FILE=/home/test/profiles.xml
ros2 run demo_nodes_cpp talker
```

Contents of my `profiles.xml`:
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
  <transport_descriptors>
    <transport_descriptor>
      <transport_id>udp_transport</transport_id>
      <type>UDPv4</type>

      <!-- Different cases checked here (see below) -->
      <interfaceWhiteList>
        <address>127.0.0.1</address>
        <address>192.168.65.3</address>
      </interfaceWhiteList>
    </transport_descriptor>
  </transport_descriptors>
  <participant profile_name="UDPParticipant" is_default_profile="true">
    <rtps>
      <name>profile_for_ros2_context</name>
      <userTransports>
        <transport_id>udp_transport</transport_id>
      </userTransports>
      <useBuiltinTransports>false</useBuiltinTransports>
    </rtps>
  </participant>
</profiles>
```

Without stopping the default configured listener on the first terminal, I checked several scenarios on the other one (i.e. stop the talker, edit the XML, and start the talker again)

| whitelisted interfaces | communication |
|-|-|
| 127.0.0.1 | no |
| 192.168.65.3 | no |
| 127.0.0.1 <br> 192.168.65.3 | yes |

Conclusion: if you want to communicate using a single interface, you should also add localhost to the whitelist.

### Additional notes

> Discovery server is sticky

I would better say that *ROS 2 daemon* is sticky. When `ros2 topic list` is executed, it starts a ros2 daemon. Changing the XML and executing `ros2 topic list` again will not update the configuration on the participant used by the daemon. In order to ensure the most recent XML configuration is used, run `ros2 daemon stop` after updating the XML.

> If we run a process as a different user, even using the profiles file, we cannot see any traffic between users

On my tests, using the same XML file on different users always worked. A complete table of the tested scenarios follow

| whitelisted interfaces | XML for listener <br> running on test | XML for talker <br> running on root | communication |
|-|-|-|-|
| N/A | no | no | no |
| 127.0.0.1 | yes | no | no |
| 127.0.0.1 | no | yes | no |
| 127.0.0.1 | yes | yes | yes |
| 192.168.65.3 | yes | no | no |
| 192.168.65.3 | no | yes | no |
| 192.168.65.3 | yes | yes | yes |
| 127.0.0.1 <br> 192.168.65.3 | yes | no | yes |
| 127.0.0.1 <br> 192.168.65.3 | no | yes | yes |
| 127.0.0.1 <br> 192.168.65.3 | yes | yes | yes |

andrewbest-tri, 2022-05-26:
Getting back to this. These results seem a little surprising to me. Our loopback interface is in our whitelist.

> I see that you are configuring a long list of whitelist interfaces.
This is true. This is deployed software, so it needs to have a config file all machines can load. You are seeing the result of forcing the entire network ecosystem to load this file.

> On my tests, using the same XML file on different users always worked. A complete table of the tested scenarios follow
This is an interesting result. I am not sure I understand the first row? No XML deployment should have given you a "yes" for communication?

Still encouraging.  Our results are inconsistent w.r.t to different users , *symptom 4* but the rest tracks. My tests without shared memory transport (in my old xml) still show issues.

Can we investigate more on *symptom 1*, specifically that discovery is sticky but communication isn't. 
Here is the issue @MiguelCompany , if you were to try to ros2 topic echo the publisher's topic while your XML is misconfigured, can you verify that you get the (expected) "This topic is not yet published". What will likely happen, as happens to me, is your node will subscribe, get a qos policy, then never get any data. So the traffic will be invisibly discarded despite the discovery mechanisms appearing to work. This is a critical problem because there isn't any way to know that transport is actually not working.

It would still also be helpful to understand if we can produce a configuration which is EXACTLY the default behavior aside from the whitelist. Is this possible? Its still onerous to need a complex configuration file that may or may not be polluting other aspects of the transport when the ONLY thing we want to do is prevent a second NIC from being used.

MiguelCompany, 2022-05-27:
> a config file all machines can load

OK. That makes sense.

> I am not sure I understand the first row? No XML deployment should have given you a "yes" for communication?

Running the talker as `root` and the listener as `test` does not work without configuration because they try to communicate over the shared memory transport, which does not work due to permissions on the memory-mapped files.

> a configuration which is EXACTLY the default behavior aside from the whitelist

It would be the following one. Bear in mind that the default behavior is the one I have previously described, and communication between different users may not work.

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >

  <!-- Descriptors for the two default transports -->
  <transport_descriptors>

    <!-- Shared memory transport -->
    <transport_descriptor>
      <transport_id>shm_transport</transport_id>
      <type>SHM</type>
    </transport_descriptor>

    <!-- UDP transport with whitelisted interfaces -->
    <transport_descriptor>
      <transport_id>udp_transport</transport_id>
      <type>UDPv4</type>
      <interfaceWhiteList>
        <!-- Your address list here -->
        <address>127.0.0.1</address>
        <address>192.168.65.3</address>
      </interfaceWhiteList>
    </transport_descriptor>

  </transport_descriptors>

  <!-- Participant configuration. ROS 2 needs is_default_profile="true" -->
  <participant profile_name="nic_whitelist_participant" is_default_profile="true">
    <rtps>
      <!-- We are overriding the default transports, so we discard the builtin ones -->
      <useBuiltinTransports>false</useBuiltinTransports>

      <!-- Include the two transports declared above -->
      <userTransports>
        <transport_id>udp_transport</transport_id>

        <!-- Comment the following line if you need to ensure cross-user communication -->
        <transport_id>shm_transport</transport_id>
      </userTransports>
    </rtps>
  </participant>
</profiles>
```

> Can we investigate more on _symptom 1_, specifically that discovery is sticky but communication isn't.

On my tests, running discovery commands with `--no-daemon` is always consistent with communication behavior.

clalancette, 2022-06-06:
The problem in general with `--no-daemon` is that we quickly spin up a node, gather some data from it, then shut it down and print the result.  It's entirely possible (and frequently the case) that we won't have gotten all of the discovery data by the time we shutdown the node, and hence be missing pieces of the graph.  You could add some kind of timeout there, but it is unclear how long to wait around.  Hence the daemon, which is a persistent node which participates in graph discovery with the rest of the network.  Then (when not using `--no-daemon`) the tools query the daemon for the graph information.

MiguelCompany, 2022-06-23:
@andrewbest-tri I would just recommend to perform `ros2 daemon stop` whenever the XML configuration is changed.

This will make the next ros2 discovery-related command to start a new daemon using the new configuration.