ros2/rmw_fastrtps #597: Assertion max_num_payloads failed.

fujitatomoya, 2022-04-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/f8d68a8194b546f1cef8bb0b84629f2a5a314d1a
- DDS implementation:
  - rmw_fastrtps
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

- terminal-1

```bash
# ros2 run demo_nodes_cpp add_two_ints_server
```

- terminal-2

```bash
# cat test.sh
#!/bin/bash

for i in {1..15}
do
    ros2 run demo_nodes_cpp add_two_ints_client &
done

# ./test.sh
```

#### Expected behavior

test.sh succeeds w/o error

#### Actual behavior

assertion failture

```bash
add_two_ints_client: /root/ros2_ws/colcon_ws/src/eProsima/Fast-DDS/src/cpp/rtps/history/TopicPayloadPool.cpp:263: bool eprosima::fastrtps::rtps::TopicPayloadPool::shrink(uint32_t): Assertion `payload_pool_allocated_size() - payload_pool_available_size() <= max_num_payloads' failed.
[ros2run]: Aborted
```

#### Additional information

core stack trace with debug build.

<details>
<summary>gdb ./build/demo_nodes_cpp/add_two_ints_client core</summary>

```bash
# gdb ./build/demo_nodes_cpp/add_two_ints_client core
GNU gdb (Ubuntu 12.0.90-0ubuntu1) 12.0.90
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
[... 133 lines cut ...]
    __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:705
#41 0x000055bef2c053da in std::__shared_ptr<rclcpp::node_interfaces::NodeBaseInterface, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x55bef49d7a68, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#42 0x000055bef2c05424 in std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>::~shared_ptr (this=0x55bef49d7a68,
    __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr.h:122
#43 0x00007f4fb6350f46 in rclcpp::Node::~Node (this=0x55bef49d7a50, __in_chrg=<optimized out>)
    at /root/ros2_ws/colcon_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:279
#44 0x000055bef2c13e89 in __gnu_cxx::new_allocator<rclcpp::Node>::destroy<rclcpp::Node> (this=0x55bef49d7a50,
    __p=0x55bef49d7a50) at /usr/include/c++/11/ext/new_allocator.h:168
#45 0x000055bef2c13495 in std::allocator_traits<std::allocator<rclcpp::Node> >::destroy<rclcpp::Node> (__a=...,
    __p=0x55bef49d7a50) at /usr/include/c++/11/bits/alloc_traits.h:535
--Type <RET> for more, q to quit, c to continue without paging--
#46 0x000055bef2c126e9 in std::_Sp_counted_ptr_inplace<rclcpp::Node, std::allocator<rclcpp::Node>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x55bef49d7a40) at /usr/include/c++/11/bits/shared_ptr_base.h:528
#47 0x000055bef2c0777b in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x55bef49d7a40)
    at /usr/include/c++/11/bits/shared_ptr_base.h:168
#48 0x000055bef2c05c39 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7ffd77311c18,
    __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:705
#49 0x000055bef2c058dc in std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7ffd77311c10,
    __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#50 0x000055bef2c05926 in std::shared_ptr<rclcpp::Node>::~shared_ptr (this=0x7ffd77311c10, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr.h:122
#51 0x000055bef2c03a0b in main (argc=1, argv=0x7ffd77312778)
    at /root/ros2_ws/colcon_ws/src/ros2/demos/demo_nodes_cpp/src/services/add_two_ints_client.cpp:75
```

</details>

fujitatomoya, 2022-04-13:
@MiguelCompany i am not sure if this is bug, could you take a look?
CC: @Barry-Xu-2018 @iuhilnehc-ynos

MiguelCompany, 2022-04-19:
@fujitatomoya Sorry for the late answer (we were on Easter holidays here). This seems to have been previously reported at eProsima/Fast-DDS#2533, but we didn't have time to check it yet.

We will reproduce and debug it this week.

clalancette, 2022-05-05:
@MiguelCompany Any thoughts on what might be going on here?

MiguelCompany, 2022-05-05:
> Any thoughts on what might be going on here?

Not much yet. I could reproduce it, but not in a deterministic way (so nothing that could be turned into a regression test). I've been debugging it a bit, but didn't get to a conclusion yet.

VladimirVassiliev, 2022-06-21:
@MiguelCompany Should the TopicPayloadPool::shrink() function be something like this?

```
bool TopicPayloadPool::shrink (
        uint32_t max_num_payloads)
{
    assert(max_num_payloads == 0 || payload_pool_allocated_size() - payload_pool_available_size() <= max_num_payloads);

    while (max_num_payloads < all_payloads_.size())
    {
        if (free_payloads_.empty())
        {
            // This may happen if max_num_payloads is zero.
            break;
        }

        PayloadNode* payload = free_payloads_.back();
        free_payloads_.pop_back();

        // Find data in allPayloads, remove element, then delete it
        all_payloads_.at(payload->data_index()) = all_payloads_.back();
        all_payloads_.back()->data_index(payload->data_index());
        all_payloads_.pop_back();
        delete payload;
    }

    return true;
}
```
I understand that zero is a valid value of max_num_payloads. Is it correct?

MiguelCompany, 2022-07-21:
@fujitatomoya @clalancette This should have been fixed by eProsima/Fast-DDS#2853, backport to `2.6.x` pending CI on eProsima/Fast-DDS#2861

fujitatomoya, 2022-07-28:
@MiguelCompany 

```bash
add_two_ints_client: /root/ros2_ws/colcon_ws/src/eProsima/Fast-DDS/src/cpp/rtps/history/TopicPayloadPool.cpp:263: bool eprosima::fastrtps::rtps::TopicPayloadPool::shrink(uint32_t): Assertion `payload_pool_allocated_size() - payload_pool_available_size() <= max_num_payloads' failed.
[ros2run]: Aborted
```

above problem has been solved, i cannot reproduce that assertion.

but, we can see `service not available` message instead.

```bash
# ./test.sh
root@tomoyafujita:~/ros2_ws/colcon_ws# [INFO] [1658999884.962398783] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999884.971446955] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999884.975172233] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999884.980538959] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999884.987139210] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999884.988729905] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999884.990040116] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999884.998537695] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999885.001958420] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999885.061093777] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999885.061093287] [add_two_ints_client]: Result of add_two_ints: 5
[INFO] [1658999885.974434720] [add_two_ints_client]: service not available, waiting again...
[INFO] [1658999885.976093427] [add_two_ints_client]: service not available, waiting again...
[INFO] [1658999885.979039142] [add_two_ints_client]: service not available, waiting again...
[INFO] [1658999885.985281537] [add_two_ints_client]: service not available, waiting again...
```

closed 2022-10-13 by fujitatomoya