ros2/rmw_fastrtps #607: [BUG]The performance degradation of shm-datasharing when msg size is over 2MB.

ZhenshengLee, 2022-05-20:
## Bug report

**Required Info:**

- Operating System: Ubuntu2004
- Installation type: from sources
- Version or commit hash: galactic-5.0.2
- DDS implementation: Fast-RTPS-2.3.4
- Client library (if applicable): rclcpp

#### Steps to reproduce issue

I test the performance of rmw_fastrtps_cpp(including udp(shm-default), intra, and zero-copy(shm-datasharing)) with performance_test suit from apex.ai
the new function is extened and hosted in https://github.com/ZhenshengLee/performance_test
the test results in hosted in https://zhenshenglee.github.io/ros2_jetson_benchmarks/benchmark_results/galactic

#### Expected behavior

the zero-copy performance is better than what with udp

for example https://zhenshenglee.github.io/ros2_jetson_benchmarks/benchmark_results/galactic/pc_ros2_galactic_v5.13.0-41-generic_amd64/rmw_compare/report_compare_transport__rmw_fastrtps_Array1m.html

#### Actual behavior

when msg size is over 2m, the latency of rmw_fastrtps_cpp is up to 10ms, which is unacceptable.

https://zhenshenglee.github.io/ros2_jetson_benchmarks/benchmark_results/galactic/pc_ros2_galactic_v5.13.0-41-generic_amd64/rmw_compare/report_compare_transport__rmw_fastrtps_Array2m.html

#### Additional information

there are other dds and rmw performance tests results running in pc and jetson xavier

MiguelCompany, 2022-05-20:
@ZhenshengLee Thank you for your efforts on using data-sharing with performance_test!

I've been digging into your repo, and I've seen several mistakes that I think imply you are not using data-sharing correctly:
* Looking at [this method](https://github.com/ZhenshengLee/performance_test/blob/ddb48614775116d8824de7acff7f3aca9d8efcfd/performance_report/performance_report/utils.py#L420), I see that you are not setting `RMW_FASTRTPS_USE_QOS_FROM_XML=1`. Sounds like a copy-paste issue on line 424.
* I see you are setting `historyMemoryPolicy` to `DYNAMIC`, which is incompatible with data-sharing. You should set it to `PREALLOCATED`

ZhenshengLee, 2022-05-20:
@MiguelCompany Thanks four your help!

> Looking at [this method](https://github.com/ZhenshengLee/performance_test/blob/ddb48614775116d8824de7acff7f3aca9d8efcfd/performance_report/performance_report/utils.py#L420), I see that you are not setting RMW_FASTRTPS_USE_QOS_FROM_XML=1. Sounds like a copy-paste issue on line 424.

I will fix the typoes

> I see you are setting historyMemoryPolicy to DYNAMIC, which is incompatible with data-sharing. You should set it to PREALLOCATED

The setting is from @fujitatomoya in https://github.com/ros2/rmw_fastrtps/issues/596#issuecomment-1122705910_

> I've been digging into your repo, and I've seen several mistakes that I think imply you are not using data-sharing correctly:

I will check the docs of fastdds and test it again.

MiguelCompany, 2022-05-20:
> I will check the docs of fastdds and test it again.

https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/datasharing.html#constraints

ZhenshengLee, 2022-05-20:
@MiguelCompany Thanks again. 

The new test results seem to be reasonable.

Closing this.

closed 2022-05-20 by ZhenshengLee