ros2/rmw_fastrtps #611: AsyncWriterThread takes 60% CPU time

micxiaoc, 2022-06-17:
## Bug report

**Required Info:**

- Operating System:
  - Linux
- Installation type:
  - foxy
- Version or commit hash:
  - not sure
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
When using rosbridge, asyncwriterthread will be generated after a few hours, which will occupy a high CPU and remain at 60%. Is there any way to reduce the CPU on foxy

#### Expected behavior
CPU can remain stable
#### Actual behavior
Asyncwriterthread raises CPU
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

EduPonz, 2022-07-08:
Hi,

Each Participant holds an `AsyncWriterThread` instance that is used to:

1. Send user data coming from asynchronous writers (as all ROS 2 writers are by default).
2. Take care of re-sendings in the case of reliable writers.
3. Take care of feeding data to late joiners which DurabilityQosPolicy is other than `VOLATILE`.

> asyncwriterthread will be generated after a few hours

Because of the previous, I'm not sure I understand what you mean with this; the async thread may de stalling but it should always exist.

Could you try to reproduce in 2.1.x?

clalancette, 2023-06-13:
No response in a year here, so closing this out.  If you are still having problems, please open a new issue.

closed 2023-06-13 by clalancette