ros2/rmw_fastrtps #613: Subscriber which is created with dedicated callback group at runtime not working

gezp, 2022-06-21:


EduPonz, 2022-06-27:
We are currently working on a solution for this (#619) which fixes this issue for us locally. It is still missing some work which will be ready this week.

clalancette, 2022-06-27:
> We are currently working on a solution for this (#619) which fixes this issue for us locally. It is still missing some work which will be ready this week.

Fantastic, thanks for the update.

afrixs, 2022-07-21:
Just to note, we have recently discussed a similar issue in navigation2 with a timer being registered to an executor from an external thread. [Conclusion](https://github.com/ros2/rmw_fastrtps/issues/606#issuecomment-1125824269) was that if a node is assigned to multiple executors running asynchronously, it is unable to notify all of them about creating/destroying subscriptions etc..

There is a note about violating RMW API by this behavior in the linked comment, stating that it is more of rclcpp problem (although making FastDDS support this would be really nice of course). There has already been [an issue about it](https://github.com/ros2/rclcpp/issues/1611) for a long time in rclcpp and even a [pull request](https://github.com/ros2/rclcpp/pull/1640) fixing the problem by using a separate guard condition per callback group instead of per node, but it hasn't been merged for some reason until now, which is a shame, since this has a strong impact on elegance of the system, causing so many head-aches.

Maybe @gezp or someone who uses rclcpp built from source, could you try to test [the example](https://github.com/gezp/ros2_test_subscriber) with [this patch](https://github.com/ros2/rclcpp/pull/1640) applied? I'm quite convinced it will work.

Another way to make the example work is to use a separate node for each asynchronous executor, i.e. applying [this diff](https://github.com/ros2/rmw_fastrtps/files/9159822/ros2_test_subscriber_diff.txt). However it's quite inelegant, introducing many unneeded nodes in the system.

MiguelCompany, 2022-07-21:
@afrixs 

> although making FastDDS support this would be really nice of course

That is what #619 is about. Would be great if someone could also give it a try.

SteveMacenski, 2022-07-21:
@afrixs I'll try to escalate https://github.com/ros2/rclcpp/pull/1640 - I agree, it probably just got buried

fujitatomoya, 2022-08-25:
@SteveMacenski @gezp I confirmed that this problem has been addressed by https://github.com/ros2/rmw_fastrtps/pull/619. please feel free to reopen if you still have this issue in rolling.

closed 2022-08-25 by fujitatomoya