ros2/rmw_fastrtps #615: ros2 node list doesn't work with containers

Aposhian, 2022-06-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 5.0.2-1focal.20220430.124017
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp and rclpy

#### Steps to reproduce issue

Here is a Dockerfile
```Dockerfile
FROM ros:galactic
RUN apt-get update && apt-get install -y \
    ros-$ROS_DISTRO-demo-nodes-cpp \
    ros-$ROS_DISTRO-rmw-fastrtps-cpp \
    ros-$ROS_DISTRO-rmw-cyclonedds-cpp
```
and a compose file
```yaml
services:
  talker:
    build: ./
    command: ros2 run demo_nodes_cpp talker
    network_mode: host
    ipc: host
    environment:
      - RMW_IMPLEMENTATION
      - ROS_DOMAIN_ID
```
Run the following test script:
```bash
#!/usr/bin/env bash

source /opt/ros/galactic/setup.bash

run_node_and_check() {
    export RMW_IMPLEMENTATION=$1
    ros2 daemon stop &> /dev/null
    docker-compose down &> /dev/null
    docker-compose up -d &> /dev/null
    echo "$RMW_IMPLEMENTATION node list:"
    ros2 node list
}

run_node_and_check rmw_cyclonedds_cpp

run_node_and_check rmw_fastrtps_cpp
```

#### Expected behavior
```
rmw_cyclonedds_cpp node list:
/talker
rmw_fastrtps_cpp node list:
/talker
```

#### Actual behavior
```
rmw_cyclonedds_cpp node list:
/talker
rmw_fastrtps_cpp node list:
```

Aposhian, 2022-06-22:
If I run `ros2 node list` in another container, then it works, on both `galactic` and `humble` images.
```yaml
services:
  talker:
    build: ./
    command: ros2 run demo_nodes_cpp talker
    network_mode: host
    ipc: host
    environment:
      - RMW_IMPLEMENTATION
      - ROS_DOMAIN_ID
  lister:
    build: ./
    command: ros2 node list
    network_mode: host
    ipc: host
    environment:
      - RMW_IMPLEMENTATION
      - ROS_DOMAIN_ID
    depends_on:
      - talker
```

jsantiago-eProsima, 2022-06-24:
@fujitatomoya We have actually managed to reproduce this issue following your steps. Are you running the host commands as the `root` user or as a regular user? 

This is a permissions issue. When using --ipc=host shared memory files for the container are being created on the host with root permissions so an unprivileged user will not be able to use them for communication. 

We are working on a way to detect and prevent this.

fujitatomoya, 2022-06-24:
> Are you running the host commands as the root user or as a regular user?

regular user.

> This is a permissions issue. When using --ipc=host shared memory files for the container are being created on the host with root permissions so an unprivileged user will not be able to use them for communication.

are you guys using rootless container runtime(or bind specific uid/gid for container)? if that so, then that makes sense.
I was thinking that just using `docker run` then `root` (same uid/gid/groups) user in the container. if we bind the same IPC namespace, it should be accessible to&from container.

fujitatomoya, 2022-08-29:
related to https://github.com/ros2/ros2/issues/1318, this is user id problem for the file system.

fujitatomoya, 2022-08-29:
@Aposhian I will close this, please reopen if you have any issues. i opened a issue for Fast-DDS doc, see https://github.com/eProsima/Fast-DDS-docs/issues/398, we can track doc enhancement on this issue.

closed 2022-08-29 by fujitatomoya