ros2/rmw_fastrtps #617: Static Discovery Support

Seiwert, 2022-06-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binary (Humble)
- Version or commit hash:
  - Humble
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Working on reproducing the [Static Hello World Examples](https://github.com/eProsima/Fast-DDS/tree/master/examples/C%2B%2B/StaticHelloWorldExample) but by utilizing the ROS 2 Talker/Listener nodes @agoeckner and I are running into a few issues. Notably the participants are showing up in the Fast DDS Monitor and as we anticipated the DataWriter and DataReader never propagate indicating that the Static EDP is enabled for both the talker and listener nodes. From what we can tell though, it seems as though the DataReader and DataWriter are never matching and we're not quite sure why. 

Here is the xml being set to FASTRTPS_DEFAULT_PROFILES_FILE for both the talker and listener node
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <participant profile_name="participant_profile_ros2" is_default_profile="true">
        <rtps>
            <name>profile_for_ros2_static_edp</name>
            <builtin>
                <discovery_config>
                    <EDP>STATIC</EDP>
                    <static_edp_xml_config>file:///home/adam/Downloads/demos-master/demo_nodes_cpp_static/resource/STATIC_EDP_COMBINED.xml</static_edp_xml_config>
                </discovery_config>
            </builtin>
        </rtps>
    </participant>

    <!-- Publisher profile for topic helloworld -->
    <data_writer profile_name="rt/chatter">
        <userDefinedID>1</userDefinedID>
    </data_writer>

    <!-- Request subscriber profile for services -->
    <data_reader profile_name="rt/chatter">
        <userDefinedID>2</userDefinedID>
    </data_reader>

</profiles>
```

And here is the Static EDP file
```xml
<staticdiscovery>
	<participant>
		<name>listener</name>
		<reader>
			<userId>2</userId>
            <topicName>rt/chatter</topicName>
			<topicDataType>std_msgs::msg::dds_::String_</topicDataType>
		</reader>
	</participant>
	<participant>
		<name>talker</name>
		<writer>
			<userId>1</userId>
            <topicName>rt/chatter</topicName>
			<topicDataType>std_msgs::msg::dds_::String_</topicDataType>
		</writer>
	</participant>
</staticdiscovery>

```
We've attempted a variety of things such as :
[... 58 lines cut ...]
[37;1m2022-06-23 18:36:45.257 [32;1m[[37;1mRTPS_WRITER[32;1m Info] [37mSending relevant changes as DATA/DATA_FRAG messages[34;1m -> Function [36madd_data[m
[37;1m2022-06-23 18:36:45.257 [32;1m[[37;1mRTPS_WRITER[32;1m Info] [37mSending INFO_TS message[34;1m -> Function [36madd_info_ts_in_buffer[m
[37;1m2022-06-23 18:36:45.257 [32;1m[[37;1mRTPS_MSG_OUT[32;1m Info] [37m(ID:140668864160960) SharedMemTransport: 1384 bytes to port 7412[34;1m -> Function [36msend[m
[37;1m2022-06-23 18:36:45.257 [32;1m[[37;1mRTPS_MSG_IN[32;1m Info] [37m(ID:140668805404224) InfoTS Submsg received, processing...[34;1m -> Function [36mprocessCDRMsg[m
```
Kind of at a loss at this point and looking for any insight. Also here is exactly how we're starting the nodes.

Terminal 1
```
source /opt/ros/humble/setup.bash
export FASTRTPS_DEFAULT_PROFILES_FILE=/home/adam/Downloads/demos-master/demo_nodes_cpp_static/resource/STATIC_FASTRTPS_PROFILE.xml
ros2 run demo_nodes_cpp talker
```
Terminal 2
```
source /opt/ros/humble/setup.bash
export FASTRTPS_DEFAULT_PROFILES_FILE=/home/adam/Downloads/demos-master/demo_nodes_cpp_static/resource/STATIC_FASTRTPS_PROFILE.xml
ros2 run demo_nodes_cpp listener
```

#### Expected behavior
Talker node runs and listener node outputs "hello world #"

#### Actual behavior
Talker node runs but the listener node never receives any publications

agoeckner, 2022-06-29:
I seem to have static EDP functioning with ROS2 now. The key is to set the userID field for all topics to `65535`. ROS2 must be setting this userId value on all newly created topics. Thus, we need to set that value in the static EDP configuration so that FastRTPS is able to match it.

See our configuration files below:

**Talker XML configuration**
```
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <log>
        <use_default>FALSE</use_default>
        <consumer>
            <class>FileConsumer</class>
            <property>
                <name>filename</name>
                <value>fastdds_log_datawriter.log</value>
            </property>
            <property>
                <name>append</name>
                <value>FALSE</value>
            </property>
        </consumer>
    </log>

    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <participant profile_name="HelloWorldPublisher" is_default_profile="true">
            <rtps>
                <name>HelloWorldPublisher</name>
                <builtin>
                    <discovery_config>
                        <EDP>STATIC</EDP>
                        <static_edp_xml_config>file://./HelloWorldCombined.xml</static_edp_xml_config>
                    </discovery_config>
                </builtin>
            </rtps>
        </participant>

        <!-- Publisher profile for topic /chatter -->
        <data_writer profile_name="rt/chatter">
            <userDefinedID>65535</userDefinedID>
        </data_writer>

    </profiles>
</dds>
```

**Listener XML configuration**
```
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <log>
        <use_default>FALSE</use_default>
        <consumer>
            <class>FileConsumer</class>
            <property>
                <name>filename</name>
                <value>fastdds_log_datareader.log</value>
            </property>
            <property>
                <name>append</name>
                <value>FALSE</value>
            </property>
        </consumer>
    </log>

    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <participant profile_name="HelloWorldSubscriber" is_default_profile="true">
            <rtps>
                <name>HelloWorldSubscriber</name>
                <builtin>
                    <discovery_config>
                        <EDP>STATIC</EDP>
                        <static_edp_xml_config>file://./HelloWorldCombined.xml</static_edp_xml_config>
                    </discovery_config>
                </builtin>
            </rtps>
        </participant>

        <!-- Subscriber profile for topic /chatter -->
        <data_reader profile_name="rt/chatter">
            <userDefinedID>65535</userDefinedID>
        </data_reader>

    </profiles>
</dds>
```

**Static EDP configuration file (common):**
```
<staticdiscovery>
	<participant>
		<name>HelloWorldPublisher</name>
		<writer>
			<userId>65535</userId>
			<topicName>rt/chatter</topicName>
			<topicDataType>std_msgs::msg::dds_::String_</topicDataType>
			<topicKind>NO_KEY</topicKind>
			<reliabilityQos>RELIABLE_RELIABILITY_QOS</reliabilityQos>
		</writer>
	</participant>
	<participant>
		<name>HelloWorldSubscriber</name>
		<reader>
			<userId>65535</userId>
			<topicName>rt/chatter</topicName>
			<topicDataType>std_msgs::msg::dds_::String_</topicDataType>
			<topicKind>NO_KEY</topicKind>
			<reliabilityQos>RELIABLE_RELIABILITY_QOS</reliabilityQos>
		</reader>
	</participant>
</staticdiscovery>
```

agoeckner, 2022-07-06:
We now have static discovery / static EDP working with ROS2.

The key discovery is that the main QoS profiles file takes ROS2 topic names as e.g. `/chatter`. However, the static EDP configuration file takes ROS2 topic names as `rt/chatter`.

I've attached the configuration files necessary to get the ROS2 talker/listener demo running with static EDP. Place those files in your working directory.

Then run the following commands for the talker:
```
source /opt/ros/humble/setup.bash
export FASTRTPS_DEFAULT_PROFILES_FILE="$(pwd)/STATIC_FASTRTPS_PROFILE_WRITER.xml"
ros2 run demo_nodes_cpp talker
```

And for the listener:
```
source /opt/ros/humble/setup.bash
export FASTRTPS_DEFAULT_PROFILES_FILE="$(pwd)/STATIC_FASTRTPS_PROFILE_READER.xml"
ros2 run demo_nodes_cpp listener
```

If I get time soon I'll try to create a ROS2 documentation page on setting up static EDP.

**[working_static_edp_config.zip](https://github.com/ros2/rmw_fastrtps/files/9056324/working_static_edp_config.zip)**

EduPonz, 2022-07-07:
Hi @agoeckner ,

I'm glad you made it work, and a ROS 2 tutorial would definitely help. You may consider adding a a use case in [Fast DDS' documentation](https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2.html#ros2).

On a side note, mind that the profile name specified in the XML would not always be `/chatter`; the table in [Creating publishers/subscriptions with different profiles](https://github.com/ros2/rmw_fastrtps#creating-publisherssubscriptions-with-different-profiles) summarizes the profile required depending on the Node's namespace.

closed 2022-07-07 by Seiwert

Syamkamalnadhc, 2025-01-20:
@agoeckner is it possible to configure static EDP for one publisher multiple subscribers, if so, what will be new additions in STATIC_FASTRTPS_PROFILE_WRITER and HelloworldSubscriber.xml's