ros2/rmw_fastrtps #618: CPU usage increase in v2.6.0 comparing with v2.2.0

Barry-Xu-2018, 2022-06-24:
Hardware platform is i.MX 6.   
Test program is the same and the used version of fastdds is v2.6.0 and v2.2.0   

One subscriber is on this hardware device.  There are other 5 devices which will be started and send message to this device.   
At subscriber side,  there are about 500 messages received immediately after matching.   
During this period,  
v2.2.0:   CPU usage (30% ~ 70%) will last for about 1 minute.  
v2.6.0:   CPU usage (28% ~ 72%) will last for about 3 minutes. 

While receiving messages from one device,   
v2.2.0: CPU usage is about 2.5%.  
v2.6.0: CPU usage is about 7.5%.  

----
For v2.6.1  
If receive 100 messages, CPU usage is about 20%.  
If receive 500 messages, CPU usage is about 50%.
If receive 1000 messages, CPU usage is about 100%.

After analysis, the function using much CPU is

https://github.com/eProsima/Fast-DDS/blob/master/src/cpp/fastdds/subscriber/DataReaderImpl.cpp#L659  

And

https://github.com/eProsima/Fast-DDS/blob/master/src/cpp/fastdds/subscriber/DataReaderImpl.cpp#L673 (It also calls above `lookup_instance`).   

Above information isn't enough. I will continue to supplement information.  

I am modifying DDS sample code to reproduce it and attach code.

fujitatomoya, 2022-06-24:
This seems to be a digression for performance aspect. this directly affects number of sender devices we can connect or support to the receiver as requirement of business logic.

MiguelCompany, 2022-06-27:
@Barry-Xu-2018 @fujitatomoya I think this can be closed in favor of eProsima/Fast-DDS#2779, since I think it will only affect topics with keys, which are not used in ROS 2.

Barry-Xu-2018, 2022-06-27:
@MiguelCompany 

> @Barry-Xu-2018 @fujitatomoya I think this can be closed in favor of [eProsima/Fast-DDS#2779](https://github.com/eProsima/Fast-DDS/issues/2779), since I think it will only affect topics with keys, which are not used in ROS 2.

Yes. I am wrong to create issue here and forgot to close after creating https://github.com/eProsima/Fast-DDS/issues/2779.

closed 2022-06-27 by Barry-Xu-2018