ros2/rmw_fastrtps #624: guard_condition argument is null on fastrtps only

Aposhian, 2022-06-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - `ros-galactic-rmw-fastrtps-cpp` @ 5.0.2-1focal.20220430.134850
  - `ros-galactic-fastrtps` @ 2.3.4-1focal.20220430.115156
- DDS implementation:
  - Fast RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Running https://github.com/FormantIO/ros2-adapter which is a fairly simple ROS2 node to subscribe to topics and forward them to a cloud service.

#### Expected behavior
`rclcpy.create_node` works just fine. It works on CycloneDDS.

#### Actual behavior (with FastRTPS only)
```
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:413'

with this new error message:

  'Unable to fini publisher for node., at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:378'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Unable to fini publisher for node., at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:378'

with this new error message:

  'node argument is null, at /tmp/binarydeb/ros-galactic-rmw-fastrtps-cpp-5.0.2/src/rmw_node.cpp:80'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'node argument is null, at /tmp/binarydeb/ros-galactic-rmw-fastrtps-cpp-5.0.2/src/rmw_node.cpp:80, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:384'

with this new error message:

  'guard_condition argument is null, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/guard_condition.c:126'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py:174: RuntimeWarning: Failed to fini node: guard_condition argument is null, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/guard_condition.c:126, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:389
  self.__node = _rclpy.Node(
Traceback (most recent call last):
  File "/opt/ws/install/formant_ros2_adapter/lib/formant_ros2_adapter/main.py", line 314, in <module>
    Adapter()
  File "/opt/ws/install/formant_ros2_adapter/lib/formant_ros2_adapter/main.py", line 59, in __init__
    self.node = rclpy.create_node("formant_ros2_bridge")
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/__init__.py", line 146, in create_node
    return Node(
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 174, in __init__
    self.__node = _rclpy.Node(
RuntimeError: Invalid argument
```

#### Additional information
This is with the following node, which is difficult to make an MCVE for since it interfaces with a cloud service. https://github.com/FormantIO/ros2-adapter

fujitatomoya, 2022-07-05:
@Aposhian this problem only happens on Galactic but Humble?

Aposhian, 2022-07-05:
I will be trying this on humble soon, although I don't have a good self-contained setup to do so, so it will probably be a few days.