ros2/rmw_fastrtps #635: use_sim_time with this->now() and callback inconsistency after #619 in Humble

benjinne, 2022-09-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - latest humble
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
After #619 merged into humble, my simulation became very unreliable. It worked well before that PR, so I'm wondering if anyone else is having issues with it.

I don't have a good example yet, but with `use_sim_time` I have some callbacks that eventually stop updating about a minute after launching. Other times, in my callback `this->now()` freezes and stops updating even though the /clock topic is always updating. For example I have a callback from one subscription which transforms a message into another format then grabs `this->now()` and publishes it into a new topic. Since the time stops updating on the output of that callback, it breaks my simulation.

I've tried using various callback groups and the multi-threaded executor, but nothing seems to help.

#### Expected behavior

this->now() with use_sim_time should always update and never freeze

audrow, 2022-09-27:
I'm going to see if I can reproduce this on 22.04. @MiguelCompany, have you seen anything like this?

fujitatomoya, 2022-09-27:
@Barry-Xu-2018 @iuhilnehc-ynos any idea? sound really critical issue.

iuhilnehc-ynos, 2022-09-28:
I can't reproduce this problem.
I tried using simple applications to get now() from a node with `use_sim_time` enable, but it seems work as I expected.

```sh

# terminal 1 (add a test application that a publisher publishes '/clock' with `rosgraph_msgs.msg.Clock` starting from 1)

ros2 run demo_nodes_py timer

# terminal 2 (update the callback to call `this->now()`)

ros2 run demo_nodes_cpp talker --ros-args -p use_sim_time:=True
		or
ros2 run demo_nodes_cpp talker  ( use terminal 3 to call `ros2 param set /talker use_sim_time True` later )

# terminal 2 log
[INFO] [1664331704.865874420] [talker]: Publishing: 'Hello World: 1' time: 1664331704865865605
[INFO] [1664331705.865777557] [talker]: Publishing: 'Hello World: 2' time: 1664331705865774037
[INFO] [1664331706.865919796] [talker]: Publishing: 'Hello World: 3' time: 1664331706865913545
[INFO] [1664331707.865875875] [talker]: Publishing: 'Hello World: 4' time: 1664331707865871912
[INFO] [1664331708.865777110] [talker]: Publishing: 'Hello World: 5' time: 1664331708865773788
[INFO] [1664331709.865772521] [talker]: Publishing: 'Hello World: 6' time: 1664331709865769315
[INFO] [1664331710.866109847] [talker]: Publishing: 'Hello World: 7' time: 0
[INFO] [1664331711.865835457] [talker]: Publishing: 'Hello World: 8' time: 15000000000
[INFO] [1664331712.866043628] [talker]: Publishing: 'Hello World: 9' time: 16000000000
[INFO] [1664331713.865918251] [talker]: Publishing: 'Hello World: 10' time: 17000000000
...
[INFO] [1664331873.866060468] [talker]: Publishing: 'Hello World: 170' time: 177000000000   # check it more than 2 minutes

```

The backtrace correctly shows to get the time if using simulator time for the node. I have no idea what kind of situation could block calling the now().

<details><summary> now() backtrace </summary>
<p>

```cpp

(gdb) bt
#0  rcl_get_ros_time (data=0x555555767760, current_time=0x7fffffffab68) at /home/chenlh/Projects/ROS2/ros2-master/src/ros2/rcl/rcl/src/rcl/time.c:71

			ready to use `rcutils_atomic_load_uint64_t` get the `current` from `t->current_time`

#1  0x00007ffff6cabb9b in rcl_clock_get_now (clock=0x5555557676b0, time_point_value=0x7fffffffab68)
    at /home/chenlh/Projects/ROS2/ros2-master/src/ros2/rcl/rcl/src/rcl/time.c:261
#2  0x00007ffff7bd32f7 in rclcpp::Clock::now (this=0x555555767690)
    at /home/chenlh/Projects/ROS2/ros2-master/src/ros2/rclcpp/rclcpp/src/rclcpp/clock.cpp:74
#3  0x00007ffff7c3fe9a in rclcpp::Node::now (this=0x555555602c40)
    at /home/chenlh/Projects/ROS2/ros2-master/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:544
#4  0x00007ffff4302dee in demo_nodes_cpp::Talker::Talker(rclcpp::NodeOptions const&)::{lambda()#1}::operator()() const (__closure=0x5555558a35c0)
    at /home/chenlh/Projects/ROS2/ros2-master/src/ros2/demos/demo_nodes_cpp/src/topics/talker.cpp:48
#5  0x00007ffff430dbb6 in rclcpp::GenericTimer<demo_nodes_cpp::Talker::Talker(rclcpp::NodeOptions const&)::{lambda()#1}, (void*)0>::execute_callback_delegate<{lambda()#1}, (demo_nodes_cpp::Talker::Talker(rclcpp::NodeOptions const&)::{lambda()#1})0>() (this=0x5555558a3590)
    at /home/chenlh/Projects/ROS2/ros2-master/install/include/rclcpp/rclcpp/timer.hpp:244
#6  0x00007ffff430d8be in rclcpp::GenericTimer<demo_nodes_cpp::Talker::Talker(rclcpp::NodeOptions const&)::{lambda()#1}, (void*)0>::execute_callback()
    (this=0x5555558a3590) at /home/chenlh/Projects/ROS2/ros2-master/install/include/rclcpp/rclcpp/timer.hpp:230
#7  0x00007ffff7bf8bd7 in rclcpp::Executor::execute_timer (timer=warning: RTTI symbol not found for class 'std::_Sp_counted_ptr_inplace<rclcpp::WallTimer<demo_nodes_cpp::Talker::Talker(rclcpp::NodeOptions const&)::{lambda()#1}, (void*)0>, std::allocator<void*>, (__gnu_cxx::_Lock_policy)2>'
warning: RTTI symbol not found for class 'std::_Sp_counted_ptr_inplace<rclcpp::WallTimer<demo_nodes_cpp::Talker::Talker(rclcpp::NodeOptions const&)::{lambda()#1}, (void*)0>, std::allocator<void*>, (__gnu_cxx::_Lock_policy)2>'
std::shared_ptr<class rclcpp::TimerBase> (use count 3, weak count 1) = {...})
    at /home/chenlh/Projects/ROS2/ros2-master/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:650
#8  0x00007ffff7bf766e in rclcpp::Executor::execute_any_executable (this=0x7fffffffb750, any_exec=...)
    at /home/chenlh/Projects/ROS2/ros2-master/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:512
#9  0x00007ffff7c089ca in rclcpp::executors::SingleThreadedExecutor::spin (this=0x7fffffffb750)
    at /home/chenlh/Projects/ROS2/ros2-master/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/single_threaded_executor.cpp:37
#10 0x0000555555566318 in main (argc=4, argv=0x7fffffffc548)
    at /home/chenlh/Projects/ROS2/ros2-master/build/demo_nodes_cpp/rclcpp_components/node_main_talker.cpp:62

```

</p>
</details>

fujitatomoya, 2022-09-28:
@audrow can you reproduce this issue?

audrow, 2022-09-28:
I haven't been able to reproduce it. I tried for a good while yesterday.

benjinne, 2022-09-28:
Anyone have tips on how I can debug a callback to see why this->now() gets stuck and keeps repeating the same value after a few minutes? I'm able to debug it with gdb but I don't know what to look for.

benjinne, 2022-09-28:
Edit, here's a better stacktrace with all of ros2 built with debug flags

<details>
  <summary>stack trace</summary>

  ```
(gdb) thread apply all backtrace

Thread 27 (Thread 0x7fffd8ff9700 (LWP 1718540)):
#0  __lll_lock_wait (futex=futex@entry=0x7ffffffe90d0, private=0) at lowlevellock.c:52
#1  0x00007ffff75750a3 in __GI___pthread_mutex_lock (mutex=0x7ffffffe90d0) at ../nptl/pthread_mutex_lock.c:80
#2  0x00007ffff7d32aa1 in __gthread_mutex_lock (__mutex=0x7ffffffe90d0) at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749
#3  std::mutex::lock (this=0x7ffffffe90d0) at /usr/include/c++/9/bits/std_mutex.h:100
#4  std::lock_guard<std::mutex>::lock_guard (__m=..., this=<synthetic pointer>) at /usr/include/c++/9/bits/std_mutex.h:159
#5  rclcpp::executors::MultiThreadedExecutor::run (this=0x7ffffffe8e90, this_thread_number=<optimized out>) at /home/ben/projects/ros2_humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:81
#6  0x00007ffff7a20de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7572609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#8  0x00007ffff785c133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 26 (Thread 0x7fffd97fa700 (LWP 1718539)):
#0  __lll_lock_wait (futex=futex@entry=0x7ffffffe90d0, private=0) at lowlevellock.c:52
#1  0x00007ffff75750a3 in __GI___pthread_mutex_lock (mutex=0x7ffffffe90d0) at ../nptl/pthread_mutex_lock.c:80
#2  0x00007ffff7d32aa1 in __gthread_mutex_lock (__mutex=0x7ffffffe90d0) at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749
#3  std::mutex::lock (this=0x7ffffffe90d0) at /usr/include/c++/9/bits/std_mutex.h:100
#4  std::lock_guard<std::mutex>::lock_guard (__m=..., this=<synthetic pointer>) at /usr/include/c++/9/bits/std_mutex.h:159
#5  rclcpp::executors::MultiThreadedExecutor::run (this=0x7ffffffe8e90, this_thread_number=<optimized out>) at /home/ben/projects/ros2_humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:81
#6  0x00007ffff7a20de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7572609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#8  0x00007ffff785c133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 25 (Thread 0x7fffd9ffb700 (LWP 1718538)):
#0  __lll_lock_wait (futex=futex@entry=0x7ffffffe90d0, private=0) at lowlevellock.c:52
#1  0x00007ffff75750a3 in __GI___pthread_mutex_lock (mutex=0x7ffffffe90d0) at ../nptl/pthread_mutex_lock.c:80
#2  0x00007ffff7d32aa1 in __gthread_mutex_lock (__mutex=0x7ffffffe90d0) at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749
--Type <RET> for more, q to quit, c to continue without paging--c
#3  std::mutex::lock (this=0x7ffffffe90d0) at /usr/include/c++/9/bits/std_mutex.h:100
#4  std::lock_guard<std::mutex>::lock_guard (__m=..., this=<synthetic pointer>) at /usr/include/c++/9/bits/std_mutex.h:159
#5  rclcpp::executors::MultiThreadedExecutor::run (this=0x7ffffffe8e90, this_thread_number=<optimized out>) at /home/ben/projects/ros2_humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:81
#6  0x00007ffff7a20de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7572609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#8  0x00007ffff785c133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 24 (Thread 0x7fffda7fc700 (LWP 1718537)):
#0  __lll_lock_wait (futex=futex@entry=0x7ffffffe90d0, private=0) at lowlevellock.c:52
#1  0x00007ffff75750a3 in __GI___pthread_mutex_lock (mutex=0x7ffffffe90d0) at ../nptl/pthread_mutex_lock.c:80
#2  0x00007ffff7d32aa1 in __gthread_mutex_lock (__mutex=0x7ffffffe90d0) at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749
#3  std::mutex::lock (this=0x7ffffffe90d0) at /usr/include/c++/9/bits/std_mutex.h:100
#4  std::lock_guard<std::mutex>::lock_guard (__m=..., this=<synthetic pointer>) at /usr/include/c++/9/bits/std_mutex.h:159
#5  rclcpp::executors::MultiThreadedExecutor::run (this=0x7ffffffe8e90, this_thread_number=<optimized out>) at /home/ben/projects/ros2_humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:81
#6  0x00007ffff7a20de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7572609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#8  0x00007ffff785c133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 23 (Thread 0x7fffdaffd700 (LWP 1718536)):
#0  __lll_lock_wait (futex=futex@entry=0x7ffffffe90d0, private=0) at lowlevellock.c:52
#1  0x00007ffff75750a3 in __GI___pthread_mutex_lock (mutex=0x7ffffffe90d0) at ../nptl/pthread_mutex_lock.c:80
#2  0x00007ffff7d32aa1 in __gthread_mutex_lock (__mutex=0x7ffffffe90d0) at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749
#3  std::mutex::lock (this=0x7ffffffe90d0) at /usr/include/c++/9/bits/std_mutex.h:100
#4  std::lock_guard<std::mutex>::lock_guard (__m=..., this=<synthetic pointer>) at /usr/include/c++/9/bits/std_mutex.h:159
#5  rclcpp::executors::MultiThreadedExecutor::run (this=0x7ffffffe8e90, this_thread_number=<optimized out>) at /home/ben/projects/ros2_humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:81
#6  0x00007ffff7a20de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7572609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#8  0x00007ffff785c133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

[... cut ...]
#5  rclcpp::executors::MultiThreadedExecutor::run (this=0x7ffffffe8e90, this_thread_number=<optimized out>) at /home/ben/projects/ros2_humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:81
#6  0x00007ffff7d32f95 in rclcpp::executors::MultiThreadedExecutor::spin (this=0x7ffffffe8e90) at /home/ben/projects/ros2_humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:62
#7  0x000055555558e794 in main (argc=<optimized out>, argv=<optimized out>) at /home/ben/projects/humble/src/communication/px4_ros2_interface/src/px4_ros2_interface_node.cpp:14
  ```
</details>

benjinne, 2023-01-11:
> @benjinne I think this might have been solved by [eProsima/Fast-DDS#3194](https://github.com/eProsima/Fast-DDS/pull/3194) and its corresponding backports ([eProsima/Fast-DDS#3195](https://github.com/eProsima/Fast-DDS/pull/3195)) for Humble.
> 
> Could you try?

I'm having a hard time replicating the issue. Sometimes it takes over 10 minutes for it to happen and I have to leave my computer run for a while. I can give this a try and let it run for an hour and see what happens

benjinne, 2023-01-11:
@MiguelCompany is there an easy way to check which fast-dds version is being used for a binary install of ros2?

benjinne, 2023-01-11:
I believe those fastdds changes helped. Everything is running more smoothly and no crashes after running for an hour. I'd say it's fixed.

clalancette, 2023-01-26:
Given the latest comments, I'm going to say that this is a duplicate of #650 , and close it as such.

closed 2023-01-26 by clalancette