ros2/rmw_fastrtps #636: [BUG] multicast disable and data-sharing enable causes unpredictable communication behavior.

ZhenshengLee, 2022-09-27:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu2204
- Installation type:
  - binary
- Version or commit hash:
  - humble
- DDS implementation:
  - fastdds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

the source code is in https://github.com/ZhenshengLee/ros2_shm_msgs/tree/humble

```sh
# t0
colcon build
# t1
cd ./colcon/install/shm_msgs/lib
ros2 launch shm_msgs image_talker.launch.py
# t2
ros2 launch shm_msgs image_listener.launch.py
```

the fastdds xml is in the follow, which is not committed into the github reop, the xml is according 
https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2_configure.html#creating-ros-contexts-and-nodes 
https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/disabling_multicast.html#disabling-all-multicast-traffic 
https://discourse.ros.org/t/unconfigured-dds-considered-harmful-to-networks/25689/24

```xml
<?xml version="1.0" encoding="UTF-8"?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <participant profile_name="participant_profile_ros2" is_default_profile="true">
        <rtps>
            <name>profile_for_ros2_context</name>
        </rtps>
        <builtin>
            <metatrafficUnicastLocatorList>
                <locator />
            </metatrafficUnicastLocatorList>
            <initialPeersList>
                <locator>
                    <udpv4>
                        <address>127.0.0.1</address>
                    </udpv4>
                </locator>
            </initialPeersList>
        </builtin>
    </participant>
    <data_writer profile_name="default publisher profile" is_default_profile="true">
        <qos>
            <publishMode>
                <kind>ASYNCHRONOUS</kind>
            </publishMode>
            <data_sharing>
                <kind>AUTOMATIC</kind>
            </data_sharing>
        </qos>
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </data_writer>
    <data_reader profile_name="default subscription profile" is_default_profile="true">
        <qos>
            <data_sharing>
                <kind>AUTOMATIC</kind>
            </data_sharing>
        </qos>
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </data_reader>
</profiles>

```

#### Expected behavior

actually  the image_talker.launch.py creates 8 image publisher processes.

So the ros2 graph should containe 8 nodes and the ros2cli should display

```sh
ros2 node list
ros2 topic list
```

#### Actual behavior

ros2 node list and ros2 topic list show the wrong info.
the communication is abnormal

#### Additional information

the bug exist in galactic, in which the fastdds version is 2.3.2

## Feature request

null

#### Implementation considerations

null

MiguelCompany, 2022-09-27:
@ZhenshengLee 

First, tag `builtin` should be inside tag `rtps`.
Second, in order to achieve what you intend, you'll need to configure `maxInitialPeersRange`. See the note [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/discovery/simple.html?highlight=maxInitialPeersRange#initial-peers)

I've modified your XML with the necessary settings

```xml
<?xml version="1.0" encoding="UTF-8"?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <transport_descriptors>
        <transport_descriptor>
            <transport_id>UDP_transport</transport_id>
            <type>UDPv4</type>
            <maxInitialPeersRange>10</maxInitialPeersRange>
        </transport_descriptor>
    </transport_descriptors>

    <participant profile_name="participant_profile_ros2" is_default_profile="true">
        <rtps>
            <name>profile_for_ros2_context</name>
            <userTransports>
                <transport_id>UDP_transport</transport_id>
            </userTransports>
            <useBuiltinTransports>false</useBuiltinTransports>

            <builtin>
                <metatrafficUnicastLocatorList>
                    <locator />
                </metatrafficUnicastLocatorList>
                <initialPeersList>
                    <locator>
                        <udpv4>
                            <address>127.0.0.1</address>
                        </udpv4>
                    </locator>
                </initialPeersList>
            </builtin>
        </rtps>
    </participant>

    <data_writer profile_name="default publisher profile" is_default_profile="true">
        <qos>
            <publishMode>
                <kind>ASYNCHRONOUS</kind>
            </publishMode>
            <data_sharing>
                <kind>AUTOMATIC</kind>
            </data_sharing>
        </qos>
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </data_writer>

    <data_reader profile_name="default subscription profile" is_default_profile="true">
        <qos>
            <data_sharing>
                <kind>AUTOMATIC</kind>
            </data_sharing>
        </qos>
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
    </data_reader>
</profiles>

```

ZhenshengLee, 2022-09-27:
@MiguelCompany 
Thanks for your fast reply!

> First, tag `builtin` should be inside tag `rtps`.

That's a typo in the issue page.

> ```
> <name>profile_for_ros2_context</name>
>             <userTransports>
>                 <transport_id>UDP_transport</transport_id>
>             </userTransports>
>             <useBuiltinTransports>false</useBuiltinTransports>
> ```

As in the docs, the data-sharing is a shared-memory-based function(but different from default shm transport) and is only available in intra-machine communication, so why should we use this config? Is data-sharing really working?

How to check if the data-sharing in working? I read from this the issue https://github.com/ros2/rmw_fastrtps/issues/579#issuecomment-1122709868 
 which suggest to check `/dev/shm` to see the mem files. But the shm(which is the default) transport also creates mem files in the same path. Is there a better way to check data-sharing function?

MiguelCompany, 2022-09-27:
> why should we use this config?

In order for the `<maxInitialPeersRange>` configuration to work, as it is configured as part of the transport.

> How to check if the data-sharing in working?

Data-sharing related files start with prefix *fast_datasharing*, so `ls /dev/shm/fast_datasharing*` should work

ZhenshengLee, 2022-09-28:
> In order for the `<maxInitialPeersRange>` configuration to work, as it is configured as part of the transport.
>
>
> Data-sharing related files start with prefix _fast_datasharing_, so `ls [...]

It's tested and confirmed in https://github.com/ZhenshengLee/ros2_shm_msgs/tree/humble
Thank you!

closed 2022-09-28 by ZhenshengLee