ros2/rmw_fastrtps #640: Lost large messages across subnets

Eumenes0362, 2022-10-06:
Question:

I am now trying to communicate ROS2 using fastDDS between different subnets.
But, large message drop in size (ex. RGB, 640x480) between "different subnets in the same domain".
Messages are not dropped between the "same subnets".
What are the possible causes and remedies for this?

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - ros2

#### Steps to reproduce issue

```
On Publisher PC(Subnet 1):
$ros2 bag play <pub_image_s.bag> -l

On Subscriver PC(Subnet 2):
$rviz2 -d image.rviz
```
#### Expected behavior
- Images are displayed in RViz2.

#### Actual behavior
* No image displayed.
* message was los
$ ros2 topic echo /image/rgb
A message was lost!!!
        total count change:1
        total count: 1---
A message was lost!!!
        total count change:4
        total count: 5---
A message was lost!!!
        total count change:1
        total count: 6---
A message was lost!!!
        total count change:1
        total count: 7---

fujitatomoya, 2022-10-06:
It would be probably better to open questions like this on [https://answers.ros.org](https://urldefense.com/v3/__https:/answers.ros.org__;!!JmoZiZGBv3RvKRSx!4yqoFSNjQX_iHeocuBfYJuswESHtJZGNBawU4QZxH7nk68dcbMATaNW3r8WeQFAC96qFXK01QiJU6XVwUvpVsEIrRr0$), which is our central Question and Answer site.

i think that multicast discovery does not work with different subnet.

closed 2022-10-06 by fujitatomoya

Eumenes0362, 2022-10-07:
Thanks for advice.
I will ask this in ROS Answers.