ros2/rmw_fastrtps #648: Cannot execute security in the keys saved in a softhsm user.

Juliamarsal, 2022-12-07:
## Bug report
Hello,
I am trying to save a security _key.pem_  from a security enclave in in a softHSM. I am working with environmental variables (not passing enclaves by argument), for example,  (export ROS_SECURITY_ENCLAVE_OVERRIDE=/gazebo).
I create a specific token and then I make an _import_ of the key.pem to the softHSM. Then, I delete the key.pem and then I add a symbolic link in the file key.p11 in order to get known where the _key.pem_ is saved in the softhsm.
When I execute the specific simulation, the following errors appears:

[robot_state_publisher-3] [INFO] [1670193287.558653675] [rcl]: Found security directory: /home/user/security_ws/turtlebot_keystore/enclaves/gazebo
[robot_state_publisher-3]
[robot_state_publisher-3] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[robot_state_publisher-3] This error state is being overwritten:
[robot_state_publisher-3]
[robot_state_publisher-3]   **'couldn't find all security files!, at /vulcanexus_ws/packaging/src/eProsima/RMW-Fast-DDS/rmw_fastrtps_shared_cpp/src/participant.cpp:396, at ./src/rcl/node.c:263'**
[robot_state_publisher-3]
[robot_state_publisher-3] with this new error message:
[robot_state_publisher-3]
[robot_state_publisher-3]   'rcl node's rmw handle is invalid, at ./src/rcl/node.c:415'
[robot_state_publisher-3]
[robot_state_publisher-3] rcutils_reset_error() should be called after error handling to avoid this.
[robot_state_publisher-3] <<<
[robot_state_publisher-3] [ERROR] [1670193287.562731377] [rcl]: Failed to fini publisher for node: 1
[robot_state_publisher-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[robot_state_publisher-3]   what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at ./src/rcl/node.c:415
[ERROR] [robot_state_publisher-3]: process has died [pid 28892, exit code -6, cmd '/opt/ros/humble/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=robot_state_publisher --params-file /tmp/launch_params_ttmdd7ti'].
[spawn_entity.py-4] [INFO] [1670193288.527872715] [rcl]: Found security directory: /home/user/security_ws/turtlebot_keystore/enclaves/gazebo

**Required Info:**

- Operating System:
 - Ubuntu 22.04
- Installation type:
   - sudo apt update
   - sudo apt install --yes --no-install-recommends vulcanexus-humble-tools vulcanexus-humble-simulation
- Version or commit hash:
  - humble
- DDS implementation:
  - FastDDS with SROS2
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
``` sudo softhsm2-util \
   --init-token \
   --free \
   --label vulcanexus_gazebo \
   --pin VulcanexusGazebo \
   --so-pin VulcanexusGazebo
```
sudo softhsm2-util \
   --import /home/user/security_ws/turtlebot_keystore/enclaves/gazebo/key.pem  \
   --token vulcanexus_gazebo \
   --label key \
   --pin VulcanexusGazebo \
   --id 123456789ABCDEF123456789ABCDEF
​
```
sudo rm -rf /home/user/security_ws/turtlebot_keystore/enclaves/gazebo/key.pem
```
GNUTLS_PIN=VulcanexusGazebo sudo  p11tool \
--provider /usr/lib/softhsm/libsofthsm2.so \
--list-tokens \
--login | grep "token=vulcanexus_gazebo" | awk '{print $2}' >  /home/user/security_ws/turtlebot_keystore/enclaves/gazebo/key.p11

````
cd
source /opt/vulcanexus/humble/setup.bash
export FASTDDS_PKCS11_PIN=VulcanexusGazebo
export ROS_SECURITY_KEYSTORE=/home/user/security_ws/turtlebot_keystore
export ROS_SECURITY_ENABLE=true
export ROS_SECURITY_STRATEGY=Enforce
export TURTLEBOT3_MODEL=burger
export ROS_SECURITY_ENCLAVE_OVERRIDE=/gazebo
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py

#### Expected behavior

Should appear the gazebo simulation without any errors.

#### Actual behavior

The gazebo simulation does not open.

---- 
## Feature request

#### Feature description
The feature should allow to execute keys saved in other hardware.

EduPonz, 2022-12-07:
I think this issue belongs in [eprosima/rmw_fastrtps](https://github.com/eprosima/rmw_fastrtps), but I cannot transfer it.

Juliamarsal, 2022-12-07:
Hello @EduPonz I just tried to open the same issue in [eprosima/rmw_fastrtps](https://github.com/eprosima/rmw_fastrtps), but I cannot add any issues as issues are not available. Thanks and sorry for the inconvenience.

EduPonz, 2022-12-07:
Hi @Juliamarsal , I just enabled issues on the eProsima fork. You should be able to open the issue there now

Juliamarsal, 2022-12-08:
Hello @EduPonz! Lots of thanks! I just created the same issue here [eprosima/rmw_fastrtps](https://github.com/eprosima/rmw_fastrtps). This issue can be closed now.

closed 2022-12-08 by Juliamarsal