ros2/rmw_fastrtps #650: Deadlock when using MultiThreadedExecutor

hplatou, 2022-12-12:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 22.04, Humble
- Installation type: 
  - binaries
- Version or commit hash:
  - v6.2.2
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Running this simple example with a `MultiThreadedExecutor` eventually causes a deadlock (typically within a few seconds). From the debugger it seems to be stuck on `WaitSetImpl::wait` (see attached stack trace). This behavior seems to be introduced by #633 

```C++ 
class Sender : public rclcpp::Node
{
  public:
    Sender()
    : Node("sender"), count_(0)
    {
      publisher_ = this->create_publisher<std_msgs::msg::UInt32>("tick", 100);
      timer_ = this->create_wall_timer(
        10ms, std::bind(&Sender::callback, this));
    }

  private:
    void callback()
    { 
      auto message = std_msgs::msg::UInt32();
      message.data = count_ ++;
      std::cout << "Sending tick: " << message.data << "..." << std::endl;
      publisher_->publish(message);
    }

    rclcpp::TimerBase::SharedPtr timer_;
    rclcpp::Publisher<std_msgs::msg::UInt32>::SharedPtr publisher_;
    size_t count_;
};

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);

  auto node = std::make_shared<Sender>();
  rclcpp::executors::MultiThreadedExecutor executor;
  std::cout << "Created multithreaded executor. Num threads: " << executor.get_number_of_threads() << std::endl;
  executor.add_node(node);
  executor.spin();
  
  rclcpp::shutdown();
  return 0;
}

```

Stacktrace when stuck:
```
libc.so.6!__futex_abstimed_wait_common64(int private, _Bool cancel, const struct timespec * abstime, int op, unsigned int expected, unsigned int * futex_word) (futex-internal.c:57)
libc.so.6!__futex_abstimed_wait_common(_Bool cancel, int private, const struct timespec * abstime, clockid_t clockid, unsigned int expected, unsigned int * futex_word) (futex-internal.c:87)
libc.so.6!__GI___futex_abstimed_wait_cancelable64(unsigned int * futex_word, unsigned int expected, clockid_t clockid, const struct timespec * abstime, int private) (futex-internal.c:139)
libc.so.6!__pthread_cond_wait_common(pthread_mutex_t * mutex, pthread_cond_t * cond) (pthread_cond_wait.c:503)
libc.so.6!___pthread_cond_wait(pthread_cond_t * cond, pthread_mutex_t * mutex) (pthread_cond_wait.c:627)
libfastrtps.so.2.6!eprosima::fastdds::dds::detail::WaitSetImpl::wait(std::vector<eprosima::fastdds::dds::Condition*, std::allocator<eprosima::fastdds::dds::Condition*> >&, eprosima::fastrtps::Time_t const&) (Unknown Source:0)
libfastrtps.so.2.6!eprosima::fastdds::dds::WaitSet::wait(std::vector<eprosima::fastdds::dds::Condition*, std::allocator<eprosima::fastdds::dds::Condition*> >&, eprosima::fastrtps::Time_t) const (Unknown Source:0)
librmw_fastrtps_shared_cpp.so!rmw_fastrtps_shared_cpp::__rmw_wait(char const*, rmw_subscriptions_s*, rmw_guard_conditions_s*, rmw_services_s*, rmw_clients_s*, rmw_events_s*, rmw_wait_set_s*, rmw_time_s const*) (Unknown Source:0)
librmw_fastrtps_shared_cpp.so![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libc.so.6!start_thread(void * arg) (pthread_create.c:442)
libc.so.6!clone3() (clone3.S:81)
```

#### Expected behavior
No deadlock

#### Actual behavior
Callback stops executing, typically within a few seconds

fujitatomoya, 2022-12-12:
@hplatou could be related to https://github.com/eProsima/Fast-DDS/issues/2961 ? that is something we have experienced before.

I think https://github.com/eProsima/Fast-DDS/pull/2976 is not released yet as humble patch.

MiguelCompany, 2022-12-13:
eProsima/Fast-DDS#2976 has been released with Fast DDS v2.6.3, which is pending humble [sync](http://repo.ros2.org/status_page/ros_humble_default.html?q=fastrtps).

It seems a sync is [planned for Dec 16th](https://discourse.ros.org/t/preparing-for-humble-sync-2022-12-16/28737)

hplatou, 2022-12-19:
Thanks for your answers, but sadly, the 2.6.3 update of Fast DDS did not fix the problem. Only way I can make it work is to build the 6.2.1 version of rmw_fastrtps from source.

fujitatomoya, 2022-12-19:
@iuhilnehc-ynos @Barry-Xu-2018 CCed.

iuhilnehc-ynos, 2022-12-20:
I can reproduce this issue with [the latest humble repos](https://github.com/ros2/ros2/blob/humble/ros2.repos).

| repo | commit id|
| :-: | :-:|
| Fast-DDS | https://github.com/eProsima/Fast-DDS/commit/dfe8204aff787202764b9c2338d82752ee049442 |
| rmw_fastrtps | 8955d4e574db3469cdcba7f52211cdf2180fbb1d|

application log (replace with the issue code in examples/rclcpp/topics/minimal_publisher/member_function.cpp)

```
chenlh ros2-humble $ ros2 run examples_rclcpp_minimal_publisher publisher_member_function
Created multithreaded executor. Num threads: 16
Sending tick: 0...
Sending tick: 1...
Sending tick: 2...
Sending tick: 3...
Sending tick: 4...
Sending tick: 5...
<... omited>
Sending tick: 1399...
Sending tick: 1400...
Sending tick: 1401...
Sending tick: 1402...
Sending tick: 1403...
Sending tick: 1404...
Sending tick: 1405...
Sending tick: 1406...
Sending tick: 1407...
Sending tick: 1408...
<blocking here, it seems the timer callback is not called anymore>
```

<details><summary> thread apply all bt </summary>
<p>

```shell
chenlh gdb $ gdb
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) attach 1526191
Attaching to process 1526191
[New LWP 1526192]
[New LWP 1526193]
[New LWP 1526194]
[New LWP 1526195]
[New LWP 1526196]
[New LWP 1526197]
[New LWP 1526198]
[New LWP 1526199]
[New LWP 1526200]
[New LWP 1526201]
[New LWP 1526202]
[New LWP 1526203]
[New LWP 1526204]
[New LWP 1526205]
[New LWP 1526206]
[New LWP 1526207]
[... 559 lines cut ...]

Thread 2 (Thread 0x7f80c4826640 (LWP 1526192) "publisher_m-ust"):
#0  __recvmsg_syscall (flags=0, msg=0x7f80c48246b0, fd=3) at ../sysdeps/unix/sysv/linux/recvmsg.c:27
#1  __libc_recvmsg (fd=3, msg=0x7f80c48246b0, flags=0) at ../sysdeps/unix/sysv/linux/recvmsg.c:41
#2  0x00007f80c4992eec in ?? () from /lib/x86_64-linux-gnu/liblttng-ust.so.1
#3  0x00007f80c4967f42 in ?? () from /lib/x86_64-linux-gnu/liblttng-ust.so.1
#4  0x00007f80c4c23b43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#5  0x00007f80c4cb5a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

Thread 1 (Thread 0x7f80c4830680 (LWP 1526191) "publisher_membe"):
#0  futex_wait (private=0, expected=2, futex_word=0x7ffdd8094ce0) at ../sysdeps/nptl/futex-internal.h:146
#1  __GI___lll_lock_wait (futex=futex@entry=0x7ffdd8094ce0, private=0) at ./nptl/lowlevellock.c:49
#2  0x00007f80c4c27082 in lll_mutex_lock_optimized (mutex=0x7ffdd8094ce0) at ./nptl/pthread_mutex_lock.c:48
#3  ___pthread_mutex_lock (mutex=0x7ffdd8094ce0) at ./nptl/pthread_mutex_lock.c:93
#4  0x00007f80c5a84a01 in __gthread_mutex_lock (__mutex=0x7ffdd8094ce0) at /usr/include/x86_64-linux-gnu/c++/11/bits/gthr-default.h:749
#5  0x00007f80c5a872b2 in std::mutex::lock (this=0x7ffdd8094ce0) at /usr/include/c++/11/bits/std_mutex.h:100
#6  0x00007f80c5a87900 in std::lock_guard<std::mutex>::lock_guard (this=0x7ffdd8094848, __m=...) at /usr/include/c++/11/bits/std_mutex.h:229
#7  0x00007f80c5ac7169 in rclcpp::executors::MultiThreadedExecutor::run (this=0x7ffdd8094aa0, this_thread_number=15) at /home/chenlh/Projects/ROS2/ros2-humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:81
#8  0x00007f80c5ac6fd5 in rclcpp::executors::MultiThreadedExecutor::spin (this=0x7ffdd8094aa0) at /home/chenlh/Projects/ROS2/ros2-humble/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:62
#9  0x000055c5db6b191b in main (argc=1, argv=0x7ffdd8094e58) at /home/chenlh/Projects/ROS2/ros2-humble/src/ros2/examples/rclcpp/topics/minimal_publisher/member_function.cpp:64

```

</p>
</details>

iuhilnehc-ynos, 2022-12-20:
The notification of https://github.com/ros2/rclcpp/blob/33cbd76c07dc9a30e8dbeecd5f5e70057122a369/rclcpp/src/rclcpp/executor.cpp#L540 might be lost.

Let me try to explain this issue.

About the timer callback called in multiple thread executor, we assume that `thread a` gets the timer timeout and calls the timer callback and then calls the [interrupt_guard_condition_.trigger();](https://github.com/ros2/rclcpp/blob/33cbd76c07dc9a30e8dbeecd5f5e70057122a369/rclcpp/src/rclcpp/executor.cpp#L540) to trigger the interrupt_guard_condition_ for wait_set  of thread b, so thread b can get the notification to `continue` to add the `timer` into waitset. If the `interrupt_guard_condition_.trigger();` notification is lost, `thread b` can't `continue` to add the `timer` into its waitset. That's why `thread 25` in https://github.com/ros2/rmw_fastrtps/issues/650#issuecomment-1358770243 wait without a timeout.

thread a of executor:

```
MultiThreadedExecutor::run
  for {
    {
      std::lock_guard wait_lock{wait_mutex_};
      if (!get_next_executable(any_exec, next_exec_timeout_)) {
        continue;
      }
    }

    execute_any_executable         // keep the `CallbackGroup::can_be_taken_from_(true)` and call timer_callback
                                   // and then call `interrupt_guard_condition_.trigger();` to trigger the waitset in thread b
  }
  
  

```

thread b of executor:

```
MultiThreadedExecutor::run
  for {
    {
      std::lock_guard wait_lock{wait_mutex_};
      if (!get_next_executable(any_exec, next_exec_timeout_)) {             // wait without timer until the interrupt_guard_condition triggered
        continue;
      }
    }

    execute_any_executable
  }

```

I think it's related to https://github.com/eProsima/Fast-DDS/pull/3087 .

fujitatomoya, 2022-12-20:
@iuhilnehc-ynos thanks for checking on this, https://github.com/ros2/rmw_fastrtps/issues/650#issuecomment-1358941776 makes sense to me. So this could be `wait forever` problem but `deadlock`. did you confirm that https://github.com/eProsima/Fast-DDS/pull/3087 can solve this problem with reproducible test sample?

iuhilnehc-ynos, 2022-12-21:
> this could be `wait forever` problem but `deadlock`.

Yes, it's not a deadlock.

> did you confirm that [eProsima/Fast-DDS#3087](https://github.com/eProsima/Fast-DDS/pull/3087) can solve this problem with reproducible test sample?

Yes. Two ways I confirmed. (a. the example ran from yesterday to this morning and it seems good till now. b. rerun it after 20 seconds many times.)

hplatou, 2022-12-21:
I can also confirm that this resolves my problems. Thanks again!

hplatou, 2023-01-02:
Any chance this will be fixed for the next package sync of Humble?

fujitatomoya, 2023-01-03:
> Any chance this will be fixed for the next package sync of Humble?

This is a question for eProsima. https://github.com/ros2/ros2/blob/00f3ba9f73916a5eab322710edaaf197f0f10e31/ros2.repos#L34-L37 uses `2.6.x` branch for humble. if the fix can be backported to `2.6.x` branch, we can fix this problem ROS 2 humble as well.

CC: @MiguelCompany

MiguelCompany, 2023-01-09:
@fujitatomoya 
Fix for rolling is currently eProsima/Fast-DDS#3194
See eProsima/Fast-DDS#3195 for the backport to Fast-DDS 2.6.x (i.e. Humble)

MiguelCompany, 2023-02-07:
@hplatou The fix on Fast DDS has already been released on the Humble binaries.

Do you think this can be closed?

hplatou, 2023-02-12:
Yes, I can confirm that the released version has resolved the issue. Thanks!

closed 2023-02-12 by hplatou