ros2/rmw_fastrtps #666: busy loop consumes CPU 100%

fujitatomoya, 2023-02-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - apt / debian packages
- Version or commit hash:
  - humble
- DDS implementation:
  - Fast-DDS https://github.com/eProsima/Fast-DDS/tree/v2.6.2
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

This problem only can be reproducible in internal application evaluation. So far unable to develop simple self-contained reproducible application.

#### Expected behavior

Busy loop does not happen, and all application nodes can start up w/o error or problems.

#### Actual behavior

Busy loop happens during application initialization, that will loop forever that consumes CPU 100%.

#### Additional information

according to https://github.com/ros2/ros2/blob/e3d99dcd430b46455f7ca2e134f643810ad34e97/ros2.repos#L36-L37, https://github.com/eProsima/Fast-DDS/tree/v2.6.2 is used.

busy loop starts from here,

https://github.com/eProsima/Fast-DDS/blob/5076ebc0c5d030cac6225b94e18ef5b17c996ef3/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp#L1268

then, process following,

https://github.com/eProsima/Fast-DDS/blob/5076ebc0c5d030cac6225b94e18ef5b17c996ef3/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp#L1305

change_to_process points always the same address,

and,

https://github.com/eProsima/Fast-DDS/blob/5076ebc0c5d030cac6225b94e18ef5b17c996ef3/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp#L1321

will always be false, then end up having the next loop from the top. this mutex is locked by `WriterProxy::WriterProxy` constructor.

https://github.com/eProsima/Fast-DDS/blob/5076ebc0c5d030cac6225b94e18ef5b17c996ef3/src/cpp/rtps/reader/WriterProxy.cpp#L102

```bash

(gdb) s
eprosima::fastrtps::rtps::Endpoint::getMutex (this=0x55828b720960) at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/include/fastdds/rtps/Endpoint.h:92
92	        return mp_mutex;
(gdb) p mp_mutex
$23 = {<std::__recursive_mutex_base> = {_M_mutex = {__data = {__lock = 2, __count = 2, __owner = 634358, __nusers = 1, __kind = 1, 
        __spins = 0, __elision = 0, __list = {__prev = 0x0, __next = 0x0}}, 
      __size = "\002\000\000\000\002\000\000\000\366\255\t\000\001\000\000\000\001", '\000' <repeats 22 times>, 
      __align = 8589934594}}, <std::__timed_mutex_impl<std::recursive_timed_mutex>> = {<No data fields>}, <No data fields>}
```

I believe this mutex is locked on here,

https://github.com/eProsima/Fast-DDS/blob/5076ebc0c5d030cac6225b94e18ef5b17c996ef3/src/cpp/rtps/writer/StatefulWriter.cpp#L1867

fujitatomoya, 2023-02-03:
<details>
<summary>stack trace via gdb</summary>

ROS: humble
Install type: built from source, and apt installed binary packages.
target node: /opt/ros/humble/lib/topic_tools/relay
process id: 634333 
LPW in a busy loop: 634406

(gdb) n
1306	            while (nullptr != change_to_process)
(gdb) 
1309	                if (!async_mode.fast_check_is_there_slot_for_change(change_to_process))
(gdb) 
1314	                if (nullptr == current_writer || current_writer->getGuid() != change_to_process->writerGUID)
(gdb) 
1316	                    auto writer_it = writers_.find(change_to_process->writerGUID);
(gdb) 
1317	                    assert(writers_.end() != writer_it);
(gdb) 
1319	                    current_writer = writer_it->second;
(gdb) 
1322	                if (!current_writer->getMutex().try_lock())
(gdb) 
1324	                    break;
(gdb) 
1381	            async_mode.group.sender(nullptr, nullptr);
(gdb) 
1269	        while (async_mode.running) // loop starts here
(gdb) n
1272	            if (0 != async_mode.writers_interested_in_remove)
(gdb) n
1277	            std::unique_lock<std::mutex> lock(mutex_);
(gdb) n
1278	            fastrtps::rtps::CacheChange_t* change_to_process = nullptr;
(gdb) n
1282	                std::unique_lock<std::mutex> in_lock(async_mode.changes_interested_mutex);
(gdb) n
1284	                sched.add_interested_changes_to_queue_nts();
(gdb) n
1286	                while (async_mode.running &&
(gdb) n
1287	                        (async_mode.force_wait() || nullptr == (change_to_process = sched.get_next_change_nts())))
(gdb) s
eprosima::fastdds::rtps::FlowControllerAsyncPublishMode::force_wait (this=0x55828b641148) at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp:229
229	        return false;
(gdb) 
230	    }
(gdb) 
eprosima::fastdds::rtps::FlowControllerImpl<eprosima::fastdds::rtps::FlowControllerSyncPublishMode, eprosima::fastdds::rtps::FlowControllerFifoSchedule>::run (this=0x55828b640a60) at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp:1286
1286	                while (async_mode.running &&
(gdb) 
1287	                        (async_mode.force_wait() || nullptr == (change_to_process = sched.get_next_change_nts())))
(gdb) 
eprosima::fastdds::rtps::FlowControllerFifoSchedule::get_next_change_nts (this=0x55828b640ac8) at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp:411
411	        return queue_.get_next_change();
(gdb) 
eprosima::fastdds::rtps::FlowQueue::get_next_change (this=0x55828b640ac8) at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp:75
75	        if (!is_empty())
(gdb) 
eprosima::fastdds::rtps::FlowQueue::is_empty (this=0x55828b640ac8) at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp:58
58	        return new_ones_.is_empty() && old_ones_.is_empty();
(gdb) 
eprosima::fastdds::rtps::FlowQueue::ListInfo::is_empty (this=0x55828b640e08) at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp:125
125	            assert((&tail == head.writer_info.next && &head == tail.writer_info.previous) ||
(gdb) 
127	            return &tail == head.writer_info.next;
(gdb) p &tail
$17 = (eprosima::fastrtps::rtps::CacheChange_t *) 0x55828b640ed8
(gdb) p head.writer_info.next
[... 260 lines cut ...]
(gdb) up
#29 0x00007f9cac852f73 in std::__invoke<void (eprosima::fastrtps::rtps::ResourceEvent::*)(), eprosima::fastrtps::rtps::ResourceEvent*> (
    __fn=@0x55828b635e20: (void (eprosima::fastrtps::rtps::ResourceEvent::*)(eprosima::fastrtps::rtps::ResourceEvent * const)) 0x7f9cac84cd16 <eprosima::fastrtps::rtps::ResourceEvent::event_service()>) at /usr/include/c++/11/bits/invoke.h:96
96	      return std::__invoke_impl<__type>(__tag{}, std::forward<_Callable>(__fn),
(gdb) 
#30 0x00007f9cac852ed3 in std::thread::_Invoker<std::tuple<void (eprosima::fastrtps::rtps::ResourceEvent::*)(), eprosima::fastrtps::rtps::ResourceEvent*> >::_M_invoke<0ul, 1ul> (this=0x55828b635e18) at /usr/include/c++/11/bits/std_thread.h:253
253		  { return std::__invoke(std::get<_Ind>(std::move(_M_t))...); }
(gdb) 
#31 0x00007f9cac852e88 in std::thread::_Invoker<std::tuple<void (eprosima::fastrtps::rtps::ResourceEvent::*)(), eprosima::fastrtps::rtps::ResourceEvent*> >::operator() (this=0x55828b635e18) at /usr/include/c++/11/bits/std_thread.h:260
260		  return _M_invoke(_Indices());
(gdb) 
#32 0x00007f9cac852e68 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (eprosima::fastrtps::rtps::ResourceEvent::*)(), eprosima::fastrtps::rtps::ResourceEvent*> > >::_M_run (this=0x55828b635e10) at /usr/include/c++/11/bits/std_thread.h:211
211		_M_run() { _M_func(); }
(gdb) 
#33 0x00007f9cadbaf2c3 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) 
#34 0x00007f9cad91fb43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
442	./nptl/pthread_create.c: No such file or directory.
(gdb) 
#35 0x00007f9cad9b1a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
81	../sysdeps/unix/sysv/linux/x86_64/clone3.S: No such file or directory.
(gdb) 
Initial frame selected; you cannot go up.

</details>

fujitatomoya, 2023-02-03:
CC: @MiguelCompany

fujitatomoya, 2023-02-16:
@iuhilnehc-ynos if you have bandwidth, could you take a look at this?

iuhilnehc-ynos, 2023-02-21:
@fujitatomoya 

Could you check if the internal application can still reproduce this issue with the latest version of humble distro?
There are a lot of patches upgraded from v2.6.2 to v2.6.4.
```
$ dpkg -l | grep -i fastrtps
ii  ros-humble-fastrtps                               2.6.4-1jammy.20230117.223829            amd64        *eprosima Fast DDS* (formerly Fast RTPS) is a C++ implementation of the DDS (Data Distribution Service) standard of the OMG (Object Management Group).
```

iuhilnehc-ynos, 2023-02-21:
From the debug information, we can see that the shared_lock for the `endpoints_list_mutex` is waiting.
```shell
#6 0x00007f9cac8590ee in eprosima::shared_lockeprosima::shared_mutex::shared_lock (this=0x7f9caa84fd80, m=...)
at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/include/fastrtps/utils/shared_mutex.hpp:175
175 m->lock_shared();
(gdb) p m
$33 = (eprosima::shared_lockeprosima::shared_mutex::mutex_type &) @0x55828b635580: {mut = {std::__mutex_base = {_M_mutex = {__data = {
__lock = 0, __count = 0, __owner = 0, __nusers = 2, __kind = 0, __spins = 0, __elision = 0, __list = {__prev = 0x0,
__next = 0x0}}, __size = '\000' <repeats 12 times>, "\002", '\000' <repeats 26 times>, _align = 0}}, },
gate1 = {_M_cond = {_M_cond = {__data = {__wseq = {__value64 = 2, __value32 = {__low = 2, __high = 0}}, __g1_start = {__value64 = 0,
__value32 = {__low = 0, __high = 0}}, __g_refs = {2, 0}, __g_size = {0, 0}, __g1_orig_size = 0, __wrefs = 8, __g_signals = {0,
0}},
__size = "\002", '\000' <repeats 15 times>, "\002", '\000' <repeats 19 times>, "\b\000\000\000\000\000\000\000\000\000\000",
_align = 2}}}, gate2 = {_M_cond = {_M_cond = {__data = {__wseq = {__value64 = 2, __value32 = {__low = 2, __high = 0}},
__g1_start = {__value64 = 0, __value32 = {__low = 0, __high = 0}}, __g_refs = {2, 0}, __g_size = {0, 0}, __g1_orig_size = 0,
__wrefs = 8, __g_signals = {0, 0}},
_size = "\002", '\000' <repeats 15 times>, "\002", '\000' <repeats 19 times>, "\b\000\000\000\000\000\000\000\000\000\000",
align = 2}}}, state = 2147483649, static write_entered = 2147483648, static n_readers = 2147483647}
(gdb) up
#7 0x00007f9cac9747aa in eprosima::fastrtps::rtps::RTPSParticipantImpl::find_local_reader (this=0x55828b635080, reader_guid=...)
at /home/ubuntu/hello-ros2/ros2_humble/src/eProsima/Fast-DDS/src/cpp/rtps/participant/RTPSParticipantImpl.cpp:1158
1158 shared_lock<shared_mutex> (endpoints_list_mutex);
```
There was a PR(https://github.com/eProsima/Fast-DDS/pull/2976) to deal with this similar issue, it expects that
`the reader will receive the requested lock, even if a writer is waiting.`

After breaking this waiting lock `endpoints_list_mutex`, the lock in the `StatefulWriter::process_acknack` called at https://github.com/eProsima/Fast-DDS/blob/5076ebc0c5d030cac6225b94e18ef5b17c996ef3/src/cpp/rtps/reader/WriterProxy.cpp#L522 will be unlocked later, and then https://github.com/eProsima/Fast-DDS/blob/5076ebc0c5d030cac6225b94e18ef5b17c996ef3/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp#L1321 can try_lock sucessfully.

fujitatomoya, 2023-03-14:
Does this happen to rolling? it is worth to check if the problem still stays in rolling which uses Fast-DDS master branch.

iuhilnehc-ynos, 2023-03-15:
https://github.com/eProsima/Fast-DDS/pull/2976 fixed for the `master` at first and then backported `2.7.x` and `2.6.x`.
As we know, `rolling` use the [master](https://github.com/ros2/ros2/blob/rolling/ros2.repos#L34-L37) branch for `Fast-DDS`.

MiguelCompany, 2023-03-15:
> Actually, I don't know how to reproduce this issue.

Does this mean we can close this issue, since it was fixed by eProsima/Fast-DDS#2976 and its backport to humble (eProsima/Fast-DDS#3091)?

iuhilnehc-ynos, 2023-03-15:
@MiguelCompany 

Thanks for your reply.
I think it's better to wait for the final confirmation from @fujitatomoya , as it's reported by an `internal application`(maybe the internal application we all can't get.) that using ros2 tool(`/opt/ros/humble/lib/topic_tools/relay`).

fujitatomoya, 2023-03-15:
@MiguelCompany 

let us have some time to make sure this has been addressed to humble, we will confirm that the problem cannot be observed with humble source build.

fujitatomoya, 2023-03-23:
@MiguelCompany this has been confirmed our internal test, no problem observed with current humble source build. thanks 👍

closed 2023-03-23 by fujitatomoya