ros2/rmw_fastrtps #670: How to enable LoanedMessages (Foxy)

MarcoMagriDev, 2023-02-13:
## Bug report

**Required Info:**

- Operating System:
   - Ubuntu 20.04
- Installation type:
  - ROS2 foxy installed via apt-get
- DDS implementation:
  - Fast-DDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker_loaned_message
[INFO] [1644340760.069653573] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.
[INFO] [1644340760.069869940] [loaned_message_talker]: Publishing: '2.000000'
[INFO] [1644340760.078255330] [rclcpp]: Currently used middleware can't loan messages. Local allocator will be used.
[INFO] [1644340760.078312705] [loaned_message_talker]: Publishing: 'Hello World: 2'
[INFO] [1644340761.069560102] [loaned_message_talker]: Publishing: '3.000000'
[INFO] [1644340761.069749556] [loaned_message_talker]: Publishing: 'Hello World: 3'
[INFO] [1644340762.069596311] [loaned_message_talker]: Publishing: '4.000000'
[INFO] [1644340762.069794361] [loaned_message_talker]: Publishing: 'Hello World: 4'
[INFO] [1644340763.069584865] [loaned_message_talker]: Publishing: '5.000000'
[INFO] [1644340763.069779696] [loaned_message_talker]: Publishing: 'Hello World: 5'
```

#### Expected behavior
`LoanedMessages` should be enabled for POD data type.

#### Actual behavior
`LoanedMessages` cannot be enabled.

#### Additional information
I tried to follow the steps reported here https://github.com/ros2/rmw_fastrtps/issues/579#issuecomment-1122709868 but i get
```bash
[XMLPARSER Error] Invalid element found into 'writerQosPoliciesType'. Name: data_sharing -> Function getXMLWriterQosPolicies
[XMLPARSER Error] Error parsing publisher profile -> Function parseXMLPublisherProf
[XMLPARSER Error] Error parsing profile's tag data_writer -> Function parseProfiles
[XMLPARSER Error] Invalid element found into 'readerQosPoliciesType'. Name: data_sharing -> Function getXMLReaderQosPolicies
[XMLPARSER Error] Error parsing subscriber profile -> Function parseXMLSubscriberProf
[XMLPARSER Error] Error parsing profile's tag data_reader -> Function parseProfiles
[XMLPARSER Error] Error parsing '/home/user/foxy_ws/src/fastDDSprofile.xml' -> Function loadXMLFile
[RTPS_READER_HISTORY Error] Change payload size of '76' bytes is larger than the history payload size of '35' bytes and cannot be resized. -> Function add_change
[RTPS_READER_HISTORY Error] Change payload size of '100' bytes is larger than the history payload size of '35' bytes and cannot be resized. -> Function add_change
```
This seems to be due to the fact that ROS2 foxy uses the version 2.1.2 of FASTrtps in which the `<data_sharing>` is not valid for xml profile file. How can `LoanedMessages` be achieved in ROS2 foxy?

fujitatomoya, 2023-02-13:
@MarcoMagriDev i think that `LoanedMessages` in ROS 2 (data sharing in Fast-DDS term) is supported on Galactic or later. that means you can use `LoanedMessages` zero copy with Humble, but Foxy.

@EduPonz @MiguelCompany correct me if i am mistaken.

JLBuenoLopez-eProsima, 2023-02-14:
@fujitatomoya is right! [Fast DDS v2.2.0](https://github.com/eProsima/Fast-DDS/releases/tag/v2.2.0) introduced Data Sharing delivery and the required API that allows zero-copy delivery. ROS 2 Foxy uses Fast DDS v2.1.x (right now the latest release on that branch is Fast DDS v2.1.3) but according to [Fast DDS versioning policy](https://github.com/eProsima/Fast-DDS/blob/master/VERSIONING.md) and [ROS 2 REP 2004 Quality assurance](https://ros.org/reps/rep-2004.html#id23), no ABI breaks are allowed within a ROS 2 stable version. Consequently, only patches are backported and no new feature breaking ABI is introduced. Thus, LoanedMessages API is not supported with Fast DDS in ROS 2 Foxy. If you want to use this feature, please upgrade to ROS 2 Humble.

MarcoMagriDev, 2023-02-14:
Thank you for the quick reply. 

For others who are in the my same situation: my goal was to achieve `LoanedMessages` on Ubuntu 20.04 (regardless of the ros distribution) so i moved to Rolling that still has compatibility with Ubuntu 20.04 but with a FastDDS version (2.3.4 in my case) that supports Data Sharing delivery.

Closing the issue.

closed 2023-02-14 by MarcoMagriDev