ros2/rmw_fastrtps #673: Discovery server in AP mode

roni-kreinin, 2023-03-02:
**Required Info:**

- Operating System:
  - Ubuntu 22.04 Server
- Installation type:
  - Binary
- Version or commit hash:
  - latest
- DDS implementation:
  - rmw_fastrtps_cpp

I am working on setting up the TurtleBot 4 to use the discovery server. The TurtleBot 4  has a Raspberry Pi 4B 4GB as the main processing unit, and the base is a [Create 3](https://iroboteducation.github.io/create3_docs/). The RPi and Create 3 are connected via a USB-C cable, which functions as the `usb0` interface. The RPi also connects to Wi-Fi on the `wlan0` interface, which is generally used to communicate with a desktop PC for visualisation. 

The network looks like this:

(User PC){192.168.0.10} <- Wi-Fi -> (Router){192.168.0.1} <- Wi-Fi -> {192.168.0.9}(RPi4){192.168.186.3} <- USB-C -> {192.168.186.2}(Create 3)

I enable ipv4 forwarding and run the discovery server on the RPi4, and configure both the User PC and the Create 3 to use the RPi as the discovery server. I also configure the RPi4 and User PC to be super clients. Finally, I add an ip route on the user PC to see the Create 3: `sudo ip route add 192.168.186.0/24 via 192.168.0.9`.

In this configuration, everything works well. I am able to see ros2 topics on both the PC and RPi, and pub/sub correctly.

What I am trying to do now is place the RPi4 into AP mode and have the system function without an external router like this:  
(User PC){10.42.0.170} <- Wi-Fi ->  {10.42.0.1}(RPi4){192.168.186.3} <- USB-C -> {192.168.186.2}(Create 3)

When I try this, I am able to `ros2 topic list` all of the topics on the User PC, but I cannot echo any data from Create 3 topics. RPi4 topics work fine still. I am still able to ping `192.168.186.2` from the PC as well. I can see in wireshark that the PC is sending packets to the Create 3, but there is no response. Interestingly, if I connect the PC to the RPi with an ethernet cable, then things start to work. In wireshark I can see that the PC sends data over Wi-Fi, and then receives a response over ethernet. 

I am assuming that there is either some setting that is preventing the Create 3 from forwarding data to the PC, or I am missing some routing configurations in AP mode. If anyone has some insight into what could be blocking ros2 data while allowing pinging and discovery, that would be appreciated.

clalancette, 2023-03-09:
@kscottz Is there a place for Turtlebot 4/Create 3 specific documentation?

roni-kreinin, 2023-03-09:
@clalancette https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html

kscottz, 2023-03-10:
This seems like it should be of reasonably high priority. My knowledge of DDS / FastRTPS is fairly scant. Would it be helpful if I escalate this issue to eProsima? I believe they could shed some light on the proper configuration to address this issue.

clalancette, 2023-03-10:
> Would it be helpful if I escalate this issue to eProsima?

Yeah, that would be the best.  I don't really know anything about the Discovery Server either.

MiguelCompany, 2023-03-15:
The issue is most probably in the routing configuration of the RPi4. Traffic is most probably sent from the create 3 with a destination that is being routed to the ethernet interface of the RPi4.

The discovery server only forwards DDS discovery traffic, that is why ros2 topic list works. User traffic works the standard way.
Have a look at the locators announced on the DATA(w) and DATA(r) messages. The addresses there are the ones that will be used for user traffic exchange.

MiguelCompany, 2023-03-15:
One more thing that can be useful is to have a specific port forwarding, and making the Create3 announce the WiFi address of the RPi4.

This could be achieved with the following XML

```xml
<?xml version="1.0" encoding="UTF-8"?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <profiles>

        <!-- Domain Participant Profile -->
        <participant profile_name="domainparticipant_profile_name" is_default_profile="true">
            <rtps>
                <!-- Reception unicast locator for user data traffic -->
                <defaultUnicastLocatorList>
                    <locator>
                        <udpv4>
                            <!-- Change this to RPi4 public address -->
                            <address>10.42.0.1</address>
                            <!-- Change this to forwarded port -->
                            <port>12345</port>
                        </udpv4>
                    </locator>
                </defaultUnicastLocatorList>

                <builtin>
                    <!-- Use same locator for builtin metatraffic -->
                    <metatrafficUnicastLocatorList>
                        <locator>
                            <udpv4>
                                <!-- Change this to RPi4 public address -->
                                <address>10.42.0.1</address>
                                <!-- Change this to forwarded port -->
                                <port>12345</port>
                            </udpv4>
                        </locator>
                    </metatrafficUnicastLocatorList>
                </builtin>
            </rtps>
        </participant>

    </profiles>
</dds>
```