ros2/rmw_fastrtps #678: foxglove-bridge err occur when use fastrtps

sgf201, 2023-03-23:
## Bug report

**Required Info:**

- Operating System:
  docker ros::humble
- Installation type:
  docker
- Version or commit hash:
 
- DDS implementation:
 fastdds
- Client library (if applicable):
 

#### Steps to reproduce issue
sudo apt install ros-humble-foxglvoe_bridge*
ros2 launch foxglove_bridge foxglove_bridge_launch.xml

#### Expected behavior
no error msg

#### Actual behavior
[image]

#### Additional information
no error occur  when I change dds to cyclonedds

## Feature request

#### Feature description

#### Implementation considerations
no error occur  when I change dds to cyclonedds

[image]

[image]

sgf201, 2023-03-23:
seems like get_default_topic_qos  the qos_policies is a abnormal value

clalancette, 2023-03-30:
@amacneil Could you have somebody take a quick look at this and see if this is something on the foxglove side or the Fast-RTPS side?

sgf201, 2023-04-06:
https://github.com/foxglove/ros-foxglove-bridge/pull/205

sgf201, 2023-04-06:
done

closed 2023-04-06 by sgf201