ros2/rmw_fastrtps #699: Cannot discover the endpoint after multiple nodes created and exited

fujitatomoya, 2023-07-24:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 22.04
- Installation type:
  - binaries, source build
- Version or commit hash:
  - Humble
- DDS implementation:
  - Fast-DDS
- Client library (if applicable):
  - rclcpp, rclpy

#### Steps to reproduce issue

The following can be reproducible only if it is in the localhost. (shared memory transport is enabled)

1. start the subscription.

```bash
# ros2 run examples_rclpy_minimal_subscriber subscriber_member_function
```

2. start the multiple publishers and kill all of them after 10 seconds.

```bash
# cat > pub_nodes.launch.py << EOS
from launch import LaunchDescription
from launch_ros.actions import Node
node_num_ = 100
def generate_launch_description():
    ld = LaunchDescription()
    for i in range(node_num_):
        ld.add_action(
            Node(
                package='examples_rclpy_minimal_publisher',
                executable='publisher_member_function',
                name='pub_node_{}'.format(i),
            )
        )
    return ld
EOS

# ros2 launch ./pub_nodes.launch.py
...<snip>

### Send Ctrl-C after a short while (about 10 seconds) to kill all publishers
```

3. Start a single publisher in another terminal

```bash
# ros2 run examples_rclpy_minimal_publisher publisher_member_function
```

4. Check the subscription terminal see if that can receive the message from the publisher with procedure-3.

#### Expected behavior

subscription can receive the message with procedure-4.

#### Actual behavior

subscription cannot receive the message with procedure-4.

#### Additional information

- this problem cannot be observed with rmw_cyclonedds.
- this problem only can be observed with localhost communication. (via network interfaces, there is no such problem.)\
- when this issue occures, `ros2 node list --no-daemon` does not show subscriber running in first terminal.
- when SHM for fastrtps disabled and UDP transport enabled, this issue does not reproduce.

```xml
 <?xml version="1.0" encoding="UTF-8" ?>
 <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
     <transport_descriptors>
         <transport_descriptor>
             <transport_id>CustomUdpTransport</transport_id>
             <type>UDPv4</type>
         </transport_descriptor>
     </transport_descriptors>

     <participant profile_name="participant_profile" is_default_profile="true">
         <rtps>
             <userTransports>
                 <transport_id>CustomUdpTransport</transport_id>
             </userTransports>

             <useBuiltinTransports>false</useBuiltinTransports>
         </rtps>
     </participant>
 </profiles>
```

fujitatomoya, 2023-07-24:
CC: @iuhilnehc-ynos @Barry-Xu-2018 @MiguelCompany

MiguelCompany, 2023-07-24:
@fujitatomoya We have recently merged some changes to Fast DDS dealing with this exact problem. Could you verify if this is fixed with the current head of [2.6.x](https://github.com/eProsima/Fast-DDS/tree/2.6.x) (i.e. https://github.com/eProsima/Fast-DDS/commit/3fffea5ab04fbf2241a32d51f1513886edf44c72) ?

Barry-Xu-2018, 2023-07-24:
@MiguelCompany 

The version I am using is [d0cc5a6d4cf124ed42b8df061cd857144e372888](https://github.com/eProsima/Fast-DDS/commit/d0cc5a6d4cf124ed42b8df061cd857144e372888), which is two commits behind the latest version.  
And these 2 commits aren't related to SHM codes.   
At this version, this problem also can reproduce in my environment.

MiguelCompany, 2023-07-24:
@Barry-Xu-2018 As the report stated `Humble` as the version, I understood it was related to Fast DDS 2.6.x. Your comment seems to suggest this is also reproduced in `Rolling` ?

Barry-Xu-2018, 2023-07-24:
@MiguelCompany Yes. I also speculated whether subsequent updates have fixed this issue, so I tried using the rolling version.

Barry-Xu-2018, 2023-07-24:
@MiguelCompany
CC: @fujitatomoya 

While issue occurs，

There is one thread in the `subscriber_member_function` that consumes almost 100% of CPU usage.

<details><summary> abnormal CPU usage </summary>
<p>

```shell
top - 15:16:41 up  4:33,  2 users,  load average: 14.46, 9.73, 8.67
Threads:  12 total,   1 running,  11 sleeping,   0 stopped,   0 zombie
%Cpu(s):  6.2 us,  0.8 sy,  0.0 ni, 92.9 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :  31804.4 total,   5983.9 free,  18137.0 used,   7683.6 buff/cache
MiB Swap:  65536.0 total,  65475.2 free,     60.8 used.  13057.1 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND  
 581173 chenlh    20   0  698376 105084  68464 R  99.9   0.3   0:26.56 subscriber_memb    # consume 100% CPU usage
 581140 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:09.58 subscriber_memb
 581167 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.00 subscriber_memb
 581168 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.00 subscriber_memb
 581169 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.00 subscriber_memb
 581170 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.85 subscriber_memb
 581171 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.48 subscriber_memb
 581172 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.00 subscriber_memb
 581174 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.00 subscriber_memb
 581175 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.19 subscriber_memb
 581176 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.60 subscriber_memb
 581177 chenlh    20   0  698376 105084  68464 S   0.0   0.3   0:00.03 subscriber_memb    
```

Check the backtrace of the thread `581173`  

<details><summary> backtrace </summary>
<p>

```cpp

(gdb) info thread
  Id   Target Id                                            Frame 
* 1    Thread 0x7f713f38e1c0 (LWP 581140) "subscriber_memb" __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55adaa858260) at ./nptl/futex-internal.c:57
  2    Thread 0x7f713cc6e640 (LWP 581167) "subscriber_memb" __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7f713dd7c1e0 <(anonymous namespace)::g_signal_handler_sem>) at ./nptl/futex-internal.c:57
  3    Thread 0x7f71387ff640 (LWP 581168) "subscriber_memb" __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x7f71387fede0, op=137, expected=0, futex_word=0x55adaa1d7900) at ./nptl/futex-internal.c:57
  4    Thread 0x7f7137ffe640 (LWP 581169) "subscriber_memb" __futex_abstimed_wait_common64 (private=939514416, cancel=true, abstime=0x7f7137ffdc10, op=137, expected=0, futex_word=0x55adaa1da710) at ./nptl/futex-internal.c:57
  5    Thread 0x7f71377fd640 (LWP 581170) "subscriber_memb" __futex_abstimed_wait_common64 (private=72506686, cancel=true, abstime=0x7f71377fcc10, op=137, expected=0, futex_word=0x55adaa1dd45c) at ./nptl/futex-internal.c:57
  6    Thread 0x7f7136ffc640 (LWP 581171) "subscriber_memb" 0x00007f713f127934 in __libc_recvfrom (fd=15, buf=0x55adaa1eb720, len=65500, flags=0, addr=..., addrlen=0x7f7136ffb69c) at ../sysdeps/unix/sysv/linux/recvfrom.c:27
  7    Thread 0x7f71367fb640 (LWP 581172) "subscriber_memb" 0x00007f713f127934 in __libc_recvfrom (fd=16, buf=0x55adaa1fbed0, len=65500, flags=0, addr=..., addrlen=0x7f71367fa69c) at ../sysdeps/unix/sysv/linux/recvfrom.c:27
  8    Thread 0x7f7135ffa640 (LWP 581173) "subscriber_memb" 0x00007f713d3681e0 in _Unwind_Find_FDE () from /lib/x86_64-linux-gnu/libgcc_s.so.1
  9    Thread 0x7f71357f9640 (LWP 581174) "subscriber_memb" 0x00007f713f127934 in __libc_recvfrom (fd=18, buf=0x55adaa20e8a0, len=65500, flags=0, addr=..., addrlen=0x7f71357f869c) at ../sysdeps/unix/sysv/linux/recvfrom.c:27
  10   Thread 0x7f7134ff8640 (LWP 581175) "subscriber_memb" __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x7f7134ff74a0, op=265, expected=0, futex_word=0x7f713dda0110) at ./nptl/futex-internal.c:57
  11   Thread 0x7f71347f7640 (LWP 581176) "subscriber_memb" __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55adaa227418) at ./nptl/futex-internal.c:57
  12   Thread 0x7f7133ff6640 (LWP 581177) "subscriber_memb" __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7f7124000c04) at ./nptl/futex-internal.c:57
(gdb) thread 8
[Switching to thread 8 (Thread 0x7f7135ffa640 (LWP 581173))]
#0  0x00007f713d3681e0 in _Unwind_Find_FDE () from /lib/x86_64-linux-gnu/libgcc_s.so.1
(gdb) bt
#0  0x00007f713d3681e0 in _Unwind_Find_FDE () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#1  0x00007f713d364833 in ?? () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#2  0x00007f713d3661e4 in _Unwind_RaiseException () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#3  0x00007f713d0ae50b in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f713b3e18f7 in boost::interprocess::shared_memory_object::priv_open_or_create (this=0x7f7135ff9150, type=boost::interprocess::ipcdetail::DoOpen, filename=0x7f712021fee0 "fastrtps_fed84726591b50b4", mode=boost::interprocess::read_write, perm=...) at /home/chenlh/Projects/ROS2/ros2-master/src/eProsima/Fast-DDS/thirdparty/boost/include/boost/interprocess/shared_memory_object.hpp:363
[... cut ...]
1296	            segment = std::make_shared<SharedMemSegment>(boost::interprocess::open_only, segment_name);
(gdb) p segment_name
$5 = "fastrtps_fed84726591b50b4"
```

but the file `fastrtps_fed84726591b50b4` does not exist in `/dev/shm`.

```shell
$ ls /dev/shm/fastrtps_fed84726591b50b4
ls: cannot access '/dev/shm/fastrtps_fed84726591b50b4': No such file or directory
```

At this scenario，subscriber always try to open deleted shared memeory file .  

BTW, we think this problem is relevant to another issue https://github.com/ros2/ros2cli/issues/582#issuecomment-1321799828.

MiguelCompany, 2023-07-24:
@Barry-Xu-2018 Thank you very much for the investigation. I was getting almost to the same place on my debugging session.

I've been testing some changes that solve the 100% CPU usage, but they do not always solve the main issue (i.e. step 4 sometimes succeeds, and sometimes fails).

We did a workaround on https://github.com/eProsima/Fast-DDS/commit/687104a269eb58491d7d9498390dd99543ff86e9 which fixes this issue. We are considering whether to add that commit into the supported branches, but we first need to evaluate the impact of incorporating those changes.

Barry-Xu-2018, 2023-07-25:
@MiguelCompany Thank you for your quick response.  

The workaround https://github.com/eProsima/Fast-DDS/commit/687104a269eb58491d7d9498390dd99543ff86e9 is to avoid using SHM for metatraffic if there is other transport (such as UDP).  
Could you briefly introduce what information belongs to metatraffic? If this information is transmitted through UDP, would it affect the performance of SHM transmission?

Barry-Xu-2018, 2023-07-25:
I found the definition on metatraffic in DDS spec.  
[image]

So this information doesn't directly affect the performance of SHM transmission.

MiguelCompany, 2023-07-25:
@Barry-Xu-2018 Correct. Metatraffic is basically discovery traffic.

MiguelCompany, 2023-07-27:
@Barry-Xu-2018 I made some bugfixes on eProsima/Fast-DDS#3759, and made the meta-traffic be transmitted on UDP by default on eProsima/Fast-DDS#3753.

With those two on a ROS 2 rolling workspace, I could not reproduce this issue anymore.

Barry-Xu-2018, 2023-07-28:
@MiguelCompany  

Thank you for the further correction on this issue.   
In my environment, I also cannot reproduce this issue with these 2 patches.

fujitatomoya, 2023-07-28:
> In my environment, I also cannot reproduce this issue with these 2 patches.

neither can I. thanks for the patches.

> BTW, those 2 patches will backport to the version of FastDDS used by Humbel. Right?

https://github.com/eProsima/Fast-DDS/pull/3753 includes ABI break change, but I am not sure Fast-DDS policy to backport. @MiguelCompany friendly ping?

MiguelCompany, 2023-07-31:
@fujitatomoya There's no ABI break. We will backport both changes to the iron and humble branches

ZhenshengLee, 2023-08-09:
> We have already released Fast DDS v2.6.6 (for humble), and will release v2.11.2 (rolling), and v2.10.2 (iron) this week or next.

@MiguelCompany  Any possibility of backportting this to v2.0.x(foxy) which is the last LTS ros2 version?

Thanks.

clalancette, 2023-08-09:
> @MiguelCompany Any possibility of backportting this to v2.0.x(foxy) which is the last LTS ros2 version?

Unfortunately, no.  Foxy is now End-of-Life, so we have no plans to do updates to it.

The good news is that Humble is the next LTS, which should have these fixes in it.

fujitatomoya, 2023-09-01:
@MiguelCompany yeah, right. i will go ahead to close this one.

closed 2023-09-01 by fujitatomoya

iceriver97, 2025-12-19:
I'm using Fast DDS v2.6.6 (for humble) and I still encounter situations where the publisher is not visible in the ROS2 node list; the Fast DDS version in the Docker container containing the publisher is 2.6.8, and the subscriber version is 2.6.6. I've found that after restarting another ROS2 node multiple times, the publisher's related maps are sometimes deleted.

```
root@integration02:/dev/shm# cat /proc/3717572/maps | grep fastrtps_port
ffffb9923000-ffffb9930000 rw-s 00000000 00:19 64055                      /dev/shm/fastrtps_port28150 (deleted)
ffffb99a3000-ffffb99b0000 rw-s 00000000 00:19 64102                      /dev/shm/fastrtps_port28163
ffffb99e3000-ffffb99f0000 rw-s 00000000 00:19 64055                      /dev/shm/fastrtps_port28150 (deleted)
ffffb9e33000-ffffb9e40000 rw-s 00000000 00:19 64099                      /dev/shm/fastrtps_port28162
ffffba333000-ffffba340000 rw-s 00000000 00:19 63982                      /dev/shm/fastrtps_port28161
ffffbbf34000-ffffbbf41000 rw-s 00000000 00:19 63977                      /dev/shm/fastrtps_port28160
```